jeelabs/esp-link #1: Usage Question About ESP Board Versions

kd-kraken, 2015-05-28:
The readme states that GPIO pins 12 and 13 are needed. I have the ESP-01 breakout version of the esp8266. It doesn't appear to have pins 12 and 13 broken out. Can I still use this module or do I need to purchase a different version?

vickash, 2015-05-28:
I'm wondering this as well. GPIO 0 and 2 (which ARE broken out on the ESP-01) are both used for status LEDs according to the README. If that's correct, then no, it won't work on the 01.

This project looks like it exactly suits my needs, so I assumed the 01 wouldn't work and ordered an ESP-12 to try with. Still waiting on it, but GPIO 12 and 13 are available there, so it should work.

However, the ESP-01 is more appealing if you're "slaving" the ESP to another microcontroller. It's much smaller in (almost) bread-board-friendly form and you don't really need the extra pins.

I think it would be better to use GPIO 0 and 2 to reset the "master" boards, and use the less-available 12 and 13 for status LEDs. Unless there is some technical issue that prevents this. I have no idea about programming the ESP itself.

tve, 2015-05-28:
The source or binary is not usable with the -01 module as-is. I started with a -01 myself and used gpio0/2 as you guys say. The annoying thing is that the ESP requires these pins to be tied high at boot (and gpio2 low for reflashing). By putting the LEDs on these pins I make double-use of them as pull-ups and I don't end up with side-effects when driving them low to reflash the ESP. It would be relatively straight-forward to change the pins in the code. It's something I want to make easier, but I don't know that I will get to it this week.
I'm using the -03 module 'cause it's nice and small. It's also pretty easy to add normal headers to it with some soldering...

vickash, 2015-05-28:
Ah. That makes sense. Forgot that those are used to flash the esp itself.
Will check out the 03.

kd-kraken, 2015-05-28:
I had already ordered a -01, it's still in the mail. Without having it in front of me I don't have a sense of how small it is. Would it be reasonable to solder a couple jumpers wires directly to the esp8266 chip's solder pads and break out gpio 12 and 13 that way? Then just use your code as is?

BTW, I'm in agreement with vickash, this project looks like exactly what I'm looking for. Thank you tve.

tve, 2015-05-28:
While it is possible to solder to the chip directly, it's really really extremely small. I solder 0603 SMD components regularly and have dealt with 0201 and I wouldn't try this :-). Looks easier to me to change the #defines in the code...

tve, 2015-05-28:
I'm reopening this issue to make it easier to switch pin assignments. This is bound to come up more... The steps are:
 - create global variables to hold gpio pin numbers for reset, isp, conn-led, ser-led
 - create ability to store these settings in flash
 - add drop-down to esp-log web page to change the pin assignments

kd-kraken, 2015-06-02:
I'll be looking forward to this enhancement.

For the time being, to switch the pins all I need to do is edit lines 15-22 in serbridge.c? Or will there be more to it?

tve, 2015-06-02:
Also lines 317-318

tve, 2015-06-05:
This is addressed in release 0.9.4 using definitions in the Makefile. Still open is to make it more configurable through the web UI and store the settings in flash..

tve, 2015-06-19:
gpio pin assignments are configurable on the web pages as of release 0.10.0

closed 2015-06-19 by tve