jeelabs/esp-link #4: Build Instructions Tips/Clarifications

kd-kraken, 2015-06-02:
I just did a fresh install of everything on a Mint 17.1 32bit virtual machine in VirtualBox.

I took a few notes of what I had to do to get it running and thought it would be helpful to other newbies. These steps were omitted in the readme.

I had to install zlib1g-dev:
```
sudo apt-get install zlib1g-dev
```

I also had to install esptool:
```
sudo apt-get install python-serial
cd <path to esp8266 directory>
git clone https://github.com/themadinventor/esptool.git
```
I also went ahead and added the absolute path for esptool.py to the esp-link Makefile.

Double check the esp-link makefile for paths, make sure you have the correct path to the esp_iot_sdk (mine was the incorrect version).

Finally, don't forget to make sure that the esp-open-sdk/xtensa-lx106-elf/bin is in the PATH.

tve, 2015-06-02:
Thanks for the info, i will incorporate. I now have a set-up that can use esptool.py so i can test it. You've figured this out already, but https://github.com/jeelabs/esp-link/blob/master/BOARDS.md may help someone else reading this.

tve, 2015-06-19:
I incorporated what I could into the readme #1dfb379

closed 2015-06-19 by tve