jeelabs/esp-link #5: Adjustable baud rates and inverting resets

kd-kraken, 2015-06-05:
It would be good to have an adjustable baud rate for the mcu. Some microcontrollers require different bauds to reprogram, ie the arduino pro minis I'm using only accept new sketches at 57600.

Also, because my arduino is 5v I'm using an npn transistor to switch the arduino's reset. This seemed safer to do than wiring the esp's 3v gpio directly to the arduino's 5v reset pin. But to do this I had to invert all the GPIO_OUTPUT_SET(mcu_reset_pin, X) calls in serbridge.c.

Having these settings user adjustable could be very handy.

Anyway, now I am up and running. OTA flashing of the esp-01 and arduino pro mini working like a charm.

jcw, 2015-06-05:
There's no risk connecting a 3.3V output to a 5V input pin. The reset pull-up is about 50 kΩ, which will generate a weak 30 µA current through the ESD protection diodes of a logic chip running at 3.3V.

tve, 2015-06-05:
I'm glad you were able to get running! I will keep your baud rate request in mind (and I also welcome pull requests!).

Note that I committed some bug fixes a few minutes ago that you will want to pull (the change in user/cgi.c in https://github.com/jeelabs/esp-link/commit/c7c1d510ef4d7d596b62a83efddaa10edbd7a312 specifically)

vickash, 2015-06-05:
+1 for this feature. Adjustable baud rate would be great.

If, somewhere within the web interface, we could save a "running baud rate", and a "flashing baud rate" separately, it would be even more useful.

Consider the Leonardo boards, which need to be flashed at 57600. What if I want my sketch to communicate at 115200? Assuming the sketch is running and I want to upload a new version, I'd first have to change from 115200 to 57600, upload, then go from 57600 back to 115200 to continue communicating with the board.

If esp-link can detect whether it's getting a "regular" vs "upload" connection, and it has the two appropriate rates saved, it should be able to automatically do this switching for us.

The user will still need to know the upload rate for their particular chip. This file from the Arduino project shows just that:
(Search for "upload.speed")
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/boards.txt

jcw, 2015-06-05:
The telnet protocol has support for changing baudrates. Getting the upload rate right is essential, but The second one would fall more into the "nice to have" category, IMO (and ARM chips can often do auto-baud in their boot loaders). The more modes the esp-link software can be in, the more trouble it becomes to move this unit around and re-use it.

kd-kraken, 2015-06-05:
@jcw, Ah okay, perhaps the use of the transistor isn't necessary for safety's sake. But having the ability to invert the reset pin's output calls could still be useful. In issue #1 tve explained that using the ESP-01 module was a problem because using GPIO0 & 2 are needed low for OTA flashing the ESP. The use of transistors and this output inverting could get around that problem, correct?

jcw, 2015-06-05:
Agree - if the power up defaults prevent proper startup, then sure, a transistor or inverter can help.

kd-kraken, 2015-06-05:
These are the changes I made to make life easier for changing baud rates and inversing the reset pin states: https://github.com/jeelabs/esp-link/compare/master...kd-kraken:master

I hesitate to open a pull request because it seems to me that these may be better to make this adjustable at runtime from the web interface.

tve, 2015-06-07:
FYI, release 0.9.5 supports changing the baud rate from the web interface. The new rate isn't saved, so it will revert after a reset. Changes in https://github.com/jeelabs/esp-link/commit/1b1ee532deb53fa000b6279a60a6d56d6df15867

tve, 2015-06-19:
Open item: inverting reset

tve, 2015-06-28:
I'm going to close this, if someone feels strongly about it, please reopen. It's difficult enough to control reset at power-up and boot when active-low, I don't want to figure it out both ways...

closed 2015-06-28 by tve