jeelabs/esp-link #8: esp-link RS-232 bridge

raboof, 2015-06-06:
I'm quite new to DIY electronics, so I'd like to run this by you.

I have a device that can be controlled over RS-232 - but I'd like to control it wirelessly. 

Do I understand correctly that an ESP8266 combined with a MAX232 could provide that? And that based on this project I could create a firmware that connects to the wifi, after which it provides a (authenticated, perhaps REST-ish) API to send commands to the device over RS-232? Or am I misunderstanding some things? :)

tve, 2015-06-07:
What you describe is certainly feasible. Currently esp-link doesn't provide a bridge between http and the serial link, but that's certainly one of the goals. First I need to get the web interface features finished.

tve, 2015-06-07:
Closing this issue as it doesn't represent any to-do item. Please reopen if you want to continue the discussion.

closed 2015-06-07 by tve