jeelabs/esp-link #11: Feature suggestions, baud rate and other parameters.

fileark, 2015-06-07:
First things first, this is the exact application I need for my home automation projects and I think a lot of others do too. A cheap  transparent wifi to serial gateway. Thank you thank you.
 
Here are some observations and requests.
1. The web interface wifi page seems to work with Firefox but not internet explorer 11, on IE it never loads the list of detected APs. (if you don't want to support IE that’s fine, just mention in documentation which browser to use)
2. After the ESP8266 connects to an AP successfully, it doesn't show the IP address assigned so I have to search for it manually. 
3. Powering down my router/AP does not make the module go back  into AP mode. Indeed the module goes from consuming 20ma to 70ma which indicates it is trying to go back to AP mode but I can't discover it until re-flashing the unit.
4. Selectable baud rate that is saved in eprom is a critical feature. Lots of devices either cant or shouldn't communicate at 115200. Right now when I try to communicate through the ESP8266 to an Arduino uno using modbus it wont work. I don't know if this is due to the speed or something else.
5. You use port 23, it would be nice to be able to customize the port but not critical, at the very least you might consider putting the port on the web interface as a status so people will more easily know what port is used?
6. It would be nice to be able to see and set serial port settings. For instance: Speed, Parity, Data bits, Stop bits. I don't know what parameters you are using, I am not sure you have mentioned it in documentation? 115200,8,n,1?
7. Static IP Please. Dynamic works well for more modern protocols like MQTT however lots of common protocols like Modbus really requires a static IP if you are going to put something in production (i.e. not have to fix it when your router resets).
8. All of these settings need to persist through power cycle. If the default firmware that comes on these units would allow persistence for everything I would just bite the bullet and power through the pain of AT commands,  but some of the critical settings reset on power cycle making the ESP8266 a toy so far... 

I am sure someone has done the things I need before, maybe with an Arduino sketch or something else, however I have spent many hours and can't find an example anywhere. Sorry, I know this is a laundry list. I am used to having these features in industrial devices, obviously for my personal use I would like it in a $5.00 package! Keep up the good work and I look forward to your progress!
Thanks.

P.S. I am using an ESP-7 that came soldered down on a white breakout board.

tve, 2015-06-07:
Thanks for the long feeback! You kind'a want everything, no ;-)

I'll see how far I get before I run out of steam. At some point you may just have to bite the bullet and compile your own version in order to change some of the hard-coded settings...

WRT returning to AP, when you power-up the module it will try to reassociate. If after ~20 seconds it has not been able to it should switch into STA+AP mode again on its own. The "conn" LED, if you have that connected, should give you an indication of that (see README for the blink codes). If the LED indicates that it's connected and has an IP address (LED mostly on with a short interruption every 3 seconds) and you can't get to it then it may have gotten a different IP by your DHCP server.

WRT static IP, I recommend you create a static lease in your DHCP server, that's what I've been doing.

Persistence for settings is next on the to-do list, after I fix the refactoring of all the web pages that I'm currently deep into. I hope this fixes IE problems (I haven't tried IE myself).

kd-kraken, 2015-06-08:
It occurs to me that it might be prudent to have some kind of security when it comes to mcu reset and flashing. Maybe something as simple as a having a toggle switch in the webgui that enables/disables the mcu reset functionality of the esp. This toggle switch would be password protected, and disabled by default.

tve, 2015-06-19:
Issues fixed in release 0.10.0:
1. IE support fixed (I don't know which versions exactly)
2. Show IP address after connecting
5. port 23 command line example shown on home page
8. pin assignments now saved to flash, baud rate saving coming soon

Issues I don't understand:
3. Going back in to AP mode: after a reset the esp-link turns on its AP if it cannot connect to the configured network within 15 seconds. Is this not occurring? What does the CONN LED flash pattern indicate?

tve, 2015-06-19:
I'm closing this issue. Some items have been fixed. For some I created separate issues. If I'm missing something critical, please open a separate issue.

closed 2015-06-19 by tve