jeelabs/esp-link #13: attached uC reset multiple times at boot-up

tve, 2015-06-12:
Seen using esp-bridge and arduino

JohnOH, 2015-06-13:
There is more to this powering up with a device connected. I mentioned the apparent multiple resets at power on of my JeeNode Block earlier. It transpires that I can typically only get a working TCP connection to it if I connect to the ESP after powering it up. If they are connected at ESP power on then nc 192.168.1.24 23 typically connects with no action - although there are exceptions where a "." or a "?" can get the RF12Demo menu working.
The upset appears to be in the JeeNode since hitting its reset button restores operation.

JohnOH, 2015-06-13:
With a JeeNode connected to the ESP-Bridge the uC Console initially contains:
������怘�怘�����
Perhaps a hex option?
After first connection is made there is no change in the uC Console.
After sending a "." towards the ESP-Bridge I am unable to get a page from the Bridge.

jcw, 2015-06-13:
is the uploaded sketch set to 115200 baud?

JohnOH, 2015-06-13:
JeeNode interface is at 115200.

JohnOH, 2015-06-13:
Having corrected the JeeNode speed mistake the problem is in the main as before. The web server is sometimes corrupting the returned webpage and then going absent other times just going immediately absent.

JohnOH, 2015-06-13:
esp-link - 0.9.4 - 2015-06-03 21:44:51 - c5eb65a

Is being overwhelmed by 115200. I will hold for a 0.9.5 version of ESP-Bridge.

tve, 2015-07-04:
I verified that 0.10.3 no longer resets the attached uC if esp-link is reset

closed 2015-07-04 by tve