jeelabs/esp-link #18: JN-ESP Overheating

JohnOH, 2015-06-13:
With a program outputing a few text lines each second the second chip (AH1444) on the ESP board is getting hot, 60 degree centigrade and rising.
https://files.gitter.im/jeelabs/n9KV/blob

Is this normal?

JohnOH, 2015-06-18:
Tried using 0.9.3 and still seeing the overheating of what may be the flash chip.

tve, 2015-06-20:
I modified the vcc sketch to output every 100ms and I can't get the esp to go above 120F (~50C). This is using a physical probe, so IR may show a little higher. The esp8266 gets about 2-3 degrees C hotter than the flash chip. If you think that yours gets much hotter, please let it run and let's see what happens. If you kill it I'll send you a replacement ;-).

jcw, 2015-06-20:
Note that 50°C is not very hot for a chip (and only ≈ 30° above ambient), everything can probably rise to well over 100° without any risk. Not saying there's no issue, just that it won't die soon from this, IMO.

JohnOH, 2015-06-20:
That is a relief, no plans for battery operation.

susisstrolch, 2015-06-20:
Ehmmm....
> 5V power supply.

 =without= regulator? Those ESP8266 thingies are supposed to run with 3.3V - max. 3.6V.
So I wouldn't wonder at all that its start sweating...

JohnOH, 2015-06-20:
There is a regulator, down to 3v3.

piperpilot, 2015-06-25:
One thing I have noticed...when in SOFTAP mode, the ESP8266 gets pretty hot...but once switched over to STA mode, it runs nice and cool.  Might have to do with the overhead of SSID broadcast, etc.

tve, 2015-08-05:
the esp8266 always has the radio RX enabled while in AP mode, this causes ~70mA of current consumption, while it mostly sits at ~12mA while in STA mode.
I'm going to close this ticket, I don't think this is something that esp-link really has an impact on. Please correct me if you think otherwise.

closed 2015-08-05 by tve