jeelabs/esp-link #19: Secure configuration flag

tve, 2015-06-19:
Insecure:
- esp-link runs in STA+AP mode for 15 seconds at reset
- reprogramming and reset via port 23 is allowed
Secure: (not really secure, sigh)
- AP mode only after reset if it cannot connect
- no reprogramming or reset via port 23

replaces #11 item 3

vickash, 2015-06-29:
The web interface also needs some kind of authentication to be secure. An attacker could change baud rate and cause you to lose data, or change the WiFi settings and you lose connection altogether.

Realized this while scripting GET requests to toggle between "operating" and "flashing" baud rates. Nice feature though. :+1:

dreamlayers, 2017-02-03:
> esp-link runs in STA+AP mode for 15 seconds at reset

> AP mode only after reset if it cannot connect

These don't have to be a problem anymore because you can configure the AP to use encryption and provide a password.

For the paranoid the problem is that the password is initially input insecurely via the web interface. Though you could flash the AT command firmware and configure there before flashing esp-link. (The WiFi settings are managed by the SDK and shared.) You can also set these things via the Makefile.

Unless you're interfacing to something where security or safety is critical, messing with port 23 would mainly be a nuisance. Though there certainly ought to be a way to have authentication there.

The most disturbing security issue is the ability to upload new ESP8266 firmware without authentication.  That could create a rogue device on your network!

Earlier today I created a new issue regarding security, #258. It may be considered a duplicate of this issue.

tve, 2017-02-03:
Yup. IMHO in the current state it does not make a whole lot of sense to add security features without also going to HTTPS for the web pages and I don't have the energy to mess with that and all the certs fumbling on the esp8266...
Setting crypto on the AP is supported in the Makefile. It's also an easy way to hang yourself :-)

dreamlayers, 2017-02-03:
If you're sending cleartext passwords that's kind of pointless. It can only stop the least determined attackers. It's probably better than nothing, but it also gives a false sense of security.

Maybe [digest authentication](https://en.wikipedia.org/wiki/Digest_access_authentication) could be used. That way the password should be protected even without HTTPS. If you perform the initial setup securely, and then only interact with the attached microcontroller, this could be reasonable. I think the only risks are eavesdropping and man in the middle attacks then.

overisberg, 2018-07-09:
I have a proposal for basic security for esp-link.
Why not just use a pin on the esp for security, when the pin is low you are allowed to change configuration and change firmware on the esp or the connected arduino and if it is high you are not allowed to change configuration but the web interface still works.
You need physical access to the esp to make changes and you can even add a switch with a physical key if you  want.
HTTPS + a password stored as a hash is much better but much harder to implement... this works now.

You can look at my implementation:
https://github.com/overisberg/esp-link/commit/ad2ba1caa7da1daa53d4f0bf986fc2ad633fab06

All constructive comments are welcome.

tve, 2018-07-09:
Nice idea!