jeelabs/esp-link #24: Improve look on small/mobile screen

piperpilot, 2015-06-24:
On a small screen in portrait mode, the wifi config screen is pretty much unusable.  The same goes for the home screen.  It would be great if this were adaptive and put the boxes in a vertical orientation when the screen size is smaller.  I have attached a couple of images.  Landscape looks fine, but you will see what I mean about portrait.
[image]
[image]

tve, 2015-06-24:
Yup. Thanks for trying it out! Do you have the CSS skills to add the appropriate media queries?
The layout uses the Pure-CSS grid and so is totally ready for the adaptive layout. There's even the Pure-CSS adaptive grid, but the files are too big to just pull in (sigh). Someone has to spend time extracting just the right CSS fragments...
It's on my to-do list, but I first want to get to v1 feature complete...

piperpilot, 2015-06-24:
I was just looking at this.  I'm gonna try adding the grids-responsive-min.css via Yahoo CDN and see what it does...from there I might be able to strip out just whats needed.

vickash, 2015-06-24:
Quick and dirty fix:

````css
@media screen and (max-width: 1023px) {
	.pure-u-12-24 {
  		width: 100%;
	}
}
````

You can try to break at 767px instead, but the wifi table ends up overlapping the other div @ 768px even, just like your screenshots. Depends on the length of your IP and Network name though.

You could make all that wrap, but that isn't much easier to read anyway Better to just push the other div below.

piperpilot, 2015-06-24:
OK, I have a fix for this...it added a custom grid-responsive-min.css which is 706b on disk.  Total flash size:

** user1.bin uses 239216 bytes of 241664 available

I'd be happy to contribute...it changes the grid to 4 columns since not much more is really needed and defines a medium size of 1024 which will stack at smaller than that.  I'm not exactly sure how to do a pull request/diff, if someone wants to help, I can submit it.

tve, 2015-06-24:
Awesome! I will look at both versions and incorporate into the next release. I need to finish the static-IP and hostname stuff for that.

vickash, 2015-06-24:
@tve, according the repo, you already are including the entire Pure CSS file in the responsive version. Without it the side navigation would not collapse I think. I compared `pure-min.css` from the repo to the one from purecss.io and they're identical. The non-responsive one should be named `pure-nr-min.css`.

Do you mean that you want to reduce what's included to just the necessary components to save space?

Also, what about gzip? Can the web server on the 8266 serve up a CSS file in gzip encoding? The gzipped version of that file is a good 12 to 13K less.

piperpilot, 2015-06-24:
no @vickash none of the grid media queries are in the included pure-min.css.  I have a grid.css that has them...looks something like:

@media screen and (min-width: 48em) {
    .pure-u-md-1,
    .pure-u-md-1-1,
    .pure-u-md-1-2,
    .pure-u-md-1-4,
    .pure-u-md-2-4,
    .pure-u-md-3-4,
    .pure-u-md-4-4 {

with more obviously.  I have a diff created, but can't figure out where to upload it for the issue.

vickash, 2015-06-24:
Ah, I see. Just read all the way to the end of the Get Started page for Pure. I assumed that the default media queries were included. That the classes were already named things like "pure-12-24" also threw me off. I understand how it works now.

Dropping down to a max of 4 columns is a good idea to keep the file size down. 197 lines vs 900+ before minifying when I checked. gzip still won't hurt though.

You need to be editing files in the html folder of the repo. Inside there you'll find all the CSS. You can add css files in there and edit "home.tpl" to include them. You'll need to change some class names in "home.tpl" as well I think.

vickash, 2015-06-24:
Wow. Just realized the entire project is compressed with heatshrink. gzip will be probably no help.

tve, 2015-06-24:
All CSS and js files are compressed using gzip and shipped as-is to the browser. The .tpl files use heatshrink and get a boilerplate header prefixed, which is something I want to eliminate and prefix at make-time and then gzip. I believe this will make things smaller overall, but I have to actually try it out...

tve, 2015-06-29:
fixed in #1f8cdb6

closed 2015-06-29 by tve