jeelabs/esp-link #29: improve power when disconnected

tve, 2015-07-04:
When the esp8266 is configured to connect but can't it keeps scanning and tring to connect, using 70mA the whole time. Something should be done to make it reconnect less often.

Code provided by iothing http://www.esp8266.com/viewtopic.php?f=34&t=3445&start=80#p22358:
```
static void ICACHE_FLASH_ATTR resetTimerCb(void *arg) {
  int x = wifi_station_get_connect_status();
  int m = wifi_get_opmode() & 0x3;
  os_printf("Wifi check: mode=%s status=%d\n", wifiMode[m], x);

  if (x == STATION_GOT_IP) {  //connected successfully and got ip
    if (m != 1) {  //not already in sta mode

#ifdef CHANGE_TO_STA
      // We're happily connected, go to STA mode
              os_printf("Wifi got IP. Going into STA mode..\n");
              wifi_set_opmode(1);  //stay in the mode we are in!
              wifi_set_sleep_type(SLEEP_MODE);
#endif
    }
    //continue to check on the wifi state
    os_timer_arm(&resetTimer, RESET_TIMEOUT, 0); //check again in RESET_TIMEOUT seconds
  } else {
    if (m == 3) { //if we are in AP+STA Mode
      if (retry) {
        os_printf("WiFI: no ip, retry...\n");
        wifi_station_set_reconnect_policy(true); //attemp to reconnect
        wifi_station_connect();//reconnect
        retry = false; //unset the retry flag
        os_timer_arm(&resetTimer, 10000, 0); //check again in 10s
      }
      else  {
        os_printf("WiFI: no ip, wait...\n");
        wifi_station_set_reconnect_policy(false); //disable auto-reconnect
        retry = true; // set retry flag
        os_timer_arm(&resetTimer, 120000, 0); //check again in 120s
      }
    }

    if (m == 1) {  //in STA mode and no ip
      os_printf("STA->AP+STA\n");
      wifi_set_opmode(3); //change to AP+STA mode so the user can correct the wifi settings
      wifi_station_set_reconnect_policy(false); //disable auto-reconnect
      retry = true; // set retry flag
      os_timer_arm(&resetTimer, 120000, 0); //check again in 120s
    }
    //if in AP mode (default after flash) do nothing
  }
}
```

tve, 2015-08-05:
added in https://github.com/jeelabs/esp-link/commit/cf66979595b57c03c09865a797b8800a95dec1b9

closed 2015-08-05 by tve

blackadmin, 2017-07-24:
Was facing the same issue, also when AP+STA mode, AP is very unstable due to fast reconnects, so did a fix:

```
// reset timer changes back to STA+AP if we can't associate
// #define RESET_TIMEOUT (15000) // 15 seconds
#define RESET_TIMEOUT (5000) // 5 seconds
static ETSTimer resetTimer;

// This routine is ran some time after a connection attempt to an access point. If
// the connect succeeds, this gets the module in STA-only mode. If it fails, it ensures
// that the module is in STA+AP mode so the user has a chance to recover.
static void ICACHE_FLASH_ATTR resetTimerCb(void *arg) {
  int x = wifi_station_get_connect_status();
  int m = wifi_get_opmode() & 0x3;
  DBG("Wifi check: mode=%s status=%d\n", wifiMode[m], x);

  if (m == 2) return; // 2=AP, in AP-only mode we don't do any auto-switching

  if ( x == STATION_GOT_IP ) {
    // if we got an IP we could switch to STA-only...
    if (m != 1) { // 1=STA
#ifdef CHANGE_TO_STA
      // We're happily connected, go to STA mode
      DBG("Wifi got IP. Going into STA mode..\n");
      wifi_set_opmode(1);
      os_timer_arm(&resetTimer, RESET_TIMEOUT, 0); // check one more time after switching to STA-only
#endif
    }
    log_uart(false);
    // no more resetTimer at this point, gotta use physical reset to recover if in trouble
  } else {
    // we don't have an IP address
    if (m != 3) {
      DBG("Wifi connect failed. Going into STA+AP mode..\n");
      wifi_set_opmode(3);
      wifi_softap_set_config(&apconf);
    }
    if (m == 3) {
      DBG("In STA+AP mode. Wifi connect try...\n");
      wifi_station_set_reconnect_policy(true); //attemp to reconnect
      wifi_station_connect();
    }
    log_uart(true);
    DBG("Enabling/continuing uart log\n");
    os_timer_arm(&resetTimer, RESET_TIMEOUT, 0);
  }
}
```

and in `wifiHandleEventCb`

```
case EVENT_STAMODE_DISCONNECTED:
    wifiState = wifiIsDisconnected;
    wifiReason = evt->event_info.disconnected.reason;
    DBG("Wifi disconnected from ssid %s, reason %s (%d)\n",
      evt->event_info.disconnected.ssid, wifiGetReason(), evt->event_info.disconnected.reason);
    statusWifiUpdate(wifiState);
    if(3 == (wifi_get_opmode() & 0x3)) {// AP+STA mode
      DBG("In STA+AP mode. Wifi disconnect...\n");
      wifi_station_disconnect();
    }
    break;
```

So now reconnect is done every 5seconds.