jeelabs/esp-link #34: espfs won't build under windows

brunnels, 2015-08-02:
Build fails when trying to build with mingw under windows.  I replaced the espfs directory with the one [here](https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/examples/esphttpd/libesphttpd/espfs) and got it to build but that of course removes anything you've modified.

In the Makefile around line 286 wget for yui-compressor line needed skip cert check and output file defined:
```
cd yui; wget --no-check-certificate https://github.com/yui/yuicompressor/releases/download/v2.4.8/$(YUI-COMPRESSOR) -O $(YUI-COMPRESSOR)
```

I also needed to make sure the path is set for java bin, python, and git bin.  This will also cause a build error if anything on the path is in "Program Files (x86)".  I added symbolic links in my c:\espressif folder to these locations to resolve this issue.

Here's my resulting espmake.cmd that I call from visual studio.
```
@echo off

REM remove automatic created obj folder
rd obj /S /Q >nul 2>&1

PATH=%PATH%;C:\Espressif\xtensa-lx106-elf\bin;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\espressif\git-bin;C:\espressif\java-bin;C:\Python27
make -f Makefile %1 %2 %3 %4 %5
```

And visual studio NMake properties I have:
```
Preprocessor Definitions: ESP8266;__MINGW32__;__WIN32__
Include Search Path: .\serial;.\user;.\espfs;.\httpd;.\include;c:\tools\sming\Sming\system\include;c:\tools\sming\Sming\Libraries;c:\tools\sming\Sming;c:\tools\mingw64\x86_64-w64-mingw32\include\c++\x86_64-w64-mingw32;c:\tools\mingw64\x86_64-w64-mingw32\include\c++\backward;c:\tools\mingw64\x86_64-w64-mingw32\include\c++;c:\tools\mingw64\x86_64-w64-mingw32\include;c:\tools\mingw64\lib\gcc\x86_64-w64-mingw32\4.8.3\include-fixed;c:\tools\mingw64\lib\gcc\x86_64-w64-mingw32\4.8.3\include;c:\Espressif\sdk\include
```

brunnels, 2015-08-03:
espfs issue was resolved when I added __esp__ pre-processor  definition but now I'm having other issues with serial_uart.c

tve, 2015-08-05:
I have to admit that I don't have the time to explore building esp-link under windows at the moment. If you end up with a PR and instructions I'll be happy to work from that.

brunnels, 2015-08-07:
I have it building under windows now.  I had to copy the contents of eagle_soc.h from the espressif SDK into espmissingincludes.h, replace the espfs with the one from [here](https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/examples/esphttpd/libesphttpd/espfs), and edit the wget parameters for yuicompressor in the Makefile.  

I'm not sure what the issue with eagle_soc.h is.  I wasn't able to include it directly without errors.  I'll send a pull request if you want.  I tested and everything still builds fine under linux too.

I know it was reported in your ESP8266.com thread but wiflash is not working in master.  Says "IROM magic missing" after the upload is complete.  It works fine in the 1.0 tag.  I can create a new issue for that if you want.

My next step is to add https://github.com/tuanpmt/esp_mqtt support.  I'm interested in your simple serial protocol as well for telling the arduino to do stuff based on mqtt messages.

tve, 2015-08-07:
Thanks! A diff or PR would certainly help.

I'm not sure which ESP8266.com thread you mean, or did you mean the gitter chat? Currently the size of the V2 firmware exceeds what fits into a 512KB flash chip, so you have to use a module with a 4MB flash and first serially flash it before you can do OTA upgrades. I want to bring the size back down, but I'd rather get the new features working well first.

brunnels, 2015-08-07:
That would be the issue then.  I modified the makefile to be 512KB when I built V2 branch.  I didn't have any problems flashing it via serial and it runs???

I'll have to order some 4mb flash module to play with.  Is it possible to just solder a new chip in there?

Do you have a gitter channel?  I've been hanging out in the sming channel but I'm getting frustrated with bugs in sming so looking for other avenues to accomplish what I want.

I resolved the eagle_soc.h error with #include <eagle_soc.h> at the top of espmissingincludes.h.

esp_mqtt module has lots of warnings and will only build without -Werrors but it builds.  Haven't tested to see if it works yet.

tve, 2015-08-07:
The gitter channel is linked at the top of the readme https://gitter.im/jeelabs/esp-link
If you have a 512KB flash chip and want to use V2 I would delete the logo gif, that may be enough to squeeze it back in. There's also the html changes that kriegste made that reduce space: http://www.esp8266.com/viewtopic.php?f=34&t=3445&start=105

I'm torn between just going with larger flash and dumping the 8-10 unused esp-03 modules I still have vs. making the effort to keep the firmware size down. The new modules have much better RF (10dB difference, which is huge) and they all have larger firmware sizes. Most likely the next SDK will add more KB also.

tve, 2015-09-16:
I believe you fixed this, thanks! :-)

closed 2015-09-16 by tve