jeelabs/esp-link #38: ESP-12E data not passed to clients

pkastelik, 2015-08-28:
Data sent by attached uC (uC Tx ->ESP Rx) is not passed to clients (port 23) (even if can be seen in ESP uC console). Checked using putty, telnet, Socket Test, HW Virtual Serial Port. 
Checked on releases  V1, V1.0.1 and V2beta.
Tested on ESP -12E  - can be HW specific.

tve, 2015-08-29:
If you see the text in the uC Console web page, then it's not a hardware issue.
I normally use 'nc' on a linux box, but putty works fine for me too. I'm trying to picture what could be causing your issue and the only thing I can come up with is a DNS issue. Perhaps you're using a hostname and are connecting to the wrong box? The esp-link home page has the IP address, how does it look if you use that? And to make sure: you have some simple sketch that prints something every couple of seconds and you're not seeing this come through? The uC web page shows the last 1KB of history but when you connect to port 23 you only get new stuff, you don't get a replay of the history.
.

pkastelik, 2015-08-29:
Hi Thorsten.
I'm using Windows environment, so can't check 'nc' (I'll try on RPI). I successfully connected using IP address on 23 port from any of mentioned tools (putty, telnet, Socket Test, HW Virtual Serial Port).
Data from putty(and other tools) is passed correctly to uC, but it's problem with returning data. As I mentioned earlier data from uC can be seen on web uC console but is not passed to putty(and other) client.
All windows firewalls are disabled. I also tried directly connection of ESP to FTDI (usb2serial). This loop works in the same way - no data is passed to IP client. One thing I discovered is if I send (to ESP RX pin) "~" before some character (ex "~aaaaa") Putty start receiving "~a~a~a~a~a". 
Exact the same behaviour is for ESP-12E and ESP-01.
Please correct me if I'm wrong - it should work as fully transparent bridge?
P.S. Great stuff at all.
Best Regards,
Piotr

tve, 2015-08-29:
Yes, it works as a fully transparent bridge. I'm a bit stumped how there could be a difference between what you see in the uC Console page and the port 23 connections. The code in v1.x is very straightforward for that. v2 does interpret a ~ as a start of SLIP message. So if you have a small sketch that prints "Hello world" every second you see that appear in the uC Console and not on port 23?

Did you compile the firmware or did you just download the ready-made one?

BTW, there's a gitter chat channel that may be more convenient for interactive help: https://gitter.im/jeelabs/esp-link?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge I'm about to head to bed, though (past midnight here).

pkastelik, 2015-08-29:
Yes, I also tried a sketch that prints something every second and  other that replies for any  value that uc receive - without a success - uc console shows data OK but clients not. I downloaded pre-compiled binaries and uploaded into ESP using flash_download_tool_v1.2. 
Sorry for so late interrupting you - we have 2pm in Poland - sorry.
Let's back to this topic on Monday - have a nice weekend.
Piotr

tve, 2015-08-29:
Oh, don't worry about the time, I only mentioned it because I wasn't going to be able to respond quickly. I do recommend you join the gitter chat for quicker responses, though.

I think what we need to do is start with a known firmware, for example, V1.0.1 (please download it fresh, I noticed a slight mix-up). Verify in the UI home page that you indeed are running that firmware (the yellow title should show `esp-link v1.0.1 - 2015-08-28 21:25:00 - development` for the 4MB image and the same except `b48614` at the end).

Some thing you could do to help troubleshoot:
- copy&paste the uC Console output here
- copy&paste the debug log page content here
- if you're familiar with wireshark, run that on port 23 and grab a full connection where you telnet/putty to port 23, wait a while, and disconnect

If none of these things shed any light, I can prep a debug version for you. It would help if you said something about what you want to do so I can decide whether V1 or V2 would be best. If you just want remote reflashing and transparent bridge, then v1 is simpler...

pkastelik, 2015-08-31:
Hi Thorsten.
Something is wrong with bin from link:
https://github.com/jeelabs/esp-link/releases/download/v1.0.1/esp-link-4MB.tgz
It's V1.01 but esb web tells me it's 2.0 beta1 (2015-08-01 23:49:53 5fc65bd)
Could you verify this file or send me zipped fw V1.01 for 4M flash?

Regards,
Piotr

tve, 2015-09-01:
please do a fresh download:
```
root@oc1:~/esp-link-v2.0.beta2# cd /tmp
root@oc1:/tmp# ls
root@oc1:/tmp# curl -L https://github.com/jeelabs/esp-link/releases/download/v1.0.1/esp-link-4MB.tgz | tar zxf -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   396    0   396    0     0    608      0 --:--:-- --:--:-- --:--:--   613
100  350k  100  350k    0     0   115k      0  0:00:03  0:00:03 --:--:--  257k
root@oc1:/tmp# ls
esp-link
root@oc1:/tmp# cd esp-link/
root@oc1:/tmp/esp-link# ls
blank.bin  boot_v1.4(b1).bin  user1.bin  user2.bin  wiflash
root@oc1:/tmp/esp-link# egrep -a 'esp-link v' user1.bin | head -1
esp-link v1.0.1 - 2015-08-28 21:25:00 - developmentnormalwdt resetexceptionsoft wdtrestartdeep sleepexternal//home.html/menu/flash/next/flash/upload/flash/reboot/log/text/log/dbg/console/reset/console/baud/console/text/wifi/wifi/wifi.html/wifi//wifi/info/wifi/scan/wifi/connect/wifi/connstatus/wifi/setmode/wifi/special/pins*°þ?¸þ?Äþ?Ðþ?Üþ?äþ?ðþ?rate{"rate": %ld}start{"len":%d, "start":%d, "text": "\u%04xsendtxbuffer: espconn_sent error %d on conn %p
root@oc1:/tmp/esp-link#
```
I did have the wrong version uploaded at one point, but I fixed it.

pkastelik, 2015-09-01:
Hi Thorsten.
I'll check it tis evening.
I also flashed V2 beta2  and it seems that telnet works ok.
Only issue I see  in V2 beta2 is that ESP disconnect (previously connected AP) and back  to AP mode after some time.

solidboredom, 2015-09-06:
Hello i can confirm the no echo issue for beta1. (does it hve to do with swapped UARTs?)
 i have esp-link v2.0.beta1 running on on esp-01 (1 mb) connected to Ardiono pro mini and have the same issue
(it actually took me two days of prying and trying to figure out it was not an issue with my HW setup) .
i have put a "serial  echo" sketch ontothe arduino, and the outputs are shown in the uc Console but are not transmitted to the client. when ota proramming avdude says "cannot sync".

Another thing i had noticed: if i turn the UART debug log ON then the log text gets transmitted to ARDUINO. i can see in the uc Console that my arduino "echo-code" echoes debug messages and tries to send them back . so  they are shown in the console but are not sent to the client. 
turning off the UART logging stops this again.
telneting the esp:23  transmits my inputs via ESP-Link to arduino but responses are only shown in the mc console and do not reach the telnet client (UBUNTU )  router:linksys wrt54g with original firmware
peter

tve, 2015-09-06:
>  if i turn the UART debug log ON then the log text gets transmitted to ARDUINO
This is expected, "UART debug" means to print the esp console output to the UART. With all the SLIP line protocol stuff that's no longer a good idea and I've been toying with using UART1 again.

> i can confirm the "no echo"-issue for beta1
Sounds like I need to try and repro this somehow. If you have the time, does it still happen with beta2?

solidboredom, 2015-09-06:
i tested 2.0beta2 but i can confirm the "no-data sent to client" has disappeared.
 ok thanks for clarifying the uart thing. after finding out it was a 2.0beta1 issue i also had put 1.0.1 onto esp and it worked fine too.  interestingly playing with the "echo"-sketch it seems that the way esp-link handles data transfers to ESP. esp-link 1.0.1 sent the characters i enter into telnet almost instanteniously where 2.0 is more line oriented, and the data is not sent unil i hit ENTER key, i wonder if this is true. it also seems that the length of one line is limited to about 50-50 characters and get truncated after that, though i am not shure if it is not an issue  arduino-string-library or the code i use: here the echo sketch code:
int led = 13;//2;

void setup() {
  Serial.begin(57600);
  pinMode(led, OUTPUT);
    digitalWrite(led, LOW);
}
#define ledOncyc (long)6500

long ledcyc=ledOncyc;

String content = "";

void loop()
{
//this part is just a general  led blinking
if(ledcyc==ledOncyc){  digitalWrite(led, HIGH); }
if(ledcyc==ledOncyc*2){  digitalWrite(led, LOW);}
if(ledcyc==ledOncyc*3){  digitalWrite(led, HIGH);}
if(ledcyc==ledOncyc*4){  digitalWrite(led, LOW);}
if(ledcyc==ledOncyc*9){  ledcyc=0;}
ledcyc++;

//here is the serialecho part.
  char character;

  while(Serial.available())
  {
    character = Serial.read();
    content.concat(character);
  }

  if (content != "")
  {
    //blink quickly for about a second(depending on clock)
    // and wait for more characters to arrive to add
    //them to the string and output them all together
       for(int i=0;i!=12; i++)
    {
      digitalWrite(led, LOW);   
      delay(40);
      digitalWrite(led, HIGH);  
      delay(40);
    }
    //if new chracters arrived while i was blinking restart
    //current function  to add them to the string
    if (Serial.available()) return;
    
    Serial.println("(echo:("+content+ "):echo)");
    content="";
  }
}

tve, 2015-09-06:
that's a bug :-(

tve, 2015-09-06:
Take that back, it works, but unix command line programs, such as telnet or nc used cooked input by default, i.e. line-oriented. If you set your tty into raw mode `stty raw` then what you type is sent immediately and shows up immediately too.

solidboredom, 2015-09-06:
strangely i beleive the very same telnet behaved diffferently with 1.0.1  but i would not bet on that and put too much weight on it, until somebody can confirm it (say, using the echo sketch) .  
by the way: the blog formating had broken the " echo code": the if(ledcyc==ledOncyc2)
should be if(ledcyc==ledOncyc TIMES  2 ) but the blog ate the asterisk *

solidboredom, 2015-09-06:
anyway thanks for your quick reaction and actions!  it is indeed fun to report probissues when the auhtor reacts to them like you do , so you should be careful with that... :)

tve, 2015-09-16:
OK, I can reproduce the "UART input not sent to TCP port 23 connections" problem. It is fixed in beta2. Apologies for the time this wasted on your end.

tve, 2015-09-16:
Closing this ticket, please reopen if you still encounter such issues with recent builds.

closed 2015-09-16 by tve