jeelabs/esp-link #42: Tiny typo in release notes: --baud 460880 should be --baud 460800

elmicha, 2015-08-30:
Thank you for this great software! Flashing the ESP and the µC over the air is so cool!

There is a tiny typo in the release notes which might throw some people off:

```
% esptool.py --port /dev/ttyUSB0 --baud 460880 write_flash -fs 32m -ff 80m 0x00000 boot_v1.4\(b1\).bin 0x1000 user1.bin
Traceback (most recent call last):
  File "/usr/local/src/ESP8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 481, in <module>
    esp = ESPROM(args.port, args.baud)
  File "/usr/local/src/ESP8266/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 67, in __init__
    self._port = serial.Serial(port, baud)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 282, in open
    self._reconfigurePort()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 417, in _reconfigurePort
    set_special_baudrate(self, custom_baud)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 47, in set_special_baudrate
    FCNTL.ioctl(port.fd, TERMIOS.TIOCGSERIAL, buf)
IOError: [Errno 25] Inappropriate ioctl for device
```

With `--baud 460800` it works. This happens only with a CP2104 serial adapter; the FTDI board happily accepts 460880.

tve, 2015-08-30:
Thank you! fixed.
I'm glad it's working for you. Are you using V1.0.1 or trying the new V2 stuff?

closed 2015-08-30 by tve

elmicha, 2015-08-30:
I'm using V2, although I'm not using the TCP client or REST features yet. I just started using esp-link. My Arduino Pro Mini @ 8 MHz apparently only works up to 57600 bps, even on the serial adapter. I could only get it to talk to the ESP (Adafruit Huzzah) with swap_uart, but then the reset pin didn't work (I tried the Rx and Tx pins). So I added a new entry to map_names and map_asn in cgipins.c, so that GPIO12 is used for reset. Now it works like a charm.

tve, 2015-08-30:
interesting. glad it works!