jeelabs/esp-link #43: High powerconsumption after power-on

bc547, 2015-09-05:
I notice a rather high power consumption when I powerup the esp8266 (ESP12-Q, esp-link v2.0.beta2 precompiled binaries). As can be seen in the screenshot below, it draws a current of 72mA when doing nothing and about 115mA when listening.

[image]

However, when I surf to the built-in esp8266 webpage and change the pin assignment from jn-esp-v2 to wifi-link-12 and back to jn-esp-v2, the module suddenly draws approx 50 mA less (24mA idle, 78mA listening). The mentioned behaviour is repeatable. See screenshot below.

[image]

Everything still seems to work perfectly (http server, serial,....). It seems like some extra powersaving feature of the esp8266 is turned on when I switch from wifi-link-12 pins to jn-esp-v2. These results are more in line with the scope screenshot at http://jeelabs.org/book/1526f/. 

PS. the current shown is the total current drawn by an arduino-pro-mini (3.3V,8Mhz) with a very simple serial echo-repeater example loaded and the ESP12-Q module with esp-link. I verified the power consumption of the arduino does not change at all (7mA avg, 11mA max) during the experiments.

Btw. You are doing a great job! Esp-link is exactly what I was looking for :-)

tve, 2015-09-05:
I have to admit that I haven't looked into power consumption in a while. It's coming up though...

Let me make sure I understand:
- you're using V2
- you have it powered-up for more than 30 seconds before you start to make sure it's in STA-only mode (or you confirm that on the wifi page)
- it's using a baseline of ~70mA as you browse around
- swapping the pins and back puts the functionality into the same state, but lower power consumption
- is it specifically the wifi-link-12 pin config that "fixes" the power consumption, or any pin change?

The thing that is special about the wifi-link-12 pin assignment is the swapping of the UART pins. The code is here https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L342-L352 Overall, it's unlikely that a difference of 50mA can come from current through the pins. I'm pretty sure it can't be the CPU itself. Must be the radio. I have not measured how it looks while in STA+AP mode, so make sure it's not in that mode.

bc547, 2015-09-05:
* Yes. I'm using the precompiled binaries of esp-link v2.0.beta2
* It's powered up for at least a minute. The module is in STA-only mode (see screenshot below)

[image]

* Yes.. after booting, baseline is approx 70mA
* It is not the wifi-link-12 pin that 'fixes' the power consumption. The power consumption lowers only when switching away from it. And it rises back to 70mA if switching to it.
  * Scenario 1
    *  Boot with 'jn-esp-v2' and switch from 'jn-esp-v2' to 'wifi-link-12'   => 70mA before and after switch
    *  Switch back to 'jn-esp-v2'  => 70mA before and 24mA after the switch
  * Scenario 2
    *  Boot with 'wifi-link-12' and switch to 'jn-esp-v2'   => 70mA before and 24mA after the switch
    *  Switch back to 'wifi-link-12' => 24mA before, back to 70mA after the switch
    *  Switch back to 'jn-esp-v2'  => 70mA before and 24mA after the switch
  * Scenario 3
    *  Switching from any pin with 'swap_uart:no' to any other pin with also 'swap_uart:no' => no change in current. It stays at either 70mA or 24mA.    

So it would seem the uart swap code does at least something related to power consumption. And maybe the power consumption of 70mA at boot is because initialisation code at boot time does not execute the same code as when switching uart back to normal?

I'll do the same test tomorrow with an ESP-7 module.

bc547, 2015-09-06:
I found the cause.

GPIO15 (together with GPIO0 and GPIO2) is sampled at boottime to determine the bootmode of the device. In order to boot normally, it must be pulled low. Based on comments I found on google, I wired GPIO15 to ground... and everything worked. 

However, it appears that after boot, GPIO15 is used as an output and it is set to a logical 1. The excessive current is caused by a short circuit between GPIO15 and GND. I have to admit, I'm really surprised the module survived that for longer periods. When switching from 'wifi-link-12' to any other pin mode, GPIO15 is either set low or set as input and the short circuit/excessive current is gone.

Now I placed a 1K pulldown resistor between GPIO15 and GND and the baseline consumption is now approx 25mA at boot. Also selecting 'wifi-link-12' now does not change the current to 70mA.

Since GPIO15 is set as high output at boot, there is a needless power consumption across the pulldown resistor. If, after booting, I switch from 'jn-esp-v2' to 'wifi-link-12' and back, the baseline current drops approx 3mA.

Would it be possible to set GPIO15 as input, no pullup when booting in a mode where the 2nd uart is not used?

tve, 2015-09-06:
The UART swap turns gpio15 into UART 0 RX or TX (I forget which), so it definitely changes it (this stuff is in the SDK, so I don't have the details). esp-link doesn't initialize gpio15, maybe the SDK does...

tve, 2016-12-11:
This is pretty old and I have not seen this again. Please reopen if this is an issue again.

closed 2016-12-11 by tve