jeelabs/esp-link #44: scanning for SSIDs can take tens of minutes and cause users to think something is broken in the flashed esp-link firmware

solidboredom, 2015-09-06:
circumstances:
right after flashing esp-link (either 1.0.1 and 2.0 versions) onto the new  module  and connecting to 192.168.4.1 (stangely the modules appear as AI-THINKER_XXXX in my case)  and going to WIFI setup page.
what:
 scanning for network ssids the scan never ends so there is no way to change the SSID.(it actually will end after 15 or 20 minutes but there is no way to know it, and it looks like something has went wrong)
i wonder if it has to do with how many networks are availiable  (there are 30+ of them  around where i am, if you include the very faint ones.

tve, 2015-09-06:
I'm out in the boonies with only the few networks I operate visible... Are you able to switch UART debug ON and then capture the uart output from boot time? That would tell us why it's not finishing the scan. It uses the SDK built-in scan, but maybe there's a bug in processing what that reports. I can take a module to the office and see whether there are enough networks there to trigger the problem.

solidboredom, 2015-09-06:
sorry for that but i just had unsoldered the bulky pin header to put the ESP into the circuit, so the rest of the stuff is done OTA only. i have never tried over the wire debugging yet anyway. but if i fry the module and will have to flash anotherone i am willing to try. is there any way of saving the scan output to a logfile? it also seems to me it may be  only the very first scan which takes so very long.

DangeRuss1313, 2015-10-26:
I'm having the same issue. I just grabbed the serial output, and it looks like a fatal exception is occurring on scan causing the module to reboot. I'm running an ESP-01 with the latest released firmware (1.0.4). See the output below:

273> ** esp-link v1.0.4 - 2015-09-15 21:03:54 - 2c29e29
   273> Flash config restore ok
   273> CONN led=2
   275> SER led=-1
   277> Wifi init, mode=AP+STA
   279> Wifi uses DHCP, hostname=esp-link
   283> "ip": "0.0.0.0"
   285> "netmask": "0.0.0.0"
   288> "gateway": "0.0.0.0"
   291> "hostname": "esp-link"
   293> espFsInit ok
   295> Httpd init, conn=0x3fff1148
   299> Serbridge pins: reset=0 isp=-1 swap=0
   311> Reset cause: 6=external
   311> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   312> Flash map 0, chip 001340C8
   314> ** esp-link ready
   317> mode : sta(18:fe:34:fe:96:5a) + softAP(1a:fe:34:fe:96:5a)
   323> add if0
   324> add if1
   326> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   334> bcn 100
 15293> Wifi check: mode=AP+STA status=0
 15294> Enabling/continuing uart log
 22150> add 1
 22150> aid 1
 22150> station: 6c:88:14:8e:ef:cc join, AID = 1
 22150> Wifi AP: station 6c:88:14:8e:ef:cc joined, AID = 1
 27676> 192.168.4.2:55842 GET /wifi/scan (1 conn open)
 28489> 192.168.4.2:55842 Closed, 816ms, heap=29896
 29571> 192.168.4.2:55844 GET / (5 conn open)
 29574> 192.168.4.2:55844 Closed, 1896ms, heap=30344
 29664> 192.168.4.2:55887 GET /pins (4 conn open)
 29667> 192.168.4.2:55888 GET /wifi/info (4 conn open)
 29796> 192.168.4.2:55889 GET /wifi/scan (4 conn open)
 29881> 192.168.4.2:55887 Closed, 660ms, heap=27328
 29887> 192.168.4.2:55888 Closed, 657ms, heap=29408
 29999> 192.168.4.2:55889 Closed, 763ms, heap=31488
 30294> Wifi check: mode=AP+STA status=0
 30294> Enabling/continuing uart log
 31796> 192.168.4.2:55890 GET /wifi/scan (1 conn open)
 32002> 192.168.4.2:55890 Closed, 2758ms, heap=31896
 33799> 192.168.4.2:55891 GET /wifi/scan (1 conn open)
 33997> 192.168.4.2:55891 Closed, 1730ms, heap=31896
 35910> 192.168.4.2:55894 GET /wifi/scan (1 conn open)
 35996> 192.168.4.2:55895 GET /wifi/info (2 conn open)
 36116> 192.168.4.2:55894 Closed, 206ms, heap=29784
 36197> 192.168.4.2:55895 Closed, 204ms, heap=31872
 36510> 192.168.4.2:55896 POST /wifi/scan (1 conn open)
 36709> 192.168.4.2:55896 Closed, 201ms, heap=31896
 37510> Starting a scan
 37510> f 0, 192.168.4.2:55897 GET /wifi/scan (1 conn open)
 37950> 192.168.4.2:55898 GET /wifi/scan (2 conn open)
 38107> 192.168.4.2:55897 Closed, 303ms, heap=28352
 38400> 192.168.4.2:55898 Closed, 455ms, heap=29192
 39648> scandone
 39648> Scan done: found 57 APs
 39648> bss1:  (0)
 39648> bss2: XXXXXX (-53)
 39649> bss3: XXXXXX (-56)
 39653> bss4: XXXXXX (-71)
 39657> bss5: XXXXXX (-60)
 39661> bss6: XXXXXX (-75)
 39663> bss7: XXXXXX (-80)
 39666> bss8: XXXXXX (-76)
 39669> bss9: TDC (-81)
 39671> bss10: XXXXXX (-78)
 39674> bss11: XXXXXX (-80)
 39677> bss12: XXXXXX (-53)
 39680> bss13: XXXXXX (-53)
 39683> bss14: XXXXXX (-55)
 39686> bss15: XXXXXX (-53)
 39689> bss16: XXXXXX (-53)
 39692> bss17: XXXXXX (-86)
 39695> bss18: Davis_Guest (-82)
 39698> bss19: XXXXXX (-80)
 39701> bss20: TP-LINK_61B856 (-43)
 39705> bss21: XXXXXX (-84)
 39708> bss22: XXXXXX (-68)
 39711> bss23: XXXXXX (-66)
 39714> bss24: XXXXXX (-66)
 39716> bss25: XXXXXX (-66)
 39720> bss26: XXXXXX (-68)
 39722> bss27: TP-LINK_316605 (-55)
 39726> bss28: XXXXXX (-69)
 39729> bss29: ELPASO_TPLINK_C04A00BD0769 (-53)
 39733> bss30: XXXXXX (-67)
 39736> bss31: XXXXXX (-83)
 39739> bss32: TMR (-87)
 39741> bss33: XXXXXX (-59)
 39744> bss34: XXXXXX (-85)
 39747> bss35: XXXXXX (-52)
 39751> bss36: XXXXXX (-58)
 39754> bss37: XXXXXX (-81)
 39757> bss38: XXXXXX (-79)
 39761> bss39: XXXXXX (-64)
 39763> bss40: XXXXXX (-82)
 39766> bss41: KWZ24 (-78)
 39769> bss42: Linksys04666 (-42)
 39772> bss43: XXXXXX (-75)
 39775> bss44: tplink-test (-48)
 39778> bss45: ObiwanKenobi (-52)
 39781> bss46: FL1259 (-80)
 39783> bss47: XXXXXX (-69)
 39786> bss48: XXXXXX (-83)
 39789> bss49: XXXXXX (-83)
 39792> bss50: XXXXXX (-69)
 39795> bss51: XXXXXX (-81)
 39798> bss52: XXXXXX (-81)
 39801> bss53: XXXXXX (-81)
 39804> bss54: XXXXXX (-80)
 39807> bss55: XXXXXX (-70)
 39810> bss56: XXXXXX (-50)
 39813> bss57: XXXXXX (-85)
 40370> 192.168.4.2:55899 GET /wifi/scan (2 conn open)
 40372> 192.168.4.2:55899 ERROR! httpdSend full (170 of 2600)
 40372> Fatal exception (0):
 40374> epc1=0x2c343636, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

tve, 2015-10-26:
Thanks for the details, I see where the bug is, the response json doesn't fit into a network buffer and it crashes at that point. At least now I know how to repro.

solidboredom, 2015-10-26:
i am grad DangeRuss had provided the information,at last. 
But, by the way wait, is this just the the output from the Debug window of the esp-link GUI? if so  there must  have been a misunderstanding, i had assumed you need to have over-the-wire debugging  to get the debug output from the boot time you were asking for.

DangeRuss1313, 2015-10-26:
The debug I posted is from Putty. I had an FTDI cable connected to my ESP to read the debug output. I haven't tried debugging OTA yet.

BBBits, 2015-11-13:
Hi, 
Not sure If I'm in the right place   . this post is in relation to  esp-link-2.1.beta2
I was having a few issues with the wifi scanning not finding anything when there was only one AP in the area..
It turned out to be a little code bug in wifi.js  scanResult function.
I changed 
if (data.result.inProgress == "0" && data.result.APs.length > 1) { 
To
if (data.result.inProgress == "0" && data.result.APs.length > 0) { 
& the problem floated away .
==============================================
I was Also having problems while in AP mode with the radio going offline .
A Lib replacement has solved that issue .
Here is the bug report thread
http://bbs.espressif.com/viewtopic.php?f=7&t=1257&start=10
& this is the download link for   lib_1.4.1_pre7.zip
http://bbs.espressif.com/download/file.php?id=945
==========================================
Many Thanks for all your hard work .

tve, 2015-11-13:
Nice finds, thanks for reporting! I'll fix these in the next beta.

tve, 2016-12-11:
I don't think there anything further I can fix here. Please reopen if there is.

closed 2016-12-11 by tve