jeelabs/esp-link #45: Some remarks/feature requests

bc547, 2015-09-08:
These features I miss already after 2 days of playing with esp-link:

1. Ability to change hostname and have a custom (short, e.g. 128 char) description. The hostname and description should be clearly visible on the homepage (and preferably also all subpages) to avoid user mistakes when managing multiple devices at the same time. At the moment you have to look at the browser address bar and the host ip to distinguish different devices.

2. Ability to disable remote programming. It's very easy to make a mistake if you have multiple simultaneous projects and make an ip address typo or copy Makefiles between projects. :-)

3. Serial programming does not always work with arduino optiboot loader compiled for 250000 baud programming speed. Sometimes it only works after the 4 or 5th programming attempt. My guess would be some timing issue, but I'm not yet familiar with the exact programming protocol. So i'll investigate/test this further when I have some spare time.

These features are not really important for me, but I think they would increase user-friendliness

4. Serial input field on the uC console window. It is easier on e.g. a tablet/mobile phone to open a webpage and query a device than going through telnet.

5. Have the ability to use custom baud rates on the webpage. Since the API already allows this (e.g. http://192.168.0.76/console/baud?rate=250000), it seems this is only a cosmetical issue (textbox in addition to the preset-buttons)?

6. Have the ability to configure and use a different baud rate for programming only. On arduino, different bootloaders expect a different baud rate for programming depending on version, clock speed... (often 38.4k, 57.6k up to 115.2k). But sometimes an application wants to use a different baud rate to minimize transmission errors or achieve higher throughput. Many users lack the knowledge/hardware to compile and upload a custom bootloader where the programming baud rate equals the application baud rate. At this moment, the user must go to the uC console page, select the correct programming speed, program the device, select the correct application speed, watch the console logs... (repeat this for each programming iteration)

I'm really looking forward to the already announced features (outbound TCP requests, MQTT, REST...) :-)

tve, 2015-09-08:
I really appreciate bug reports and features requests, and even more so pull requests :-). Please be aware that esp-link is not a product you're paying money for, it's something I've written in my spare time that I use myself and that I wanted to share with others in the hope to get contributions. The source code is freely available and I tried to write quite some docs and put many comments into the code to make it as easy to adapt to your requirements as I can. At this point polishing it is really taking time away from using it :-).

That being said, here are some responses to your feedback:

> Ability to change hostname and have a custom (short, e.g. 128 char) description.

You can change the hostname used for DHCP on the wifi page. It's a good idea to display the name on every page and I was also thinking of adding a description so I can remember where I deployed a particular esp-link module.

> Ability to disable remote programming.

Makes sense. Sounds like a simple change, care to submit a pull request?

> Serial programming does not always work with arduino optiboot loader compiled for 250000 baud programming speed.

Yes, and I'm not surprised. I use 16Mhz AVRs at 115200 baud and I often have to make 2 attempts. Some boards have AC coupling of the reset pin through a capacitor, which doesn't exactly help. Also, the esp8266 isn't always fast in ACKing packets and if it doesn't have a solid connection it can easily drop out of the timing. One thing you could try is to increase the timeout in optiboot. I believe the std is 500ms and increasing that to ~2s could make a huge difference. At least it would isolate whether the problem is one of round-trip timing or reset timing. Yet another problem I've seen is if the sketch is outputting at the time of reset it's easy for a few characters to sit in the esp's UART RX FIFO and mess-up the programmer sync.

> Serial input field on the uC console window.

Yup. I don't have any sketches that take input, so it's not high on my personal feature list. PRs welcome.

> Have the ability to use custom baud rates on the webpage.

Yup. Ditto...

> Have the ability to configure and use a different baud rate for programming only.

Yup. Ditto... You're not the first person who asked for this.

Again, thanks for the feedback!

bc547, 2015-09-08:
'Requests' was maybe badly worded. It was meant more like "some ideas you might find interesting and use yourself" :-)
I really appreciate the effort, time and knowledge you devote to this project.

For now, I'll look into disabling remote programming as an option and will submit a pull request once done.
I'll also investigate/experiment with the remote programming sync errors some more.

tve, 2015-09-08:
Great, thanks! I assume you want the enable-programming switch to be saved in the config, in which case you need to add it to the struct in config.h, provide a default in config.c, add a checkbox to the web page console.html (if that's where you want to put it), and add the javascript functionality to respond to toggles in console.js (there are some similar functions for the pin and new mqtt checkboxes at the end of ui.js), and update the config in console.c, and finally disable the programming stuff in serbridge.c. Oh man, I wish it didn't take so many steps and I should templatize this stuff. V3...

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve