jeelabs/esp-link #46: Can't send more than ~2140 characters at 115200

poleguy, 2015-09-11:
I am trying to use your esp-link in transparent mode to replace a serial port. The port presents a prompt and I send a dump command. It seems to work okay until the dump returns about 2140 characters. At this point I lose the final characters of the dump. Is this expected?

I do see the final characters and prompt correctly on the microcontroller console.

I suspect this is a flow control issue. I don't see any debug messages. I'm using V1.0.1 on a ESP-01

I read your blog, but I don't see any reason why a large packet coming from the UART side would cause flow control issues. Any idea?

If this is hopeless I'd like to know, otherwise I might have a look at the code and try to fix it myself.

tve, 2015-09-11:
Mhhh, this is definitely supposed to work. What baud rate are you using and do I understand correctly that the attached uC sends the 2140 chars at full line speed? I suspect that there may be some buffer overflow issue, make sure you have a good wifi connection (the -01 modules don't have a great RF section).
The buffer management for this stuff is some of the oldest code and due for a remake using more dynamic allocation and some of the newer learnings about the SDK... However, if the buffer overflows you should see "espbuffsend: txbuffer full on conn %p" in the debug console. If you see that you could increase it at https://github.com/jeelabs/esp-link/blob/v1.0.1/serial/serbridge.h#L12 but now that I look at the code, I see a bug, which is that I "free" the buffer when I send it (https://github.com/jeelabs/esp-link/blob/v1.0.1/serial/serbridge.c#L48) and I really need to keep the content intact until I get the sent callback (https://github.com/jeelabs/esp-link/blob/v1.0.1/serial/serbridge.c#L77-L84).
Let me know if you'd like a test build with that fixed and the level of urgency...

poleguy, 2015-09-11:
Wow, that's a quick response! I'm actually talking to an embedded state machine in an FPGA, and I can have it send as much data as I ask for. It will send it at full line speed because it is not doing anything else and there is no buffering required. My test case to reproduce this is to send about 2160 characters. I only see the first 2140-2146 or so. This is 100% repeatable.

I did not see any txbuffer full on the debug console.

I'd love a test build, as I don't have a build environment set up for this at the moment. I was hoping just to use it without modification. Level of urgency... well, I suppose it's not that urgent, as nobody is  waiting on me. (I'm working on my electric go-kart hobby.) However I'm actively trying to get this going, so now is a good time for me to test and verify things.

tve, 2015-09-11:
I looked at the code and it's not a trivial change, i.e., not something I can do in 10 minutes and make available. I'll try to get to it this evening (west coast time).
The underlying error is that I reset the buffer to empty after calling "send" but in fact, if the buffer can't be transmitted in one packet (e.g. 1420 bytes) then the SDK expects the buffer to remain untouched until it can send the remainder. (That's all part of the non-documented 'fun' stuff...)

poleguy, 2015-09-11:
Well I'm excited to try it when you come up with a fix. I'm quite impressed
with your esp-link. I tried out the beckdac solution (
https://github.com/beckdac/ESP8266-transparent-bridge) first because google
lists it, but I had trouble with the UART-to-TCP link not working at all
when the UART was hooked up at boot. I think it wasn't happy to see
characters coming at startup. Disconnecting the UART line at every boot is
not really a good option. :-)

I would have reported that bug, but then I saw your post on how you handled
the buffering and thought that at least you thought it through, so maybe
yours might work better.

And it did just work (well mostly)! And the web interface is convenient.

tve, 2015-09-12:
Can you please give https://github.com/jeelabs/esp-link/releases/tag/v1.0.3 a try? I did some testing trying to repro your problem and it works, but who knows whether I really got the right conditions reproduced...

poleguy, 2015-09-13:
Well, good and bad here. 1.0.3 successfully fixes my original issue. 

I have to adjust my original statement though. There is a bug in the version of realterm I had installed that was double counting my characters, so my limit was ~1070 not ~2140.

However, it now can't send more than 2128 characters. I can successfully send 2122 but I can't send 2132 (My dump sends 10 character increments.) If I try to send 2132 characters I only see the first 2128 every time. Again no issue on the console.html page. 

This time I do see a buffer full message each time:
espbuffsend: txbuffer full on conn 0x3fff31b4
on the log.html page after hitting refresh. (It may have been there before, as I hadn't realized I needed to hit refresh.)

I'm not sure how to set my expectations. To me it seems the UART to TCP side should be able to stream pretty much endless data at the full 115200 rate without any issues on a clean wifi link. Or is this a very naive presumption?

I'm again happy to provide debug support if you're willing to keep looking at this. Since I'm running with an FPGA I send data at the UART rate without any gaps at all. I pull it up on the scope and it's solid data with none of the occasional dead time typically seen if it were from an MCU. There are no interrupts or other processing holding this back. So this is the worst case to some degree.

tve, 2015-09-13:
Interesting and sadly not entirely unexpected. I've had much worse problems with this stuff. What happens is that the SDK pretty much only supports a single packet in flight (it's difficult to get LwIP to do more). So you're limited to the round-trip time data->ACK at the TCP level. What I changed is to create a fresh 2KB buffer after handing the SDK the previous one. That size is defined at https://github.com/jeelabs/esp-link/blob/v1.0/serial/serbridge.h#L12

So let's see:
- when the RX fifo has 80 characters it triggers an interrupt
- the interrupt simply triggers a task
- the task runs, empties the fifo and calls the serial bridge code
- there it gets accumulated into a 2KB buffer and sent if there isn't a send in progress (that's the way the SDK works)
- when a send happens, the next 2KB buffer is allocated

You say you get 2128 characters and a txbuffer full error. Now 2048+80=2128! So what seems to happen is that the first 80 characters come in, interrupt fires, callback happens, packet gets sent out with those 80 characters, new 2KB buffer gets allocated. Then another 2048 characters arrive, interrupt fires every 80 characters (more or less), and they get accumulated into the 2KB buffer. By the time the buffer is full, the first 80 char send has not completed and there's an overflow.

At 115200 baud 2048 characters take 177 milliseconds. When I was troubleshooting this stuff, I did see round-trip times ranging from 20ms to 600ms. The older SDKs and my access point didn't get along very well (I think there are some issues with WPA2 when using TKIP encryption). The newer esp-12, esp-13, and wrrom-02 modules are also a lot better.

So one option might be to increase the amount of buffering, but of course if you're streaming and the RTT is too long then that won't really help. I assume you're sending to a local server, can you run tcpdump there to see whether what I'm dreaming up actually happens? Something like `sudo tcpdump -n port 23`. Just the first few packets of a connection should show the problem. Another thing you could try is a non-encrypted access point (no, I don't like that). Yet another possible source of trouble is if the receiver of your data delays ACKs ("nagle's algorithm"), so turn that off if you can (the tcpdump would show us that that's the problem).

tve, 2015-09-13:
BTW, you may not have seen the gitter chat https://gitter.im/jeelabs/esp-link that's sometimes a faster way to get some back-and-forth help... Of course time zones still exist...

poleguy, 2015-09-14:
I'm on windows, so I did a wireshark capture. I could send you the whole dump if you'd like... but this is what I see:

No.     Time           Source                Destination           Protocol Length Info
    100 8.925029000    192.168.1.66          192.168.1.90          TELNET   84     Telnet Data ...
    101 8.939469000    192.168.1.90          192.168.1.66          TELNET   134    Telnet Data ...
    102 9.144563000    192.168.1.66          192.168.1.90          TCP      54     64082→23 [ACK] Seq=91 Ack=4325 Win=64160 Len=0
    103 9.147543000    192.168.1.90          192.168.1.66          TELNET   1514   Telnet Data ...
    104 9.148525000    192.168.1.90          192.168.1.66          TELNET   642    Telnet Data ...
    105 9.148583000    192.168.1.66          192.168.1.90          TCP      54     64082→23 [ACK] Seq=91 Ack=6373 Win=64240 Len=0

It's not quite the scenario you described, but it does seem that there is a delay on the acks. This entire transaction is only 224 msec long, so it doesn't seem to be a delay issue.

I set TCPNoDelay True and TcpAckFrequency to 1 in the windows registry and restarted my network connection. Now I can send dumps longer than 2128. Good call.

Just trying things out, I've tried sending dumps of 10282 and it works maybe half of the time. I haven't looked into what causes it to fail the rest of the time, but it may be due to dropped packets, etc. I presume that lost packets are not going to be noticed in time to resend them and not lose data.

But the dump then looks like a huge series of short packets alternating in length of 134 and 58 bytes... yuck... I see what these settings are trying to fix!

I played with the two options, and it seems like TcpAckFrequency set to 1 alone is enough to mostly fix the issue. 

It seems like I need to give up on the hope of using this to reliably send "long" (2k-256k byte) dumps reliably.

I also don't like relying on changing the OS setting for TCNoDelay and TcpAckFrequency in windows. That makes it hard to move an app to a different computer. I suppose I can figure out how to set this in python, but even that feels dirty. 

It makes me think of other options: can I throttle my speed in my UART to achieve reliable operation? If retries are an issue, I figure that if I send data at 115200 with periodic gaps in it to allow retries it could become more reliable.

I wonder if there is an optimal data dump length that I can use that will prove useful. I'll have to think about this. I suppose keeping it at 2122 characters is my most robust option for now.

I see the gitter chat now, but I'm not sure I want to move this conversation there at this point. If I need back and forth help I'll go there. Or if you think we should try that for some detailed debug work, I can do that too.

tve, 2015-09-15:
Thanks for the update. I've been pondering about what to do. Couple of options:

- I need to change the buffer sizes to be multiples of ~1420 so we don't end up with smaller packets than necessary. For example, in your short trace there are two consecutive packets with 1514 and 642 bytes, that's basically the 2048 byte buffer (plus IP&TCP headers) cut into a full and a partial packet.

- You could use RTS/CTS style hardware flow-control, you'd have a line going back to your FPGA telling it when the RX UART buffer is full. This would require quite some enhancement to esp-link, though.

- Perhaps I should bite the bullet and just support a queue of buffers, I already have that for MQTT, so it's not a lot of work. What I'm wondering is how to get the SDK to actually stream packets. I'm gonna have to do some research into that :-(

A longer packet trace would be very helpful. If you can upload the whole thing somewhere, that would be best. Binary format better than text.

poleguy, 2015-09-15:
Here's the wireshark capture with the original case causing dropped data. If you can handle this format I can do some more captures. Unfortunately I didn't save the captures I did earlier.
https://www.dropbox.com/s/0tjbuf7pvgdoash/dropped_data.pcapng?dl=0

If that format works, let me know what test cases you want me to run and I can produce captures. I guess the options are: 
bad messages due to delayed ack. (more than 2128, truncated to 2128 like above)
good messages (less than 2128)
other combinations of 1) setting of TCPNoDelay (T/F), 2) setting of TcpAckFrequency (int), 3) length of message (22+10*n) 
you call it.

Notice that in this case the final packet of 642 drops the last few bytes. It should contain a prompt at the end, but it's getting dropped... "\r\n>>" 

I don't have RTS/CTS lines on my board, so that's not an easy option in hardware for me.

Does a queue of buffers solve this problem? Does that let you handle delayed ACK's? If so, I'm for it! How can I help?

I don't think I understand this problem thoroughly enough to really make any suggestions at this point. Maybe I need to try to read through all the code and set up an environment to compile this myself. :-(

tve, 2015-09-15:
I just tried to repro you problem, but I can't. If you're not using an esp-12 or wroom-02 module, I highly recommend you do. I had lots of these problems early on with esp-03 modules. The -12 consistently shows 10dB better signal/noise in my tests. That's 8x stronger!

Anyway, I wrote a simple sketch that sends N kilobytes in one go at 115200 baud. I'm receiving it fine every time and I timed it at 8.7 seconds when it's supposed to take 8.8 seconds, so I'm pretty sure there's no gap in there.

I looked at some of the SDK code and I believe that if I switch to the built-in buffering I can fully pipeline TCP segments, but it's not easy code to read (sparse chinglish comments). Looking at you dump now...

tve, 2015-09-15:
The pcap file you provided on dropbox only contains the couple of packets you have above. A full trace  on one of your failed long dumps would be great. I'm not so much looking for the missing bytes as I'm looking for the overall pattern of data & ack packets to understand what's going on.

WRT "Does a queue of buffers solve this problem?" that's what I'm trying to answer by looking at your trace. If it's a matter of some start-up round-trip delay or occasional ACK delays, then yes, more buffering would help. But if we're not getting the data rate we need because there aren't enough packets in the pipeline then any amount of buffering will fill up.

tve, 2015-09-15:
I took a trace of what I'm seeing. You'll be shocked. Basically a long string of short packets with 80 or 4 bytes of data. The esp8266 has absolutely no problem keeping up. (I don't yet understand where the 4 byte packets come from.)
```
22:00:59.633747 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [S], seq 3670505175, win 29200, options [mss 1460,sackOK,TS val 482366460 ecr 0,nop,wscale 7], length 0
22:00:59.844332 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [S.], seq 2237670805, ack 3670505176, win 5840, options [mss 1460], length 0
22:00:59.844372 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 1, win 29200, length 0
22:00:59.844446 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [P.], seq 1:5, ack 1, win 29200, length 4
22:00:59.853980 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 1:81, ack 5, win 5836, length 80
22:00:59.854014 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 81, win 29200, length 0
22:00:59.855174 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 81:86, ack 5, win 5836, length 5
22:00:59.855207 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 86, win 29200, length 0
22:00:59.861068 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 86:166, ack 5, win 5836, length 80
22:00:59.861095 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 166, win 29200, length 0
22:00:59.862611 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 166:170, ack 5, win 5836, length 4
22:00:59.862639 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 170, win 29200, length 0
22:00:59.874233 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 170:250, ack 5, win 5836, length 80
22:00:59.874260 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 250, win 29200, length 0
22:00:59.876558 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 250:335, ack 5, win 5836, length 85
22:00:59.876585 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 335, win 29200, length 0
22:00:59.882250 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 335:415, ack 5, win 5836, length 80
22:00:59.882277 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 415, win 29200, length 0
22:00:59.883441 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 415:419, ack 5, win 5836, length 4
22:00:59.883468 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 419, win 29200, length 0
22:00:59.889409 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 419:499, ack 5, win 5836, length 80
22:00:59.889436 IP 192.168.0.3.59753 > 192.168.0.75.23: Flags [.], ack 499, win 29200, length 0
22:00:59.890520 IP 192.168.0.75.23 > 192.168.0.3.59753: Flags [P.], seq 499:503, ack 5, win 5836, length 4
```

vamposdecampos, 2015-09-15:
If you're looking for guaranteed delivery (in the face or arbitrary amounts of packet loss, the likes of which one might see over WiFi), you need some form of flow-control.  TCP effectively converts loss to backpressure, but this needs to be propagated upstream: in this case, to the FPGA UART transmitter.  As @tve noted CTS/RTS is one option to do this;  another may be in-band XON/XOFF if you don't have the extra pins.

poleguy, 2015-09-15:
At this point, what are we trying to debug? It seems you care more about long packets that fail while every packet is being ack'd. I'm more bothered that I won't be able to use this for any packet longer than 2128 without changing default windows settings.

Unfortunately I didn't save the captures I did earlier, so I sent you the one pcap that I had, which was only the relevant packets. If you want to look at the 10282 length packets failing I'll try to send a full trace tonight. Of course that doesn't have a chance of working with default network settings, so it'll have every packet ack'd.

I'm not entirely shocked at what you see. It's similar to what I saw when I have my machine set to ack every packet. Lots of tiny packets! I question what "length" represents in your dump. It doesn't seem to be packet length as it is not possible to have 0 length packets.

@vamposdecampos I'm not looking for guaranteed delivery. Of course you won't get this in an arbitrary loss case without flow control. I'm well aware of that. Ideally a transparent uart bridge would provide:
* Sane operation with default Windows ack and nagle settings in good RF conditions, such that you can use a telnet window without losing data at moderate message lengths (1k-256k would be ideal).
* No loss of data in the case of of moderate (10%) packet loss

Somebody correct me if I'm wrong, but 802.11g can get 8-24Mbps depending on many variables, but this is much higher (69x-208x) than the 115.2kbps we're trying to push. That says to me that you should be able to build a transparent link with reliable 115.2kbps with even very high packet loss. I'm not saying it's possible with the current buffering setup, IP stack, hardware, etc. But it seems this 'transparent' link is coming up pretty short at the moment:
* 100% loss of packets > 2128 in good RF conditions with default Windows settings.
* 50% loss of messages once messages approach 10k length, despite 100% ack. (I'll send a dump... no idea what the packet loss rate is. It should be very low, as I'm at very short range, but I can't claim zero.)

tve, 2015-09-15:
@vamposdecampos : fully agree. The esp8266 only has about 25KB of ram left after it starts up. A "few buffers here and few there" very quickly eats that up specially if one wants enough room left to be able to pull up a debug/console web page. In the end, given what poleguy is trying to do, I see the following options:
- limit the end-to-end frame size to 2-3KB and use end-to-end acks or requests for each frame, there's enough memory to make that work and perform reasonably well
- fix the RF and wireless so it performs to spec, in which case I anecdotally see >80% success rate for 100KB frames, handle the failures by retrying the whole frame
- add hardware or xon/xoff flow control on the serial port
- spend a lot of work trying to increase the the success rate without flow control with uncertain results

tve, 2015-09-15:
@poleguy what I'm trying to see from the longer packet traces is whether we're dealing with an ACK policy issue or packet loss or round-trip times or...

When you say "such that you can use a telnet window without losing data" what telnet window do you mean? Putty?

When you say "in good RF conditions" I've found that to be tricky. Have you tried an unencrypted access point? I used to see a radical difference not due to RF but due to bugs. When monitoring the RF I saw tons of RTS retries from the esp and crap like that.

I know that the Wifi is N times faster, but there is very limited buffer space on the esp. If you were to write your own firmware you could use all of the available memory just for your one stream, but in esp-link as a general-purpose tool that doesn't make sense. 2KB buffering only lasts 177ms at 115200 baud and it's easy to get that type of ACK delay with TCP. Heck, loose one packet and the retransmit timer is most likely higher than that.

tve, 2015-09-15:
One more thing about "802.11g can get 8-24Mbps depending on many variables". You are of course correct, but the issue here is the amount of buffering required to cover a round-trip. That's almost completely independent of bandwidth. With a reliable protocol, like TCP, the sender must keep a copy of the data sent until it receives an ACK from the receiver. So, @115200bps if at any point in time data is unacked for 400ms, for example, then the sender must have at least 4KB buffered for possible retransmission. These 4KB have data *already sent* and in addition to buffering new data not yet sent. If you look at TCP retransmission timeouts and such, it's easy to get into those time-scales with just a couple of lost packets, specially at the start of a connection or with lossy connections where TCP doesn't have a tight round-trip-time estimate.

The bottom line is that while there is certainly room for improvement in esp-link (and I would like to find time to improve the buffering to be able to allocate more) there are also some hard realities about reliable streaming protocols and buffering requirements proportional to round-trip times.

poleguy, 2015-09-15:
When I say telnet window, yes, I mean something like putty, realterm, zoc, or plain ol' windows command line telnet. In my application the serial port is generally a command line interface, which is sometimes accessed by a gui, but is often used for configuration or debug manually using a telnet terminal.

I have not tried an unencrypted access point. I am also using an ESP-01. So those things may be having a negative impact in my setup.

I'll capture some data to try to make things more reliable for esp-link, but it sounds like I may have to branch the code to accomplish my goals of a transparent command line interface, or else convert my boards to do something more advanced than a command line interface.

tve, 2015-09-16:
mhh, I may be reproducing some of your problem... when I use windows telnet I get one data packet, windows sends an ack, and then the connection is completely dead. Makes no sense. Most likely I'm having some memory corruption or am using the SDK in a way it's not intended -- or there is a bug in the SDK. I do see, however that my windows 7 box takes 190ms to ack the first data packet. Dunno why it's taking so long, but that's not gonna help anything for sure.
Oh, as I'm writing this, I think I know what's happening, the esp gets a buffer overflow even before it gets the ack and it locks into an infinite loop printing "espbuffsend: txbuffer full on conn 0x3fff47d8" and the ACK never gets processed. Haha! "Fix" coming...

tve, 2015-09-16:
Please try release 1.0.4. I increased the tx buffer to 2*1460 bytes and fixed the problem where it gets stuck printing errors. I did a 100KB dump to telnet on a windows 7 box and I saw one error. The packet trace shows every other ack being delayed by 200ms, which is followed by at least one full-sized packet in the other direction. Not ideal, I hope you can turn nagle off 'cause else it's going to be touch&go. If you want to compile your own version with a larger buffer, the definition is here: https://github.com/jeelabs/esp-link/blob/v1.0/serial/serbridge.h#L12 (be sure to use the v1.0 branch).

tve, 2015-09-16:
One more update: you may want to try netcat for windows, such as https://eternallybored.org/misc/netcat/ that may set TCP_NODELAY on the individual socket, I don't know. Something to explore...

poleguy, 2015-09-16:
I didn't see your messages until I already captured these:
https://www.dropbox.com/s/b5p5qvi1iao63b6/three_bad_three_good_three_bad_default.pcapng?dl=0
https://www.dropbox.com/s/pyrk7vft09p929p/0x400_and_0xd1_TcpAckFrequency1.pcapng?dl=0
https://www.dropbox.com/s/11zfe2yke7phdi6/0x400_and_0xd1_TcpAckFrequency1_TCPNoDelay1.pcapng?dl=0
https://www.dropbox.com/s/5x94viq27flhsxe/0x400_0xd1_0xd0_default.pcapng?dl=0

Various combinations of settings and packet lengths.

I agree the ACK situation is not ideal." But since it is the default
behavior on windows I'd like to try to roll with it.

I'll try 1.0.4 tonight. Thanks.

poleguy, 2015-09-16:
Here is a link to a quick test on 1.0.4:
https://www.dropbox.com/s/zzfjdzau56zm3oy/1.0.4_0x400_repeated_windows_default.pcapng?dl=0

Nice! It is now possible to get my messages through without trouble using standard windows ACK settings.
The above capture is with long packets. The message should be 10282 long. They are getting through now at last. There are still lost messages, but I haven't actually looked closely at this capture yet to see what's the cause. I will look more closely after dinner or so.

poleguy, 2015-09-18:
Well, I looked more closely at the log. The successful transfers have a very regular pattern (e.g. starting at packet 1226):

Send 80 characters
197msec delay...
Receive ACK.
Send 1460 characters
Send 809 characters
Receive ACK.
Repeat....

This totally makes sense because 2269 bytes collect in 197msec at 115.2kbaud, and windows delays acks by about 200msec. 

The instances where data are lost the sequence looks more like this...

Send 80 characters.
197msec delay...
receive ack
**additional delay**
Send 1460 characters
Send 1460 characters.
Receive ack.

Examples are at packets 173-189,  472-477, 916-920, 989-997, 1014-1018, 1036-1046, 1084-1088, 1155-1158, 1166-1171, 1191-1205.

One example where there is a delay but not long enough to cause apparent data loss: 981-984

Most of these additional delays leading to message loss seem to be due to ARP Who has? messages. Some of these additional delays seem to be associated with TCP Retransmission. Some are associated with both. One has no obvious cause (1155-1158).

Most of these delays are about 100msec...

It seems this chip regularly gets busy spending its time handling and ignoring Broadcast ARP messages and the buffer fills up.

A naive first guess would be that adding enough buffering to cover an additional 100msec or so of delay to handle ARP messages would be enough to fix most of the problems in this capture. One more 1460 character buffer would cover about 127msec of delay at 115.2kbaud.

tve, 2016-12-11:
I don't think there is much more I can do here, it seems to be what it is with LwIP and limited buffer space. Please reopen if there's something concrete. Thanks for all the troubleshooting help!

closed 2016-12-11 by tve