jeelabs/esp-link #48: Static ip settings lost after wiflash update.

bc547, 2015-09-13:
I configured esp-link to use a static ip address instead of DHCP.

After compiling the latest 'master' firmware and uploading it to my ESP-7 via wiflash, the ESP switched back to DHCP and lost the static ip settings after the reboot.

Is this the intended behaviour?

(I'm not yet familiar enough with the code to fix/analyse it myself at this moment, so i report it here)

tve, 2015-09-13:
No, that's not intended. Grr

tve, 2015-09-16:
Q: from which version did you upgrade to master from? I had to change the checksum calculation recently, so that would have lost all flash settings.

bc547, 2015-09-16:
I did upgrade from your precompiled binaries (2.0.beta2) to the master branch code at commit f057a65.

tve, 2015-09-21:
I double-checked, this is expected. I switched the flash config checksum to a proper crc after beta2. Sorry for the hassle.

closed 2015-09-21 by tve

bc547, 2015-09-21:
No problem :-) Thanks for the effort!