jeelabs/esp-link #51: Enable secure "standalone access point only" mode

Poquaruse, 2015-09-18:
Hi,

First of all thank you for that nice piece of software! :-)
For my use case I need a mobile, battery powered standalone access point with encryption that bridges UART to TCP. From what I have seen, with ESP-link the ESP8266 can only be used as a station in a network but not as a secured access point. Could you please think about implementing this feature?

Thanks and best

tve, 2015-09-18:
There is a #define in the Makefile at https://github.com/jeelabs/esp-link/blob/master/Makefile#L119 that keeps the AP going. Set that to 'no' and build your own version and you're all set :-).

Poquaruse, 2015-09-18:
Thanks for the very fast answer. :-)

And getting a secure AP is done by first flashing an AT-software, configuring the AP, then flashing ESP-link compiled with the "no"-flag set. Correct?

Then I'll just have to find a way to compile everything on Windows. ;-)

tve, 2015-09-18:
Yes, that's the way you would have to do it. There is no support in esp-link for configuring the AP wifi settings.
Compilation on windows is supported, but I'm not the maintainer of that. You should ask brunnels in the gitter chat (see badge in readme.md) for help if you're having trouble.

Poquaruse, 2015-09-18:
Ok, I will do so as soon as I find some spare time... In my experience it's a PITA to compile stuff like that on windows. Probably a virtual machine is the easier way. 

Anyhow, I'd greatly appriciate it if you at some point in the future found the time to make configuring an AP via the website possible. ;-) I'm surprised that this use case hasn't arised before by someone else?! 

Thanks again and have a nice weekend

tve, 2015-09-18:
well, someone else wanted the ap mode and that's why there's the switch in the makefile. I just can't maintain an infinite number of options... The AP mode is pretty limited in a number of ways and so it's not a great solution anyhow: no power-down and no routing are two issues.

Poquaruse, 2015-09-20:
I'll give it a shot as soon as I find a sufficient amount of time. :-)

Of course you can't support every user's wishes. At least there IS a possibility, even though it is not available via GUI...

I've read that AP mode is not without its problems. Unfortunately, for some (mobile) use cases it is the only possiblity as there simply are no access points around. BTW: setting up the toolchain on Windows isn't as easy as I thought. :-P Here comes the virtual machine... ;-)

omersiar, 2015-10-11:
Hello i have been testing AP mode, first i flashed my ESP with AT firmware -aithinker- then set the specific paremeters like "AT+CWSAP="NoWorriESSID","password",3,3" and "AT+CWMODE=2" and you can finally flash esp-link firmware without blank.bin.

Its working great with this way.

I guess its not possible to use esp-link in both ends like wireless RS-232 solution?

tve, 2015-11-16:
From gitter chat related to this topic:

>@44simon: the esp module is visible in AP mode as a ESP_XXXX, do you know exactly in source where can I change it to for example Receiver1
I have found something like "ChipID", I suppose that is located somwhere near this in source code

Search for wifi_set_opmode(3), which turns on sta+ap mode, it's at https://github.com/jeelabs/esp-link/blob/c3e1a8a83bc4ec6e346ac651d5d816458da457c8/esp-link/main.c#L199 and at https://github.com/jeelabs/esp-link/blob/c3e1a8a83bc4ec6e346ac651d5d816458da457c8/esp-link/cgiwifi.c#L309
To set the SSID I believe you need to call wifi_softap_set_config

katyo, 2015-11-21:
I think, the ability to use encrypted access point with setting password through web interface is a high priority thing. Because in case when access point, which we connect to in station mode, is (temporary) unavailable we have big security problem. Also, web interface doesn't support auth (Is it so?).

I may to try add the ssid and password options and access point only mode in near time.

omersiar, 2015-11-21:
Web interface simple HTTP auth is possible

Sergey82K, 2017-01-11:
I tested esp-link-v2.2.3. Good work! I can change "SSID" and "Soft-AP Auth Mode" via "Soft-AP Settings", but finally I get two SSID: one with default ESP_**** and open autenification and second with setings with I was set in "Soft-AP Settings" (secured AP). Configuration WEB-site was opening by both AP. Is any chance to disable first AP with open auth?
I need AP with custom SSID and password.
And I try to mark checkbox "Soft-AP SSID hidden", but both SSID still visible.

[image]