jeelabs/esp-link #53: Enabling CTS/RTS flow control.

tiltit, 2015-09-21:
Hi, thanks for the great software. Would it be possible/realistic to have an option to enable hardware flow control for the serial connection? Or would it interfere with the network side of things? I might look in to it myself if It is at all possible.

tve, 2015-09-21:
I don't think it's all that difficult. It depends a little in which direction you want it. The ESP does have pre-assigned RTS, CTS, etc pins for the UART. I don't know how they get enabled and configured. I do know that on the ESP's RX side you can configure the RX FIFO to generate an interrupt when it reaches a certain fill level and you'd use that to toggle a gpio line. It's all not very well documented (like most things esp8266...). If you have a hard time finding it, lemme know.
There's also the issue that you have to do flow-control inside of esp-link. So when the RX fifo fills and you have no more buffer space, then you need to turn the fifo interrupt off. That's the easy part. Now you have to figure out when you can turn it on again...

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve

Sergio-tix, 2023-08-23:
Has there been new information? My device only works when there are signals RTS/CTS. Signals Rx and Tx are not enough.

uzi18, 2023-08-23:
@Sergio-tix try to add bridge RTS-CTS