jeelabs/esp-link #54: esp-link not responding to serial protocol commands?

cburgess129, 2015-09-24:
I have esp-link working beautifully on my esp-03 connected to an arduino UNO.  I can upload 32K sketches via wifi with 100% success.  

My problem is when I try to make outbound TCP connections from the arduino, I get no response from the esp.  I can see the commands coming in from the uC serial port when I look at the viewer, and if I connect to port 23 with a terminal viewer, I see the commands on the serial link but esp-link is not responding at all - the command is treated like regular I/O

I am trying this command:

~@1Tarduino.cc:80
~100GET /asciilogo.txt\r\n\r\n\0

I am sending a \n after each line.

What am I doing wrong?

tve, 2015-09-24:
Ouch, I must have left the docs for the old command protocol somewhere. Since you're using an esp-3 I assume you have a 512KB flash chip. The latest V1 release is 1.0.4 and it does not currently support commands from the arduino, that is only in V2, which currently only fits into larger flash chips. @brunnels is working on a version that fits into the small flash but without over-the-air update, i.e. only serial flashing. The command protocol has changed to a slip framing as used by espruino and I'm realizing that I need to write up docs for all that.

tve, 2015-09-24:
You can find a demo sketch for some REST calls at https://github.com/tve/espduino/blob/master/espduino/examples/demo/demo.ino and for MQTT at https://github.com/tve/espduino/blob/master/espduino/examples/mqtt/mqtt.ino
If you would like to try the V2 firmware in your esp-03 the best would be to get in touch with brunnels in the gitter chat (see badge at top of the repo's readme.md).

cburgess129, 2015-09-24:
I suspected it have some ESP-12 modules on the way and will load up the v2 firmware when they arrive and give it a try - thanks.

closed 2015-09-24 by cburgess129

cburgess129, 2015-09-27:
Just wanted to let you know that after upgrading the flash chip on my esp-03 to 32Mbit, I was able to flash V2 firmware and successfully run the REST example from the espduino site.

I am also able to upload sketch via Wifi using the same firmware, which gives me all of the functionality I was looking for.  Thanks for all your efforts!

tve, 2015-09-27:
awesome, thanks for reporting back!