jeelabs/esp-link #55: RS-485 TX Enable

ATIU66, 2015-09-25:
Hi, is there a way to use the serial activity LED to control the TX enable for an RS-485 communication?  Please point me to what files to edit to make the LED "blink" only on a TX activity duration, RX can be ignored.  This should be sufficient to control the TX for an RS-485 communication.  Thanks.

tve, 2015-09-30:
Ooops, I overlooked your issue, my apologies. I'm afraid I don't remember RS-485. Do you need a signal that is active while something is being transmitted? If so, that might be a tad difficult given that the UART has a 128 byte buffer, so the software doesn't really keep track of when it becomes empty. You could perhaps use one of the modem control signals for that, but I haven't looked into the UART capabilities around that. I surmise you have an esp-01, in which case you don't have easy access to those pins anyway. But maybe I don't really understand what you're trying to do...

ATIU66, 2015-09-30:
Yes, I need a signal that goes high for the whole duration of the TX port activity.  I have the ESP-12, with most of the GPIO pins out.  My intention is to keep the ESP module with its built in antenna mounted separately where the WIFI signal is better while keeping the uC and other electronics mounted separately.  The RS485 should keep serial link more robust when running longer wires than just TTL serial signals.  Please advice the hardware flow control pin of the ESP-12 module so I can test these.  Thanks,

tve, 2015-11-16:
Someone else asked... Some more info:
This may be useful: http://stefanshacks.blogspot.com/2015/04/wi-fi-for-lego-mindstorms-nxt-esp8266.html and http://www.esp8266.com/viewtopic.php?f=21&t=1098

All the stuff received over wifi and sent to the uart goes through https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L232, so that's a place in the SW you could hook something (probably best in uart.c, which is where the function called lives: https://github.com/jeelabs/esp-link/blob/master/serial/uart.c#L147).

Abdullbasit, 2016-01-10:
You can modify the function uart0_tx_buffer to : 
uart0_tx_buffer(char *buf, uint16 len)
{
  uint16 i;
//set TX_ENABLE pin
  for (i = 0; i < len; i++)
  {
    uart_tx_one_char(UART0, buf[i]);
  }
////Reset TX_ENABLE pin
}

deece, 2016-04-03:
I'm currently working on this. This is the first time I'm working on this platform, so some assistance from someone more familiar would be good.

I plan on hooking the UART_TXFIFO_EMPTY_INT_ST interrupt to disable the TX_ENABLE pin when done, but it's not clear from the documentation whether this will trigger when the buffer is empty, or when the byte has acutally been transmitted. I'm guessing it's the former, in which case I would need to then wait for the UART hardware to complete the transaction, or trigger a timer based on the baud rate to dead reckon when the hardware has finished transmitting.

deece, 2016-04-03:
Alternatively, it may make more sense to use the RTS line for this.

DavidJRichards, 2016-09-22:
Greetings deece,
I am interested in your fork with RS485 data direction output, can I help with the testing please?
I would like to start by downloading a pre-compiled binary, later I hope to be able to build my own.
Kind regards, David.

tve, 2017-06-18:
Implemented in v3.2

closed 2017-06-18 by tve