jeelabs/esp-link #57: Buffer issues?

scargill, 2015-09-30:
Hi

I grabbed the version of this code to run on ESP-01 units... it works - I can set up a port and talk to it. I grabbed a commercial package to map a port to a COM port on my PC - and checked - bidirectional communications over the real port to the virtual port - no problem - at 115k.  but when I tried to use this to program an ESP12 remotely - no joy. Most of the time, my Eclipse setup - which programs boards no problem every day using an FTDI, would not have it (I booted the ESP12 up with GPIO0 grounded as usual). Could it be that the unit is running out of buffer RAM?

tve, 2015-09-30:
the buffering is static and at 115200 baud it's not going to overflow. What are you observing? Do you see any errors in the debug log page? Are you using v1.0.4?

scargill, 2015-09-30:
Hi there

 

I’m using 1.04 on an ESP-01.  I initially hooked that to an FTDI and programmed it up with 1.04 – I then setup the WIFI noting the address.  I ran SERIAL-IP on a PC (windows 10 64 bit) to set up the address 192.168.1.27 port 23.  No problems so now I have COM22.   (I’m using the serial-IP software as I have no idea how to tell Eclipse on a PC how to use an IP address instead of a COM port).

 

I ran Putty and checked that COM22 would work with COM6 (the FTDI) at 115k baud – no problem  - typing both directions I see text coming out. I then connected the ESP-01 to an ESP-12 board which I can program no problems with an FTDI.  I reset the ESP-12 board with GPIO0 grounded as usual – and told Eclipse to go off and program it. On a scope you can see attempts at communication to the ESP-12 but nothing back. At one time I say it make a start at programming but immediately stopped. I’ve not been able to replicate that.

 

Didn’t think of checking the debug log page.

 

Pete.

tve, 2015-09-30:
What you are doing sounds plausible. @brunnels said that he had trouble with virtual com ports. I haven't tried it. It would be helpful if you could look at the debug log to see whether it's sending the reset (which I know you don't have hooked up) and whether there are any error messages. The debug log page doesn't auto-refresh, but there's a refresh button there.
Come to think of it, I haven't tried using esp-link to flash another esp. I should try...
If you would like some live troubleshooting help, please try the gitter (see badge at top of readme on github), that's easier for back-and-forth. I'm west coast, so about to hit the sack...

tve, 2015-09-30:
Mhh, actually, is your firmware in one chunk and you are trying to flash only that chunk? Most firmwares comes in multiple chunks and I suspect that a reset is required between them, no?

scargill, 2015-09-30:
And now I’m going to ask the REALLY silly question. Where is this GITTER – I looked all over for it yesterday.

scargill, 2015-09-30:
Good idea – but 99% of the time it’s not even starting the first one – in fact I think I only once got it to start up…  I’m thinking instead of an esp 01 I should put it on another 12 and try using the reset line anyway. Does that tristate when it’s finished?

 

Pete.

tve, 2015-09-30:
Gitter chat: https://gitter.im/jeelabs/esp-link
Badge in the readme: https://github.com/jeelabs/esp-link/releases
If you have a chance to capture the debug log, that might help. I'll give it a spin tonight after work to see whether I can repro. Thanks for trying it and reporting the issue.

tve, 2015-10-01:
I just gave this a try and it worked quite well, but was a bit of an ordeal to set-up. I used open source software called com0com and com2tcp. I downloaded com0com from http://pete.akeo.ie/2011/07/com0com-signed-drivers.html because I needed signed drivers. I installed the drivers as described (running `setupc.exe` and typing `install - -`). I then ran:
```
C:\Program Files>com2tcp.exe --baud 115200 --ignore-dsr \\.\cnca0 192.168.0.75 23
OpenC0C("\\.\cnca0", baud=115200, data=8, parity=no, stop=1) - OK
Connect("192.168.0.75", "23") - OK
```
Where 192.168.0.75 is the esp-link module. I then ran some old "ESP8266 Flash Downloader" app that I had lying around specifying `\\.\cncb0` as com port, `0x01000` as address to flash, and `D:/tve/Downloads/user1.bin` as bin, which happens to be the esp-link bin. I reset the esp8266 to be programmed manually and then hit the download button. The output was what you'd expect:
```
connecting ....
Erasing flash...

Writing at 0x00000000... (0 %)

...

Writing at 0x00042400... (100 %)

Leaving...
```

So it seems possible...

closed 2016-01-01 by tve