jeelabs/esp-link #58: using V2 on ESP-01 (with 32Mbit flash) - changing PIN assignments causes ESP to lose config

cburgess129, 2015-10-02:
I am using version 2 on an ESp-01 with 32Mbit flash.  I have notices two issues:

Scenario 1:
While ESP in WAP mode, connect to esp@192.168.4.1 and set PIN assignment to esp-01(AVR)  
ESP goes into STATION mode and joins AP
Everything seems to work until you remove power from the ESP
after power loss, ESP loses PIN assignments and reverts to esp-bridge PIN assignment

Scenario 2:
While ESP in STATION mode (joined to AP), connect to esp@IP and set PIN assignment to esp-01(AVR) 
ESP reset and lose PIN and AP configuration
ESP reboots into WAP mode and needs to be re-connected to AP, PIN assignment reverts to esp-bridge

Is this expected behavior?

tve, 2015-10-02:
Neither scenario is intended I will have to repro. You are using 2.0.rc1, correct?

cburgess129, 2015-10-02:
v2.0.beta2

cburgess129, 2015-10-02:
I flashed rc1 and had the same issue.
Tried some different things and found that the following combination seems stable and does not lose the config between resets:

blank.bin @0x7e000
blank.bin @0x3fe000
boot_v1.4(b1).bin@0x00000
user1.bin@0x01000

Maybe I had some old config data floating around somewhere.  Seems good now.

tve, 2015-10-03:
Thanks for reporting back. I'm having similar sounding difficulties in another project that also uses SDK 1.4. I wonder whether there's something broken or otherwise undocumented happening... If it surfaces again, please reopen the ticket!

closed 2015-10-03 by tve