jeelabs/esp-link #59: Compile firmware for 512kB flash devices

robert-budde, 2015-10-03:
Hi!

Thank you very much for the great firmware. This was what I was really looking for - having a serial bridge with the comfortable option to change STA settings without reflashing.

As the software I am using only allows port numbers above 1023 to be used for TCP I have to alter the port number - but I am not able to flash the software because its to big. I already altered the Makefile to assume 512kB flash (which correctly results in a warning that the firmware will not fit) and removed mqtt and rest modules - still the size is too big.

Any suggestions?

BR
Robert

robert-budde, 2015-10-03:
btw: checking out the v1.0.4 tag does not work either - make fails as some stdint types are unknown!?

tve, 2015-10-03:
V2 will not fit into 512KB. @brunnels is working on making a non-OTA upgradable version that will fit. Thanks for the notes WRT 1.4, I had not realized that 1.0.4 doesn't fit into 512KB with 1.4. Mhhhh...

tve, 2015-11-14:
V2.1.beta now works with 512KB devices.

closed 2015-11-14 by tve