jeelabs/esp-link #60: tcp connection & µc Console problems

stramike, 2015-10-09:
Hi!

first of all thanks for esp-link! it's awsome! you did a great job!

I've got some questions:
I'd like to connect to a TCP server. All esp-link setup was ok! I used NodeMcu to flash the esp module with boot_v1.4(b1).bin (0x00000) , blank.bin (0x7E000) , user1.bin (0x01000). I set esp 01 avr mode, get the IP address and everything looks working fine. But, before connecting arduino and esp8266, when i check µc Console i get few characters (most of them are unreadable chars) printed every 20-30 seconds. Why? They apparently don't have any sequence. They just look like random printed chars. Shouldn't the µc console be empty when not connected to any microcontroller?

Then, when i connect Tx and Rx pins to Arduino the µc Console automatically starts printing huge amount of unreadable chars also if i don't print anything from arduino serial. What am I doing wrong ?

I tried to connect to my TCP server and i still get huge amount of unreadable chars. The same with the examples you provided.

After power-cycled esp8266 when i check µc Console huge amount of unreadable chars are printed also if not connected to arduino.  Any idea?

I also tried to flash again esp8266 but nothing changed.

It would be very helpful if you can provide more detailed examples with arduino-side code.

Thanks a lot for any advice!!

Michael

tve, 2015-10-10:
Sounds like you have the wrong baud rate set on the uC Console page? Or, if your arduino sketch reads input, it will receive some garbage at about 74k baud due to the esp8266 booting. This cannot be removed on the esp-01.

Where is your TCP server and how do you expect to connect to it? Which version of esp-link did you flash?

stramike, 2015-10-10:
I flashed esp8266 at 115200 baud rate with esp-link v1.0.4. On the uC Console page 115200 baud rate is setted. The problem starts before any microcontroller is connected to esp8266. It starts printing random (most of them unreadable) chars on the uC Console page about every 20seconds and it NEVER stops, it's not just about esp8266 booting.

The idea is to use esp8266 with esp-link to make a web based domotic project. Last week i received esp8266 modules and just started to play with them. I expect to connect to tcp server sending via arduino serial "~@\<n\>T\<hostname\>:\<port\>\n" where \<n\> is the connection number, \<hostname\> is tcp server's ip address and \<port\> is the server port number, and to get the response ~@\<n\>C\n if everything is ok. The server is a super-simple multiclient tcp server that for the moment doesn't do anything special. I already tested it sending commands through telnet client and it works fine. 

 But the problem starts before i try to connect to the server. The problem is that i have all that garbage with no arduino connection that never stops and when i connect arduino to esp8266 the garbage get worst also if i don't read or write anything via arduino serial. What could it be the problem? Any advice how to check what can be the problem?

Thanks for your help.

Michael

stramike, 2015-10-16:
After few tries, i changed the power supply. I was using a 500mA power supply (i read on esp8266 datasheet that it absorbes up to 215mA) and i thought it was enough. I was wrong! After switching to a 1A power supply everthing works perfectly!!
 I also read issue #54 and i realized i need a different way to read and store data into a mysql database.

Thanks a lot for your time, and many thanks for your great job!

Michael

closed 2015-10-16 by stramike