jeelabs/esp-link #61: esp-link creates funky files when connected to an openwrt router

zoobab, 2015-10-17:
Hi,

esp-link created some funky files when I connected it to a serial console of an openwrt router:

    root@GL-iNet:/# ls
    ls
    
    **
    CONN
    Flash
    Httpd
    MCU
    Reset
    SER
    Scan
    Serbridge
    Starting
    Turning
    Wifi
    add
    aid
    bin
    cnt
    connected
    dev
    dhcp
    download_file.socket-0
    espFsInit
    etc
    exccause=0
    f
    gateway:
    hostname:
    ip:
    ip:192.168.0.10,mask:255.255.255.0,gw:192.168.0.1
    lib
    mnt
    netmask:
    overlay
    pm
    proc
    rom
    root
    sbin
    scandone
    sys
    tmp
    usr
    var
    www
    root@GL-iNet:/#

Any idea where those files come from?

tve, 2015-10-17:
That looks like the uart debug output which comes on initially until it's associated. You can turn that off on the debug page. However, unless you're using an esp-12 with swapped uart pins, there is some stuff the ROM prints at boot time which you can't get rid of.

tve, 2015-12-31:
I don't think there's anything to be fixed in esp-link here. Please reopen if there is.

closed 2015-12-31 by tve