jeelabs/esp-link #63: MCU/avrdude synchronisation issues

bc547, 2015-10-28:
### Introduction
Due to the way avrdude is programmed and some MCU bootloader quirks, you sometimes need a lot of luck to successfully program the MCU over the air.

The intention of the algorithm below is to solve these issues by introducing more intelligence in esp-link. It is the result of a lot of experimenting and analysing bootloader/avrdude source code but it's very well possible I still missed some quirks at this moment :-).

Below I try to describe the algorithm in words. I'm also busy implementing this algorithm as a small proof-of-concept network proxy that sits between the programmer (avrdude) and esplink. Since I found another quirk yesterday evening, my current code does not yet completely implement the description below. Once it does, I'll share it here also. In order to get quick results, it is implemented in perl. I'm also trying to make it easier to port to esp-link by using a similar 'API' like the esp-sdk provides (afaik). It should run on both linux and windows (with e.g. ActiveState perl).

### Used Terminology:
MCU = microcontroller that speaks STK500 protocol
esplink = esp module with esp-link
avrdude = client pc which runs the programming tool (e.g. avrdude)

### General principle:
During the synchronisation phase of the STK500 protocol, esplink itself talks to both MCU and avrdude. Once esplink synchronisation is established with both MCU and avrdude, esplink merely performs a bridging function between avrdude and the MCU.

### Algorithm:
Once avrdude establishes a connection on port 2323 of esplink, esplink resets the MCU and starts to send STK_GET_SYNC;CRC_EOP (0x30;0x20) to the MCU at regular intervals (e.g. 50ms). There are now 2 possible events that can happen:

1. **Client activity**
	Client will send STK_GET_SYNC;CRC_EOP (0x30;0x20). If the client does not get a reply, it will retry sending this sequence until it gets a STK_INSYNC;STK_OK (0x14;0x10) reply. Due to the (naive) way avrdude is programmed, we need to ignore incoming STK_GET_SYNC;CRC_EOP sequence two times (see *1). From third time esplink receives STK_GET_SYNC;CRC_EOP, esplink replies to each sequence with STK_INSYNC;STK_OK (0x14;0x10). Once esplink receives a different sequence, communication with avrdude has reached a point where we can pass the received data unmodified to the MCU (AVRDUDE_READY state). So: if the MCU is already in sync (MCU_READY state), we send the received data (and all future data) directly to the MCU. If the MCU is not yet in sync, we wait until MCU is synced.

2. **MCU serial reply**
	The only thing the MCU should send is STK_INSYNC;STK_OK (0x14;0x10). This means the MCU is in programming mode and is awaiting further STK500 commands (MCU_READY state). 	
	Depending on the used bootloader, if the MCU does not receive a command in time (within 1 second for optiboot, maybe less for other bootloaders), it will reboot and leave programming mode. Therefore, if we reach MCU_READY state, esplink should have a watchdog timer firing every 0.5s. If the watchdog expires and the avrdude communication has not yet reached AVRDUDE_READY state (see *2), esplink should change MCU_ state, resend STK_GET_SYNC;CRC_EOP (0x30;0x20) and wait for the STK_INSYNC;STK_OK (0x14;0x10) answer. Once we received the answer, we switch back to MCU_READY state for 0.5 seconds.
	If in the meantime avrdude communication reached AVRDUDE_READY state, we send the received avrdude data (and all future data) directly to the MCU. All further incoming MCU communication can be directly passed to avrdude. The watchdogtimer can be stopped now.

#### Remarks
**1:** avrdude sends STK_GET_SYNC;CRC_EOP twice and ignores the reply deliberately. See stk500_getsync function in http://svn.savannah.nongnu.org/viewvc/trunk/avrdude/stk500.c?revision=1321. On windows, the stk500_drain function does not behave well and this causes a timing issue where the incoming response channel is not cleared properly. So if esplink behaves correctly and replies to every STK_GET_SYNC;CRC_EOP, there is a realistic chance (it happened to me) avrdude sees too much replies STK_INSYNC;STK_OK (0x14;0x10) and we get protocol errors due to the unexpected input. In an ideal world, avrdude code should be fixed to properly handle 'net:' ports and not treat it like a serial port.

**2:** Yup.. happened to me also... Seems I'm attracting timing issues :-)

bc547, 2015-10-30:
The proxy code can be downloaded from https://goo.gl/EVhbLI  It contains a fixes for even more quirks than described in my first post :-) But I'm pretty sure there are still situations where timing could fail.

The code is in perl and is developed/tested on linux. It should also run on windows using e.g. ActiveState perl.

To use it:
* Edit the file and change $esplink to the ip you use for your esp-link module.
* Open a terminal and run '**perl avrdude_fixer.pl**'. You should see

 ``` SERVER Waiting for client connection on port 5000 ```

* For avrdude, use the ip the proxy is using port 5000. (e.g. -P net:192.168.1.251:5000)

The proxy also shows some debugging output to indicate what's happening when a client connects. The first number is a timestamp indicating the relative time since a new client connection was established:

```
0.000323 Client->Proxy: New connection from 192.168.1.251
0.000441 Proxy->MCU: Initiating MCU sync
0.113275 Proxy->MCU: Resending sync
0.127407 MCU->Proxy: MCU is in sync!
0.550210 Client->Proxy: Ignoring 1st sync cmd
0.650578 Proxy->MCU: Timeout... sending keepalive
0.663243 MCU->Proxy: MCU is in sync!
0.800586 Client->Proxy: Ignoring 2nd sync cmd
1.050972 Proxy->Client: Sending sync reply to client
1.051314 Client->Proxy: Client wants to continue...
1.051345 Clienti<->MCU: Entering bridging mode...
1.409955 That's all folks... A message brought to you from 'Client socket read'

```

This is normal behaviour :-)
The last line indicates the connections are closed and the proxy is ready to accept another client.

Without the proxy, I got about a 50% change programming started when using avrdude for linux and never got avrdude for windows to work. With the proxy, programming works all the time with both windows and linux avrdude.

If you experience problems, please provide the debugging output of the proxy. Also, if it fails, try a few times to see if it is a consistent fail, or just bad luck.

tve, 2016-01-01:
@bc547: thanks for doing all this research and helping to improve the AVR programing support! I believe at this point the built-in prorgamming is working pretty well, so I'm closing this ticket.

closed 2016-01-01 by tve

ykri, 2016-01-01:
__Setup:__ 
-
   Arduino Mega2560 with stk500boot_v2_mega2560 bootloader.
    ESP-01 with esp-link (compiled today from Master), serial port connected to the Mega Tx0/Rx0, gpio0 connected to arduino reset.
__Reason for coming back to this issue:__ 
-
No optiboot for Mega2560. 
Tried *avrdude /opt/arduino-1.6.7/hardware/tools/avr/bin/avrdude -DV -p m2560 -P net:192.168.1.4:23 -b 115200 -U flash:w:Blink2.ino.mega.1000.hex:i -C /opt/arduino-1.6.7/hardware/tools/avr/etc/avrdude.conf -c arduino -v -v -v -v*

- with port 23 and port 2323
- with -c arduino and -c wiring

When manually pressing reset button on mcu prior to upload it works with both ports and both programmers.

But when try OTA:

```
avrdude /opt/arduino-1.6.7/hardware/tools/avr/bin/avrdude -DV -p m2560 -P net:192.168.1.4:23 -b
 115200 -U flash:w:Blink2.ino.mega.1000.hex:i -C /opt/arduino-1.6.7/hardware/tools/avr/etc
/avrdude.conf -c arduino -v -v -v -v

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 18:03:40
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch
         System wide configuration file is "/opt/arduino-1.6.7/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/yk/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping
         Using Port                    : net:192.168.1.4:23
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: Send: 0 [30]   [20] 
. . .
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

```
and a happy New Year!

tve, 2016-01-02:
The fact that flashing the mega with the stk500v2 protocol doesn't work is a known issue. I created #87 for that. it's going to take some logic state analyzer work to figure out what's going on...

lunix80, 2016-11-15:
Hi, esp-link is great, but there are some news about working with arduino mega?