jeelabs/esp-link #64: Sending sketch to Atmega8 with optiboot via ESP01

Petre84, 2015-10-28:
I'm trying to send the sketch to atmega8 with optiboot.
Source of otpiboot: https://code.google.com/p/optiboot/downloads/list (optiboot-v5.0a.zip )
Sketch is basic blink sketch from arduino ide (link to file : https://www.dropbox.com/s/xreb6mdbw6w3rnh/blink.hex?dl=0).
Card is ESP-01 (512kb) flashed with esp-link 1.0.4. Pin assignment : esp-01(AVR).

I've switched the baudrate to 9600 with command: http://192.168.2.107/console/baud?rate=9600

Scheme:
[image]

I try to upload the sketch from UBUNTU with command

    petre@LinuxPC:~/arduino/arduino-1.6.5-r5$ hardware/tools/avr/bin/avrdude -C hardware/tools    /avr/etc/avrdude.conf -DV -patmega8 -Pnet:192.168.2.107:23 -carduino -b9600 -U flash:w:blink.hex:i -v -v -v
    
    avrdude: Version 6.0.1, compiled on Apr 14 2015 at 18:03:40
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/petre/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:192.168.2.107:23
         Using Programmer              : arduino
         Overriding Baud Rate          : 9600
    ioctl("TIOCMGET"): Inappropriate ioctl for device
    ioctl("TIOCMGET"): Inappropriate ioctl for device
    avrdude: ser_recv(): programmer is not responding
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
    avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xff
    avrdude: ser_recv(): programmer is not responding
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xff
    avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xff
    avrdude: ser_recv(): programmer is not responding
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xff
    avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xff
    avrdude: ser_recv(): programmer is not responding
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xff
    avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xff
    avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xe7
    avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xfd
    ioctl("TIOCMGET"): Inappropriate ioctl for device

    avrdude done.  Thank you.

If I change the baudrate to 115200 the result is:

    petre@LinuxPC:~/arduino/arduino-1.6.5-r5$ hardware/tools/avr/bin/avrdude -C hardware/tools/avr/etc/avrdude.conf -DV -patmega8 -Pnet:192.168.2.107:23 -carduino -b115200 -U flash:w:blink.hex:i -v -v -v

    avrdude: Version 6.0.1, compiled on Apr 14 2015 at 18:03:40
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/petre/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:192.168.2.107:23
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
    ioctl("TIOCMGET"): Inappropriate ioctl for device
    ioctl("TIOCMGET"): Inappropriate ioctl for device
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x41
    avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
    avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x2b
    avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x43
    avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x49
    avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x50
    avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x4d
    avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x55
    avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x58
    avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x3d
    ioctl("TIOCMGET"): Inappropriate ioctl for device

    avrdude done.  Thank you.

The uC resets on start (after ioctl("TIOCMGET"): Inappropriate ioctl for device) and second reset is after ioctl("TIOCMGET"): Inappropriate ioctl for device.

I've connected Esp TX to uC RX directly too but without changes.

tve, 2015-10-29:
So the 9600 baud version tells me "wrong baud rate"
In the 115200 baud rate version, if you pull up an ascii chart and decode what avrdude received you see "AT+CIPMUX=", which tells me that whatever is loaded in the AVR processor started, i.e., the bootloader popped out of programming mode.
My guess is that the round-trip time is too large. This was a common occurrence for me when using an esp-01 or esp-03 due to the shitty RF design and went away once I switched to esp-12 modules.
We're working on adding code to esp-link to try and work-around this issue, but there's only so much we can do.

tve, 2015-11-14:
I expect this to be fixed in V2.1 using the built-in AVR programmer. Please re-open if not and thank you for the report!

closed 2015-11-14 by tve