jeelabs/esp-link #66: ESP does not connect to Cisco 1200 WPA PSK TKIP

hjgode, 2015-11-07:
I have the issue that the ESP does not connect to my wireless network. The same happens with another sieral-Wifi bridge software (ESP transparent bridge)

hjgode, 2015-11-07:
Looking at the log of the Cisco AP I found

Packet to client 10d3.8af5.16ff reached max retries, removing the client 

and then the tip form internet

packet retries 128 drop-packet

This setting inside the Cisco fixed the connection issue.

closed 2015-11-07 by hjgode

tve, 2015-11-07:
Ha, interesting! Thanks for mentioning it 'cause I'm also having trouble with the Cisco APs at work. Just haven't had the opportunity to pursue it further.

hjgode, 2015-11-07:
Hi, I did also a blog post for this at
http://www.hjgode.de/wp/2015/11/07/cisco-aironet-1200-ap1231g-wpa-psk-clients-do-not-connect/