jeelabs/esp-link #67: Cannot connect to Wifi after SLIP is enabled

NiceCircuits, 2015-11-09:
I've flashed esp-link v2.0.rc1 to ESP12 module. I've configured Wifi settings and cycle the power of module. All works fine, I can connect to esp-link via web browser and via serial emulator, serial link is working fine. Debug output is as below (I've stripped a couple of first bytes with different baudrate):
```
   294> 
   294> ** esp-link v2.0.rc1 - 2015-09-20 21:37:44 - e7311d5
   294> Flash config restore *FAILED*
   295> CONN led=0
   297> SER led=14
   299> Wifi init, mode=STA
   301> Wifi uses DHCP, hostname=esp-link
   305> "ip": "0.0.0.0"
   307> "netmask": "0.0.0.0"
   310> "gateway": "0.0.0.0"
   312> "hostname": "esp-link"
   315> Httpd init, conn=0x3fff1e10
   318> Serbridge pins: reset=12 isp=13 swap=0
   331> Reset cause: 6=external
   331> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   332> Flash map 512KB:256/256, manuf 0xEF chip 0x4016
   336> ** esp-link ready
   339> MQTT_Init
   340> mode : sta(18:fe:34:a0:19:e0)
   344> add if0
   345> f r0, scandone
  2464> state: 0 -> 2 (b0)
  2466> state: 2 -> 3 (0)
  2469> state: 3 -> 5 (10)
  2470> add 0
  2470> aid 6
  2470> cnt 
  2506> 
  2506> connected with frogland, channel 6
  2506> dhcp client start...
  2507> Wifi connected to ssid frogland, ch 6
  6665> ip:192.168.1.121,mask:255.255.255.0,gw:192.168.1.100
  6666> Wifi got ip:192.168.1.121,mask:255.255.255.0,gw:192.168.1.100
 12470> pm open phy_2,type:2 0 0
 15315> Wifi check: mode=STA status=5
 15315> Turning OFF uart log
```
I've tried powering off and on a few times, it worked fine. Then I've checked "enable SLIP" in configuration and cycled power again. Connection to Wifi wasn't restored and reset is not helping. esp is not starting it's own network too. Debug log is:
```
   297> 
   297> ** esp-link v2.0.rc1 - 2015-09-20 21:37:44 - e7311d5
   297> Flash config restore *FAILED*
   298> CONN led=0
   300> SER led=14
   301> Wifi init, mode=AP+STA
   304> Wifi uses DHCP, hostname=esp-link
   308> "ip": "0.0.0.0"
   310> "netmask": "0.0.0.0"
   313> "gateway": "0.0.0.0"
   315> "hostname": "esp-link"
   318> Httpd init, conn=0x3fff1e10
   321> Serbridge pins: reset=12 isp=13 swap=0
   334> Reset cause: 6=external
   334> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   335> Flash map 512KB:256/256, manuf 0xEF chip 0x4016
   339> ** esp-link ready
   342> MQTT_Init
   343> mode : sta(18:fe:34:a0:19:e0) + softAP(1a:fe:34:a0:19:e0)
   349> add if0
   351> add if1
   352> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   360> bcn 100
 15318> Wifi check: mode=AP+STA status=0
 15318> Enabling/continuing uart log
 30318> Wifi check: mode=AP+STA status=0
 30319> Enabling/continuing uart log
```

tve, 2015-11-09:
This is a problem:
```
297> Flash config restore *FAILED*
...
335> Flash map 512KB:256/256, manuf 0xEF chip 0x4016
```
The first line says that esp-link couldn't restore the config stored in flash (which holds the SLIP flag, among other things), and the second line says that the boot sector is configured for a 512KB flash size, which is incorrect for a -12 module and won't work and, I'm pretty sure, explains why the flash config can't be restored.
To fix this you need to re-flash the bootloader and specify the correct flash size and speed. With esptool.py these are the -ff and -fs options.
Please ask if you need more details.

NiceCircuits, 2015-11-09:
Yes, it seems to be the issue. I've been using another tool to flash the device (esptool.exe from Arduino esp package) and missed flash size options. Looks like it's working fine now. Thanks a lot for your help.

closed 2015-11-09 by NiceCircuits