jeelabs/esp-link #71: Customized builds failed

katyo, 2015-11-21:
I tried to build without `rest` and `mqtt` modules and I have failed.

```
kayo@debi:~/esp-link$ make MODULES=
VERSION: esp-link master - 2015-11-21 21:43:36 - development
make[1]: Entering directory '/home/kayo/esp-link/espfs/mkespfsimage'
cc -I.. -std=gnu99 -DESPFS_GZIP   -c -o main.o main.c
cc -o mkespfsimage main.o -lz
make[1]: Leaving directory '/home/kayo/esp-link/espfs/mkespfsimage'
Compression assets with htmlcompressor. This may take a while...
Compression assets with yui-compressor. This may take a while...
wifi/icons.png   (100%, none,  428 bytes)
wifi/wifi.js     ( 36%, gzip, 1676 bytes)
wifi/wifi.html   ( 36%, gzip, 1148 bytes)
favicon.ico      (100%, none,  500 bytes)
ui.js            ( 37%, gzip, 2860 bytes)
style.css        ( 31%, gzip, 2264 bytes)
pure.css         ( 22%, gzip, 3820 bytes)
console.js       ( 46%, gzip,  980 bytes)
log.html         ( 44%, gzip,  788 bytes)
console.html     ( 37%, gzip, 1160 bytes)
home.html        ( 37%, gzip, 1952 bytes)
20 -rw-r--r-- 1 kayo kayo 17912 Nov 21 21:43 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
/home/kayo/esp-open-sdk/esp_iot_sdk_v1.4.0_pure/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x598): undefined reference to `user_init'
/home/kayo/esp-open-sdk/esp_iot_sdk_v1.4.0_pure/lib/libmain.a(app_main.o): In function `user_rf_pre_init':
(.irom0.text+0x804): undefined reference to `user_init'
collect2: error: ld returned 1 exit status
Makefile:282: recipe for target 'build/httpd.user1.out' failed
make: *** [build/httpd.user1.out] Error 1
```

When I simply commented `MODULES ?=` entry in `Makefile`, it has another fault:

```
kayo@debi:~/esp-link$ make
VERSION: esp-link master - 2015-11-21 21:51:54 - development
CC espfs/espfs.c
CC httpd/httpdespfs.c
CC httpd/auth.c
CC httpd/base64.c
CC httpd/httpd.c
CC user/user_main.c
CC serial/crc16.c
CC serial/uart.c
CC serial/console.c
CC serial/slip.c
CC serial/serled.c
CC serial/serbridge.c
CC cmd/handlers.c
CC cmd/cmd.c
CC esp-link/cgimqtt.c
CC esp-link/mqtt_client.c
CC esp-link/cgiflash.c
CC esp-link/config.c
CC esp-link/log.c
CC esp-link/cgioptiboot.c
CC esp-link/cgi.c
CC esp-link/main.c
esp-link/main.c: In function 'cgiSystemInfo':
esp-link/main.c:124:7: error: implicit declaration of function 'mqttState' [-Werror=implicit-function-declaration]
       mqttState(), flashConfig.baud_rate, flashConfig.sys_descr
       ^
esp-link/main.c:125:7: error: format '%s' expects argument of type 'char *', but argument 12 has type 'int' [-Werror=format=]
       );
       ^
cc1: all warnings being treated as errors
Makefile:428: recipe for target 'build/esp-link/main.o' failed
make: *** [build/esp-link/main.o] Error 1
```

tve, 2015-11-21:
What is the reason you're trying to build without those modules? Brunnels added the ability to disable modules when he was trying to fit esp-link into 240KB, but that's no longer possible as far as I can tell. I have not maintained the #ifdefs necessary, so I'm not really surprised that it's failing. It's a small problem, most likely...

katyo, 2015-11-21:
Suppose, I tried to add my own js addon to my specific build and I needed some bit of free space.
Of course, it isn't a big problem. I can fix this, but I don't sure, should I offer pr for this?

tve, 2015-11-22:
I believe esp-link currently uses about 290KB out of 480KB available. If you need more space than that, I doubt the removal of the mqtt stuff is going to do it. But maybe I'm mis-understanding...

tve, 2015-12-31:
I don't see anything to fix here. Fitting the firmware into 240KB is not supported anymore, I can't maintain too many versions. Sorry. If I'm not understanding what you're trying to do, please reopen.

closed 2015-12-31 by tve