jeelabs/esp-link #75: Error compiling " expected '_Bool *' but argument is of type 'uint8_t *'"

arjenhiemstra, 2015-11-24:
I get the following message(s), any idea why?

esp-link/cgimqtt.c:88:7: error: passing argument 3 of 'getBoolArg' from incompatible pointer type [-Werror]
       &flashConfig.mqtt_clean_session);
       ^
In file included from esp-link/cgimqtt.c:4:0:
esp-link/cgi.h:21:8: note: expected '_Bool *' but argument is of type 'uint8_t *'

tve, 2015-11-25:
Looks like you're using a different compiler than I do?
```
xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2
```

arjenhiemstra, 2015-11-25:
vagrant@esp8266:~/dev/test-esp-link/esp-link$ xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2

Looks the same

arjenhiemstra, 2015-11-25:
Ok, changing the following in config.h 
uint8_t  slip_enable, [...]_session;    
to
bool  slip_enable, [...]_session;    

fixed the error

stakinnovations, 2015-12-18:
I also had to change these:

  bool      syslog_filter;                      // min. severity
  uint8_t  syslog_showtick,              // show system tick (µs)
              syslog_showdate;             // populate SYSLOG date field
  bool     mdns_enable;

deece, 2016-04-11:
This should be fixed by pull request 134

closed 2016-06-19 by tve