jeelabs/esp-link #76: Bug/Request/Question Set boot up serial baud rate

TLMNicolas, 2015-11-30:
Ive been doing some tests to create a decently reliable water temp monitor for a hobby project i have going. Its going to be outdoors so I want it to be reliable - and if it fails to fail gracefully.

One of the tests I do is kill the power and start it up again. (Ive found a few problems but i can deal with them with delays and re-tries) however....
It appears that the baud rate defaults to 57600 on power up of the ESP module. I need 115200
Is this auto detected based on the bootloader message or is this a default ?

If its a default, could i request a change
- allow picking default boot up baud rate
or
- use the last used baud rate at boot up

Ive tried looking thru the code for an answer to my own question but haven't yet found the right file.

Thanks

TLMNicolas, 2015-11-30:
So i finally found where the UART is initialized,
user_init():
  uart_init(flashConfig.baud_rate, 115200);

And it looks like it already is supposed to use the flashConfig setting.

Do you think because im killing the power to the ESP, its not saving the baud rate to the flashConfig store?

tve, 2015-11-30:
The baud rate is saved to flash as soon as you change it. Make sure you see the green confirmation at the top of the page after you change the drop-down value. I have not seen problems in this respect. Are you finding that the other settings, such as the pin assignment (on the home page) stick across reboots? If not, maybe there's some problem. Do I assume correctly that you are using v2.1.beta6?

closed 2015-11-30 by tve

TLMNicolas, 2015-12-01:
Yes im on 2.1b6

I tried what you suggested, saving other settings and non stick across power down.
It must be the ESP. The wiring or the unit itself.
Ill try and diagnose it tonight.

Thanks for this code base....also I LOVED reading your jeelabs blog the last few years. It became like a ritual for me reading the new article every day.

TLMNicolas, 2015-12-01:
Id like to say that after a full re-flash of the ESP firmware, the settings are getting saved now. 
Sorry to have wasted your time.