jeelabs/esp-link #77: can't build «eagle.app.flash.bin» does not exists

camiloavil, 2015-12-04:
I tried to do "make" but after some steps
just after ls -ls eagle*bin

does not found eagle.app.flash.bin i dont know about that file

tve, 2015-12-04:
It would help a lot if you would paste the output of the build process here in the ticket...
Are you using SDK 1.4? and which version of esp-link have you downloaded or cloned?

camiloavil, 2015-12-05:
yes didn't because its in spanish but this I just download it again and now its other error

Yes i haver SDK 1.4 and I when you ask about esp-link you mean "xtensa-lx106-elf-objdump" right?

I think the error is because i used a different SDK than "pfalcon/esp-open-sdk" when I "make it" with my SDK i get this error

20 -rw-rw-r-- 1 camilo camilo 20372 dic  4 22:54 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
/home/camilo/git/tools/ESP8266_SDK/lib/liblwip.a(sntp.o): In function `sntp_get_current_timestamp':
/home/liuhan/software/espconn/1.4.0_release/esp_iot/app/lwip/core/sntp.c:622: multiple definition of `__tzyear'
/home/camilo/git/tools/ESP8266_SDK/lib/libc.a(tzset_r.o):(.bss+0xc): first defined here
/home/camilo/git/tools/ESP8266_SDK/lib/liblwip.a(sntp.o): In function `sntp_get_current_timestamp':
/home/liuhan/software/espconn/1.4.0_release/esp_iot/app/lwip/core/sntp.c:622: multiple definition of `__tznorth'
/home/camilo/git/tools/ESP8266_SDK/lib/libc.a(tzset_r.o):(.data+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:309: recipe for target 'build/httpd.user1.out' failed
make: *** [build/httpd.user1.out] Error 1

But after that I "make it" with the SDK "pfalcon/esp-open-sdk" and finish correctly. But maybe I get wrong the "pfalcon/esp-open-sdk" because when at first I "make it" whit this get the errors dont find files .h etc

I going to find the differences between the SDKs

tks a lot

katast, 2015-12-14:
I have got the same error, any clue?

VERSION: esp-link master - 2015-12-14 11:38:06 - development
CC rest/rest.c
CC syslog/syslog.c
CC mqtt/mqtt.c
CC mqtt/mqtt_cmd.c
CC mqtt/mqtt_msg.c
CC mqtt/pktbuf.c
CC espfs/espfs.c
CC httpd/auth.c
CC httpd/base64.c
CC httpd/httpd.c
CC httpd/httpdespfs.c
CC user/user_main.c
CC serial/console.c
CC serial/crc16.c
CC serial/serbridge.c
CC serial/serled.c
CC serial/slip.c
CC serial/uart.c
CC cmd/cmd.c
CC cmd/handlers.c
CC esp-link/cgi.c
CC esp-link/cgiflash.c
CC esp-link/cgimqtt.c
CC esp-link/cgioptiboot.c
CC esp-link/cgipins.c
CC esp-link/cgitcp.c
CC esp-link/cgiwifi.c
CC esp-link/config.c
CC esp-link/log.c
CC esp-link/main.c
CC esp-link/mqtt_client.c
CC esp-link/status.c
CC esp-link/task.c
make[1]: Entering directory `/opt/ESPRESSIF/esp-link/espfs/mkespfsimage'
cc -I.. -std=gnu99 -DESPFS_GZIP   -c -o main.o main.c
cc -o mkespfsimage main.o -lz
make[1]: Leaving directory `/opt/ESPRESSIF/esp-link/espfs/mkespfsimage'
Compression assets with htmlcompressor. This may take a while...
Compression assets with yui-compressor. This may take a while...
ui.js            ( 37%, gzip, 2860 bytes)
console.js       ( 46%, gzip,  980 bytes)
pure.css         ( 22%, gzip, 3820 bytes)
mqtt.html        ( 36%, gzip, 1168 bytes)
log.html         ( 44%, gzip,  788 bytes)
style.css        ( 31%, gzip, 2264 bytes)
wifi/wifi.js     ( 36%, gzip, 1680 bytes)
wifi/wifi.html   ( 36%, gzip, 1148 bytes)
wifi/icons.png   (100%, none,  428 bytes)
home.html        ( 35%, gzip, 2064 bytes)
console.html     ( 37%, gzip, 1160 bytes)
favicon.ico      (100%, none,  876 bytes)
mqtt.js          ( 41%, gzip,  748 bytes)
20 -rw-rw-r-- 1 luxor luxor 20372 Dec 14 11:38 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
Dump : /opt/ESPRESSIF/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -x build/httpd.user1.out
Disass: /opt/ESPRESSIF/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -d -l -x build/httpd.user1.out
FW firmware
ls -ls eagle*bin
  4 -rwxrwxr-x 1 luxor luxor   2937 Dec 14 11:38 eagle.app.v6.data.bin
244 -rwxrwxr-x 1 luxor luxor 249320 Dec 14 11:38 eagle.app.v6.irom0text.bin
 16 -rwxrwxr-x 1 luxor luxor  12732 Dec 14 11:38 eagle.app.v6.rodata.bin
 32 -rwxrwxr-x 1 luxor luxor  31808 Dec 14 11:38 eagle.app.v6.text.bin
Usage: gen_appbin.py eagle.app.out boot_mode flash_mode flash_clk_div flash_size_map
mv: cannot stat ‘eagle.app.flash.bin’: No such file or directory
make: *** [firmware/user1.bin] Error 1

camiloavil, 2015-12-15:
Yes, of course.

I think you are using another toolchain than "https://github.com/pfalcon/esp-open-sdk".

I did make it but with some issues, 
First make it with your toolchain you will have that error. but after that make it with the "https://github.com/pfalcon/esp-open-sdk" toolchain and then you have your *.bin files

that works for me for now. but unfortunately I dont have so much time to fix the makefile.

I guess is the way that generates the *.bin files maybe in the esptool.py i dont now for sure.

Hope u can do it and fix it

tve, 2015-12-15:
The master branch now uses SDK 1.5 (since a day ago). On linux, please be sure to use pfalcon's esp-open-sdk or you're on your own...

tve, 2015-12-31:
I believe this is resolved? If not, I'd need more details about the problem.

closed 2015-12-31 by tve