jeelabs/esp-link #78: Trouble connecting to WRT54G

appsxt, 2015-12-07:
Hi,

Thanks for a really interesting project, which I found while looking for an ESP8266 drop-in replacement to the Arduino Ethernet library (to share as much application code for a wifi version of an ethernet project).

This is mainly some testing feedback using Release 2.1.beta6 that couldn't connect to an old Linksys WRT54G (setup with WPA Pre-Shared Key, AES with Group Key Renewal: 3600).

With SSID Broadcast disabled (hidden), esp-link continuously reports no_ap_found and with SSID Broadcast enabled, esp-link cycles between auth_expire, group_key_update_timeout, no_ap_found repeatedly. Using the original firmware and AT commands, connection was possible.

esp-link had no trouble to connect to another AP I had sitting on the shelf for testing (with open access). Built-in MQTT is great and the self reporting is working well.

Cheers,

Mick

appsxt, 2015-12-16:
Just reporting back...

I finally upgraded the WRT54G (v2) firmware to OpenWrt and now everything is working fine. The ESP-01 with esp-link 2.1.beta6 connects without problem, also with ESSID hidden.

This upgrade (from Linksys to OpenWrt firmware) has been a long time coming. I don't know why I waited so long. Now even the MacBook Air can connect :-)

closed 2015-12-16 by appsxt

tve, 2015-12-16:
I'm glad it worked and thanks for reporting back!