jeelabs/esp-link #82: Looses flash

thjubeck, 2015-12-20:
Hey, I have an issue where if I disconnect the esp from power it looses flash and the wifi connection shows "hidden network".

tve, 2015-12-21:
Which version are you using? What size flash chip does your esp8266 module have? How did you flash it or what is it saying on the home page about the flash size and flash chip? The last time someone had a similar sounding problem they had not passed the right -ff / -fs parameters to esptool.py when flashing.

thjubeck, 2015-12-21:
esp-link v2.1.beta6. 4MB:512/512. Using NODEMCU Firmware Programmer.

thjubeck, 2015-12-21:
Re flashed with Linux and the esptool and it seems to work. Any ideas on how to make it work in windows?

tve, 2015-12-31:
Use esptool.py under windows, or use the official Espressif flasher. I think the nodemcu flasher doesn't allow you to set the flash size correctly.

closed 2015-12-31 by tve