jeelabs/esp-link #86: help troubleshooting reboots (Fatal exception 0)

TLMNicolas, 2016-01-01:
using a nodemcu devboard (to remove any possibility of mis wiring on my part). 
flashed the esp with 2.1.7 (tried two different utilities). SLIP enabled, MQTT disabled.
Wifi is setup to a WPA2 access point.

it randomly will reboot with nothing attached (except debug uart).
Its a fatal exception, but dont know how to figure it out.
It is always the same " epc1=0x4010711a"

driving myself crazy trying to figure this out.

this is the output when it resets
```
 51290> 3fff2550 already freed
                                                                                               
 53748> 3fff2550 already freed
                                                                                               
 59958> Fatal exception 0(IllegalInstructionCause):
                                                                          
 59958> epc1=0x4010711a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

                            
                                                                                                                               
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

                                                                              
                                                                                                                               
load 0x40100000, len 1396, room 16 
                                                                                          
tail 4
                                                                                                                       
chksum 0x89
                                                                                                                  
load 0x3ffe8000, len 776, room 4 
                                                                                            
tail 4
                                                                                                                       
chksum 0xe8
                                                                                                                  
load 0x3ffe8308, len 540, room 4 
                                                                                            
tail 8
                                                                                                                       
chksum 0xc0
                                                                                                                  
csum 0xc0

                                                                                                                  
                                                                                                                               
2nd boot version : 1.4(b1)
                                                                                                   
  SPI Speed      : 40MHz
                                                                                                     
  SPI Mode       : QIO
                                                                                                       
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
                                                                                  
jump to run user1 @ 1000
                                                                                                  
don't use rtc mem data
```
            
Im not sure if its a bad flash, or if its a bad setting. possibly a bug in the code.
any help greatly appreciated. thanks

tve, 2016-01-01:
can you confirm that you downloaded the release 2.1.7 and are running that unmodified? And that the epc1=0x4010711a corresponds to that build? The "3fff2550 already freed" is worrysome, it says that os_free is called on a block that has already been freed. The last time I saw that was when I tried mDNS.

TLMNicolas, 2016-01-01:
stil trying things, i had first tried 2.1beta6 but subsequently downloaded a fresh 2.1.7 from github and made sure to flash a blank bin to all the locations i could find, then flashed the 2.1.7.
the epc1 is still the same as before. always that value.

the process i have done
flash 2.1.7
[image]

set up the wifi. wpa2 (ssid is of the form <something>_<somethingelse>, password is 16 characters long)
then load up the "home" via the new ip and in the terminal for the debug output, monitor the output.
within 1-30 seconds itll reset with that epc1 value.

im used to the "already freed warnings" from the first time i started trying esp-link they existed for me.
Didnt think much of it.

I spent the last 3 hours venturing down the rabbit hole of detecting when the esp resets and re-initiing all the items in the arduino, but that proved problematic in its own ways.

I have done a fresh re-flash and captured a dump of the debug log. see attached.
[fresh_log.txt](https://github.com/jeelabs/esp-link/files/76025/fresh_log.txt)

(flash files, apply wifi settings, reset esp, listen to debug port)
 

What else can I provide to help you help me debug this reset?
Any specific process you want me to try?

tve, 2016-01-01:
Please give the new 2.2.beta0 a try. it uses SDK1.5 and there were a number of fixes https://github.com/jeelabs/esp-link/releases/tag/v2.2.beta0. The crash is in the unsigned modulo library function, not sure what to think of that... Could be divide-by zero, could be a bogus jump. Unfortunately there's no stack backtrace, that makes it hard to determine where it's really happening. :-(

TLMNicolas, 2016-01-01:
does 2.2b0 need a boot_1.5.bin ?

TLMNicolas, 2016-01-01:
Got 2.2b0 flashed. Have not been able to make it reset no matter how much i fiddle with the web ui, or blow on it or touch it. Also the "already freed" errors are gone.

Makes me think if the "fix" was in the SDK or in the esp-link code.

Is the modulo library something thats part of esp-link or the SDK ?
I wonder if somehow the ESP's i have, or the way iam flashing them leaves some memory corrupted or missing. I was reading a webpage that mentioned with newer SDK's that the flashing tools would write big user program over the boundary and back to 0x0000 causing some overwriting.

Now ill have to go diving into more code but at least you gave me a big reprieve from this madness knowing at least something works properly. Thanks!

One final question for now, is this 2.2b0 indeed break the 2.1.7 SLIP compatibility, or is that in the notes as a fore-warning for 2.2 ?

tve, 2016-01-01:
I suspect the fix was in the SDK, but it's hard to tell. The udiv function is part of the GCC runtime. I really doubt that the problem has to do with your flashing, more likely your wifi or network environment contains something that triggers the failure.
2.2.b0 SLIP should be compatible with 2.1.7, I messed-up a commit and didn't include those changes...

minida28, 2016-01-02:
Hi TLMNicolas, I think I am using the same NodeMCU model with yours. I bought it used, so I don't know exactly what model it is, but I think it is the same with yours (with ESP-12 model). 
[image]
I was using native NodeMCU flasher when flashing, this is my settings:
[image]
[image]
Probably worth a try?
And also if you have other different power supply to power the ESP-8266 it also worth a try, if you have 5V, you should feed it it from 5V pins, if you have 12V, you should feed it from Vin. I am not good at coding, so I can only try to give my views on hardware side. Cheers.

TLMNicolas, 2016-01-02:
Well I tried every flash combination i could think of, and files I could find.
Even got esptool.py working on my machine.
I think for the error above, it is my network in some way. No matter what flash methods I tried, same problem happened for me with 2.1.7 (and almost briked the unit at one point).

I did learn that the speed at which you flash, does matter. After doing a FULL flash_erase in esptool.py I had lots of stability problems with just the 3 files (2.2b0) being flashed. I narrowed it down to luck + the speed at which I programmed the chip. Possibly there is no write verification ?
I also had to re-flash the AT base first then the esplink flash file in once instance - take from it what you will. Either way I was able to get back to a stable 2.2b0.

2.2b0 is working VERY stably for me now. SLIP is working great with the espduino library. MQTT + REST both working at the same time RELIABLY. 

Only problem is I cant seem to remotely flash the arduino anymore.
I used to use the http methods to sync and upload the .hex with 2.1b6 (and 2.1.7), but no matter what I cant seem to get it to sync anymore with 2.2b0 - not even manually reseting the arduino.

Thanks tve for guiding me in the right direction. and Minida28 thanks for the settings.

tve, 2016-01-02:
Mhh, thanks for the report. I'm not 100% sure what to make of it all... When you say that you can't flash the AVR anymore with v2.2b0 can you please paste the output of the debug console after a failed attempt? I made quite some improvements, but there are so many flavors of optiboot and reset circuitry out there that any change is always a bit a shot in the dark.

TLMNicolas, 2016-01-02:
Clicking "Reset uC" in the UI does reset the arduino (rest line to proper default pin by level shifter).
Data from the arduino does get to the serial console, and responses from the ESP do get to the arduino so tx/rx is working.

Some attempts to use a port forwarding tool, sometimes yield some progress to rpogram but never complete. In the past the http method was 100% and quick.

Ive tried manualy setting the console to 57600. Ive tried to reset the arduino manually just before and after.
 
The resets go fast, i wonder if there is not enough time waiting for the response from optiboot ?
If your changes are checked in, ill go have a peek to see if I can spot any differences that stand out.

Here is the debug port output after I send the request to sync.

73253> MCU reset gpio12
                                                      
 73794> UART 115200 baud
                                                      
 73794> MCU reset gpio12
                                                      
 74336> UART 115200 baud
                                                      
 74336> MCU reset gpio12
                                                      
 74877> UART 9600 baud
                                                        
 74878> MCU reset gpio12
                                                      
 75419> UART 57600 baud
                                                       
 75419> MCU reset gpio12
                                                      
 75960> UART 115200 baud
                                                      
 75961> MCU reset gpio12
                                                      
 76502> UART 115200 baud
                                                      
 76502> MCU reset gpio12
                                                      
 76967> OB sync abandoned after timeout, state=0 syncCnt=62
                   
 76967> UART 115200 baud

TLMNicolas, 2016-01-02:
I noticed line 530 of cgioptiboot.c was changed to .
 //uart0_write_char(STK_GET_SYNC);
and replaced with
uart0_write_char(CRC_EOP);

Is that character (STK_GET_SYNC) possibly required to be sent ?
From the very tiny tiny little I know (looked into how avrdude works), I think optiboot expects STK_GET_SYNC  + CRC_EOP (0x30 0x20) to trigger sync.

For quick ref, the define is as such
#define CRC_EOP             0x20  // 'SPACE'
#define STK_GET_SYNC        0x30  // '0'

minida28, 2016-01-02:
Hi mates, I also have flashed my nodemcu to 2.2beta0 firmware, and I think I can confirm I can still flashed the attached uC (I am using Arduino UNO rev3 - clone)...
I used windows version of avrdude, when I fireup the avrdude command for the first time, it seem did not work (fail to sync after 10 times retry). But the moment later I run exactly the same command it works right away.
If I use the linux version of avrdude, it never fail...

minida28, 2016-01-02:
Below is the debug log:
```
997853> Accept port 23, conn=0x3fff7140, pool slot 0
998156> MCU Reset=gpio12 ISP=gpio14
998943> serbridge: connection reset err=-9
999842> Accept port 23, conn=0x3fff7140, pool slot 0
   143> MCU Reset=gpio12 ISP=gpio14
   930> serbridge: connection reset err=-9
  2089> Accept port 23, conn=0x3fff7140, pool slot 0
  2391> MCU Reset=gpio12 ISP=gpio14
  3176> serbridge: connection reset err=-9
  4219> Accept port 23, conn=0x3fff7140, pool slot 0
  4521> MCU Reset=gpio12 ISP=gpio14
  5307> serbridge: connection reset err=-9
  6574> Accept port 23, conn=0x3fff7140, pool slot 0
  6876> MCU Reset=gpio12 ISP=gpio14
```
You can see I have to fireup the avrdude command several times before it is finally works.. (using windows version of avrdude)

minida28, 2016-01-02:
I use this command for my UNO rev3 clone:
```
avrdude -DV -patmega328p -Pnet:esp-link2.lan:23 -carduino -b115200 -U flash:w:EmonLib_voltage_and_current_Modified_MQTT_r1.cpp.standard.hex:i
```
If there is anything else you want me to share which can help, please let me know...

tve, 2016-01-02:
@minida28 Doe avrflash work for you too? The use of avrdude depends a lot on network latency and other semi-random attributes of each person's local set-up. That's why I ended up building the programming into esp-link.

@TLMNicolas You are correct that the sync uses STK_GET_SYNC + CRC_EOP and that esp-link initially only sends CRC_EOP pairs. Optiboot doesn't actually care and responds to either. The reason I changed to CRC_EOP pairs is that it avoids failed syncs if the AVR uart register overruns. I guess I should document that in the README...

What you should see:
- a brief reset pulse (1ms)
- CRC_EOP+CRC_EOP sent ~50ms later
- CRC_EOP+CRC_EOP repeated every ~70-80ms
- eventually optiboot responds with STK_INSYNC+STK_OK
- esp-link sends one CRC_EOP
- either optiboot responds with STK_INSYNC+STK_OK or nothing happens for 70-80ms
- esp-link sends STK_GET_SYNC+CRC_EOP and optiboot responds with STK_INSYNC+STK_OK and we're in sync now
- esp-link sends the next command (starts with 'u')
- if no sync is achieved, esp-link changes baud rate and the whole thing starts over with a reset pulse about 600ms, esp-link gives up after about 5 seconds

WRT your debug console output, it looks to me like reset is not working: esp-link does not get any response ever

minida28, 2016-01-02:
@tve 
avrflash does not work...
This is my debug output - similar with TLMNicolas:
```
247336> UART 115200 baud
247336> OB init
247336> UART 115200 baud
247336> MCU reset gpio12
247851> UART 9600 baud
247851> MCU reset gpio12
248392> UART 57600 baud
248393> MCU reset gpio12
248934> UART 115200 baud
248934> MCU reset gpio12
249476> UART 115200 baud
249476> MCU reset gpio12
250017> UART 9600 baud
250018> MCU reset gpio12
250560> UART 57600 baud
250560> MCU reset gpio12
251101> UART 115200 baud
251102> MCU reset gpio12
251643> UART 115200 baud
251643> MCU reset gpio12
252108> OB sync abandoned after timeout, state=0 syncCnt=62
252108> UART 115200 baud
252108> OB init
252147> OB cannot sync
```
I've tried changing the baud rate on the esp-link GUI (tried 9600, 57600, 115200). And I think my reset line is okay since I can reset the attached uC trough esp-link GUI just fine - using GPIO12 (pin D6 on nodemcu). I used a voltage divider (1k resistor connected to arduino reset pin) as the logic level shifter.

tve, 2016-01-02:
grrr, thanks for the report. I need to improve the error reporting, I think...

TLMNicolas, 2016-01-03:
Im SUPER happy to report the esplink with v2.2b0 is running 24 hours now without failing once.
Its MQTT publishing every 30 seconds, and REST posting as well. 
It seems to have an issue with trying to access the uC console now, debug port shows:
```
951154> E:M 1584                               
951155> E:M 1584                              
951155> E:M 1584                                
951155> E:M 1584                            
951155> E:M 1584                               
951155> E:M 1584                                    
951155> E:M 1584    
```
But at this point im happy its running and sending data. This is all for a hobby project for my wifes rabbits (so its purely a love affair with tinkering and the frustration level was getting up there)

Should i make a new issue for the flashing problem in v2.2b0 ?

PS. Tve do you take donations ?

tve, 2016-01-03:
I'm glad it's working! :-)
The E:M message is concerning, that says that it's out of memory and can't allocate 1584 bytes. It comes from malloc directly. I'm a bit surprised and wonder what's consuming the memory. If you look in the debug log, you'll see all http requests end with something like `h=8156`, that says that there are 8156 bytes of heap free. It should be >10k and if it drops below about 3k you're in the "red zone". I wonder whether you can see a pattern. It's possible that some rest or mqtt functionality doesn't free up memory somewhere. So it would be interesting to see whether it drops suddenly when you open connections, or whether it erodes away little by little.
Re: donations, I appreciate, but so far it's not necessary! The best you can do is post about esp-link in some public forum :-).

TLMNicolas, 2016-01-03:
After your explanation of the sync process, I went and tried to send the 0x20 0x20 on reset to another arduino i have here (from the same batch) and it doesnt go into bootloader.
However when I send 0x30 0x20 it responds instantly.
So the bootloader I have, must be different than the one you are using, and most likely is too ridig to the spec or assumptions of what is expected.

My bootloader reports version 1.16 (is this even optiboot??)
```
0x81 => 14 01 10 
0x82 => 14 10 10  (0x10 = 16)
```

I did look into the optiboot [source](https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot.c#L805) and the latest version there shows that it actually expects an EOP (space). So iam more confused now.

TLMNicolas, 2016-01-03:
Im going to seperate this out into two different issues, and close this one.
One for the flashign problem, and one for the 951154> E:M 1584  one.

closed 2016-01-03 by tve