jeelabs/esp-link #87: support Arduino Mega / stk500v2 protocol

tve, 2016-01-02:
Currently none of the flashing options work for the mega with the stk500v2 protocol. It's not clear why nothing works, so some logic analyzer work is required...

jcw, 2016-01-02:
Different baud rate, perhaps?

ykri, 2016-01-03:
Might help your investigation, I tried this with Mega:
1. Installed MCU bootloader -> uploaded OTA with esp-link -> SUCCESS!!
2. uploaded OTA with esp-link -> FAILURE!!

i.e. it uploads ok after a fresh bootloader but then any subsequent upload fails.
It's like the first upload meshes up with the bootloader???
Any subsequent upload can succeed only by manually pressing MCU RESET and only at port 23.

Xerxes3rd, 2016-03-15:
This sounds like an electrical problem- like the board isn't resetting at all or isn't resetting when the software thinks it is.  See this post on the arduino forums for more information, noting that the symptoms described in the post are very similar to this issue (resetting and programming works fine with a 328/Uno but not a 2560/Mega):

http://forum.arduino.cc/index.php?topic=310109.0

bestbuddyrohit, 2016-08-30:
hi,
Any update with atmega2560. i also experienced the same. i i press the reset button i have a bit chance that mcu is programmed. but yes if the chip is freshly loaded with the boot loader, it will program at once without external reset required. after that same behaviour.

dannybackx, 2016-12-07:
This issue is bothering me. Can anyone tell me where to look to debug/fix this ?

tve, 2016-12-08:
I think the first step is probably to read up which char sequence starts the programming and add it to esp-link: https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L169-L182
The second step is probably to look at the TX and RX lines with a scope and see what the timing looks like, I hope the Mega responds with something similar to the way the 328p does....

janchrillesen, 2016-12-08:
One thing I noticed is that I can't program my Mega2560 over USB when the ESP8266 is connected. I have two Mega2560 with ESP8266/ESP-link connected to a prototype cape. 

One board is using a simple voltage divider on the Mega2560 TX line towards the ESP8266 and TX of the ESP8266 directly connected to the Mega2560 RX. On this cape I need to remove the ESP8266 to program using USB
The other board is using "real" (MOSFET based) levelshifters on both RX and TX lines and on this board I need to remove the entire cape - so it seems like there may be some hardware level stuff stopping programming from working. 

Does anyone else see this issue?

dannybackx, 2016-12-11:
@tve If I understand this correctly, the Mega programmer sends a sequence which esplink needs to intercept, put the Mega in programming mode, and only then get avrdude and the Mega to talk to each other ?

When I run 
  avrdude -patmega2560 -cwiring -t -Pnet:esplink:23 -x snooze=1 -v -v -v -v
it says [...]
avrdude: Send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout

So I'm guessing I need this sequence in serial/serbridge.c like
        || (len == 7 && data[0] == 0x1b && data[1] == 0x01 && data[2] == 0x00
        && data[3] == 0x01 && data[4] == 0x0e && data[5] == 0x01 && data[6] == 0x14)
and then I should see the reset sequence being triggered as
 59746> SERBRIDGE Accept port 23, conn=0x3fff9828, pool slot 0, ix 0, mode 0
 59747> SERBRIDGE Accept conns 0x3fff9828 0x00000000 0x00000000 0x00000000
 59801> SERBRIDGE Accept ports 0x3fff6fb4 0x3fff6fd0
 60050> MCU Reset=gpio5 ISP=gpio-1

tve, 2016-12-11:
If the reset sync sequence is really that long, yes. You could also 
start with just matching the first 3 chars, for example.

FernandoGarcia, 2016-12-21:
Hi @tve !

I don't know if it's possible or how to make but I would like to give this suggestion.

I have tested this library with an Arduino nano and worked very well. I'll try with Arduino Mega later.

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

Maybe you could join this library to ESP-link to make possible programming via ISP instead UART.

With ISP is possible burn hex file to any type of Arduino.

Best regards.

FernandoGarcia, 2016-12-21:
With Arduino mega 2560 doesn't work using CMD on Windows but worked using AVRDUDESS.

[image]

zrc98, 2016-12-23:
Hi.
On MacOS doesn't work neither from terminal, nor with avrdudess(using with mono).
If mega2560 connected to usb avrdude/avrdudess works fine.
I tried to use v.2.2.3 and v.3.0.14 of esp-link.
Need help(

FernandoGarcia, 2016-12-23:
You needs check my suggestion again.
AVRDUDESS can't be used with ESP-link for now.
You should install this code directly on ESP8266 WITHOUT ESP-link.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266AVRISP/examples/Arduino_Wifi_AVRISP/Arduino_Wifi_AVRISP.ino

FernandoGarcia, 2016-12-23:
Follow my tutorial to see how to update the sketch using this library

https://youtu.be/Ll35jVvetA8

FernandoGarcia, 2016-12-23:
It worked using the IDE directly too with command ctrl + shift + u and programmer in Arduino as ISP.

dannybackx, 2016-12-24:
Thanks @FernandoGarcia  I'll have a look at this.

zrc98, 2016-12-25:
Thank you !!
It could be a temporary solution)
Hope one day esp-link will support mega clones.

CharlyBrok, 2017-01-19:
I did this script as a workaround :                                                            
#!/bin/bash

CMD="avrdude -p m2560 -c stk500v2 -P net:$1:2323 -U flash:w:$2 -D -v"
CMD23="avrdude -p m2560 -c stk500v2 -P net:$1:23 -U flash:w:$2 -D -v"

timeout 1 $CMD

$CMD23
echo "Done"
exit

Usage : ./flash_mega.sh esp-e62067 /tmp/arduino_build_495949/TFT_Test_Mega_3.ino.hex

rnizametdinov, 2017-01-29:
More elegant workaround for windows and mega-esplink.
Make bat file with next:

> curl --data '' http://192.168.2.92/console/reset
> SET ADDRESS=192.168.2.92
> SET BAUD=115200
> SET HEXFILE=d:\YandexDisk\SmartHome\Arduino\test\test\test.ino\test.ino.ino.mega.hex
> set prog=D:\Software\arduino-1.8.1\hardware\tools\avr/bin/avrdude
> set conf=D:\Software\arduino-1.8.1\hardware\tools\avr/etc/avrdude.conf
> :GOTOWAN
> 
> %prog% -C%conf% -v -p atmega2560 -b %BAUD% -c wiring -P net:%ADDRESS%:23 -D -Uflash:w:%HEXFILE%:i
> 
curl --data '' http://192.168.2.92/console/reset - will reset Arduino Mega

uzi18, 2017-02-02:
reset with curl and launch avrdude works form me on linux.

uzi18, 2017-02-02:
It is possible for me to ivestigate it with logic analyser if you want

dannybackx, 2017-02-04:
FYI a first cut of esp-link with mega flashing support is downloadable from http://danny.backx.info/danny/backup/esp-link-danny-20170204064017.tgz and the script at http://danny.backx.info/danny/backup/megaflash . This is a branch of 3.0alpha (by fuzzball03 and me) which contains stuff to change the telnet ports etc and to secure them. The UI of esp-link is protected with userid "admin" password "s3cr3t" by default. Feedback welcome.

uzi18, 2017-02-04:
> 20170204064017.tgz and the script at http://danny.backx.info/danny/
> backup/megaflash . This is a branch of 3.0alpha (by fuzzball03 and me)

Any patch?

dannybackx, 2017-02-04:
I'll see what I can do.
This is my third piece of development in esp-link, neither of which is in the main source tree yet, so I'll have to see how to do this without making (more of) a mess.

uzi18, 2017-02-04:
Try to attach patch for upsteam i will try it and test

dannybackx, 2017-02-05:
I've uploaded my code to https://github.com/dannybackx/esp-link.git/trunk .
This is without the serial-bridge/security changes that were in the binary version. So this is TVE's 3.0alpha with just a few changes.

uzi18, 2017-02-06:
@dannybackx does not work for me, used Your binary from tgz archive

$ bash ../esp-link/megaflash -v 192.168.2.6 sterownik_ac_plus_dc_fan.hex 
Resetting AVR with http://192.168.2.6/pgmmega/sync
*   Trying 192.168.2.6...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> POST /pgmmega/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 204 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
* Curl_http_done: called premature == 0
* Closing connection 0
*   Trying 192.168.2.6...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Curl_http_done: called premature == 0
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.2.6...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Curl_http_done: called premature == 0
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.2.6...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [64 bytes data]
* Curl_http_done: called premature == 0
* Closing connection 0
AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Sending HEX file for programming
*   Trying 192.168.2.6...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> POST /pgmmega/upload HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> Content-Length: 235208
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
> 
* Done waiting for 100-continue
} [16384 bytes data]
* Operation timed out after 10000 milliseconds with 0 bytes received
* Curl_http_done: called premature == 1
* stopped the pause stream!
* Closing connection 0

Error programming AVR

uzi18, 2017-02-06:
But smaller hex file works ok:

* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> POST /pgmmega/upload HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.52.1
> Accept: */*
> Content-Length: 100474
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
> 
* Done waiting for 100-continue
} [16384 bytes data]
* We are completely uploaded and fine
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [66 bytes data]
* Curl_http_done: called premature == 0
* Closing connection 0
Success. 37378 bytes at 115200 baud in 8.0s, 4657B/s 40% efficient
Success, took 10 seconds

dannybackx, 2017-02-06:
Can you share the big file so I can try it ?
I've had trouble with lagging in the version I distributed as binary, there might be issues with that version. The source version might not have that.
So I'd like your big file to see if I can use it to narrow things down and/or find other problems with my code.

uzi18, 2017-02-06:
https://github.com/uzi18/sterownik/blob/master/arduino/sterownik_ac_plus_dc_fan.hex

wallorchid, 2017-02-07:
Compiled dannybackx's code using the supplied docker config.  Loaded in ESP8266MOD AI-THINKER module.  Using avrdude to load got same error as std release of eep-link (...ioctl("TIOCMGET"): Device not configured...avrdude: stk500v2_ReceiveMessage(): timeout).  Verified that rx/tx comm with 2560 is working within console of esp-link web ui.  Reset also verified to work as expected from web ui.

Is danny... forked code only suppose to work with the http post method? vs using telnet 23 or 2323?

I am not clear what boot loader my Mega2560 boards are using, they are Osqyoo that appear identical to the branded Arduino boards except for logo.  Several Mega2560 boot loaders exist.

Researching boot loaders it would appear that 2560 boot loaders use stk500v2 because of the memory size of the 2560.  A fork of OptBoot, https://github.com/majekw/optiboot claims to be modified to work with the 2560 using what I understand to be stk500v1  (note version).  Presumably the existing stock release of esp-link works with OptiBoot.  Has anyone tried this path?

uzi18, 2017-02-07:
Need to mention - my "big" hex  into flash by Arduino IDE and serial connection without problems.
So it's not related to bootloader.

wallorchid, 2017-02-07:
My understanding is that the developers of esp-link had to spend time figuring out how to sit as a man in the middle to avrdude and the micro.  This detail work was done with std optiboot that uses stk500v1.  Avrdude understands both v1 and v2.  But esp-link only understands how to fake out std optiboot running stk500v1.

So in theory an atmega2560 running optiboot w/stk500v1 ought to work just fine.  I likely will try it but wanted to know if my underlying assumptions are correct.

If esp-link can be made to support stk500v2 great, but my first try with solution did not work.

dannybackx, 2017-02-07:
@wallorchid : your assumption is close to the truth. A module added to esp-link actually replaces avrdude. TVE already implemented this for stk500, I just added stk500v2.

dannybackx, 2017-02-07:
The stk500v2 (Mega) implementation currently duplicates the method offered previously, also known as the "http post" method. Use e.g. the megaflash script provided. It uses curl to talk to esp-link.
You should see something like this : 
`
acer: {1941} megaflash testesp ~/src/sketchbook/mega-esp/kippen-mega/tmp/obj/kippen.hex
AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Success. 30920 bytes at 115200 baud in 5.5s, 5536B/s 48% efficient
Success, took 7 seconds
acer: {1942} 
`

uzi18, 2017-02-07:
Where could be a problem with my big hex?  Did You tasted it?

dannybackx, 2017-02-08:
I am looking into it.

Lincolnshirenick, 2017-02-09:
@dannybackx, I have hardware and a project so would be happy to BETA test potential solutions for the Mega OTA flash problem as this is something that I need.

I'm in the process of developing a project on an Arduino Mega2560 and would like to be able to flash OTA. I have two Megas and as far as I'm aware they are both optiboot but neither will program with the stock ESP-LINK, I've tried every method that has so far been published here except for FernandoGarcias as I'm not keen on the need for all the other tool installs that are required and the loss of some of the ESP-LINK functionality. I've not currently tried yours as it seemed from this thread that it didn't initially work and that your are investigating.

One hardware point, I note that I have to disconnect the TX from the ESP to flash by USB and I'm wondering if the same issue is occurring for the ESP from the USB TX.

uzi18, 2017-02-09:
To work together with usb programmer and esp-link, you need to add
rectifier diode in direcion from arduino pin to esp-link on tx.
Works like charm.

In my 2560 project we use RPi-usb connection to remote programming but
esp-link is more lightweight.
I will be glad to resolve this issue.

Lincolnshirenick, 2017-02-09:
@uzi18 I had thought about a diode but my point was more along the lines of if the ESP TX upsets the USB then the USB might also be upsetting the ESP so there would be an argument for fitting a diode for both devices (the USB path has a 1k resistor).

dannybackx, 2017-02-09:
@uzi18 @Lincolnshirenick it is my experience also that you can't program the Mega via USB if it's connected to an ESP in a way suitable for esp-link.
The solution that I'm working on doesn't have that : it is the serial connection between ESP and Mega that is used to program the Mega.
So if you move a setup to a production environment (in my case, to the chicken house in my garden), it should be possible to reprogram the Mega without access to the hardware.

uzi18, 2017-02-09:
Did You checked my  hex file?

dannybackx, 2017-02-11:
This is not my day job so it takes a little bit of time :-)
The error message is caused by : 
1. timeout too short in the script (easy to bypass)
2. a command in the hex file that isn't interpreted yet. See the short line below : 
   :10FFE000C6010E949F4CC6010E946C4EA6960FB699
   :10FFF000F894DEBF0FBECDBFDF91CF911F910F915F
   :020000021000EC
   :10000000FF90EF90DF90CF90BF90AF909F908F9038
   :100010007F90089586E590E10E940649882319F0B3
  This is not unexpected. The Mega contains more memory so the special command is probably to move to the next piece of address space.

I'll deal with this shortly.

dannybackx, 2017-02-11:
LOL, just ignoring that line gets this file to flash correctly.
```
acer: {1962} megaflash testesp ~/sterownik_ac_plus_dc_fan.hex
AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Success. 22260 bytes at 115200 baud in 15.7s, 1411B/s 12% efficient
Success, took 17 seconds
acer: {1963} 
```
This is not a general solution obviously.
But the program runs : 
```
Lucjan v.0.1.0.109

I2C: Skanowanie ...
RTC: brak DS1307
DALLAS: Poczatek poszukiwania czujnikow.

 Czas komenda konwersji dla czujnika = 2
DALLAS: Koniec poszukiwania czujnikow.

SD: brak karty
SD: blad karty
SD: blad odczytu danych
IP: konfiguracja sieci...
Wolny RAM:4015
IP: brak EtherShield...
START
```

dannybackx, 2017-02-11:
You can grab a new binary version at http://danny.backx.info/danny/backup/esp-link-danny-20170211161419.tgz and http://danny.backx.info/danny/backup/megaflash .
The source code repo at https://github.com/dannybackx/esp-link.git is also up to date.

Note : this is a proof of concept version, the segment address record is basically ignored.
Let me know if this works.

uzi18, 2017-02-11:
> backup/esp-link-danny-20170211161419.tgz and

Ok I will check new version.
Maybe we need to send some special command to bootloader on this short
line.

Lincolnshirenick, 2017-02-11:
@dannybackx I've flashed an ESP with your bins but I have the same problem with the sync timing out. Just one quick check, I've managed to flash a micro without problem with the standard ESP-LINK, but is this the correct command line for a Mega: 
avrdude -v patmega2560 -cwiring -Pnet<net address>:23 -b115200 -D -U flash:w:mega_code.ino.hex:i -vvvvv

uzi18, 2017-02-11:
> avrdude -v patmega2560 -cwiring -Pnet:23 -b115200 -D -U

Use megaflash provided by Danny not avrdude.

Lincolnshirenick, 2017-02-11:
Thanks uzi18, I've tried it with megaflash but it just sits in a permanant waiting for sync loop with the following output:
Resetting AVR with http://192.168.1.239/pgmmega/sync
* Hostname was NOT found in DNS cache
*   Trying 192.168.1.239...
* Connected to 192.168.1.239 (192.168.1.239) port 80 (#0)
> POST /pgmmega/sync HTTP/1.1
> User-Agent: curl/7.35.0
> Host: 192.168.1.239
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 204 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
* Closing connection 0
* Hostname was NOT found in DNS cache
*   Trying 192.168.1.239...
* Connected to 192.168.1.239 (192.168.1.239) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> User-Agent: curl/7.35.0
> Host: 192.168.1.239
> Accept: */*
> 
... repeats without first line

uzi18, 2017-02-12:
@dannybackx new firmware works for me but ...
programming took 18 s. with limit 20 ... so if we have mode data to flash timeout must be longer,
maybe need to send Keep Alive header ?
other ... in the end got msg:
 Success. 22166 bytes at 115200 baud in 15.8s, 1401B/s 12% efficient 
 Success, took 17 seconds
but sketch bin is:
 Sketch uses 87,702 bytes (34%) of program storage space. Maximum is 253,952 bytes.

dannybackx, 2017-02-12:
@uzi18 I removed the timeout options from the megaflash script, also a variable didn't have the right definition to cope with >64K program size, resulting in the number of bytes being off like you showed.
Now I see : 
```
acer: {2005} /home/danny/src/github/jeelabs/mine/esp-link.git/trunk/megaflash testesp ~/sterownik_ac_plus_dc_fan.hex
AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Success. 87796 bytes at 115200 baud in 15.6s, 5596B/s 48% efficient
Success, took 18 seconds
```
Source updates are sent to github. Do you use the sources or a binary ?

Lincolnshirenick, 2017-02-12:
@uzi18 as you are make some progress and as I seem to be completely stuck and trying to understand why, I wonder if you could help a little. I am guessing that in my case it must be related to my bootloader or possibly the type of ESP so could you answer a couple of questions:
What bootloader do you have on your Mega?
How do I detect what bootloader I have?
What ESP are you using?
In my case I have had my megas for several years and they have the bootloader they came with, I have never had any cause to change this. I am using an ESP8266-01.

uzi18, 2017-02-12:
@dannybackx i'm using Your binaries, as my SDK building process stopped at some point ...
I will check if flashed hex and chip dump are same.

@Lincolnshirenick: I flashed bootloader 1-1,5y ago so dont remember exactly - but it is fixed optiboot for 2560.
Just flash fixed bootloader, If You want have hex with all fixes.
ESP-12E - "big" PCB with CH340  usb-ttl converter and goldpins on sides.

janchrillesen, 2017-02-13:
I am seeing the same behavior as uzi18 where megaflash gets stuck in the loop connecting to /pgmmega/sync. I am using the esp-link-danny-20170211161419.tgz binary and reset of the Mega board works fine through the web UI. The file I'm trying to flash is RFLink.cpp.hex from https://drive.google.com/file/d/0BwEYW5Q6bg_ZcGVES2JadGdvc2M/view
When I start megaflash the blue LED on the ESP12 flashes irregularly approx 2 times a second. It continous after stopping megaflash. I need to reboot the Mega afterwards to talk to it

dannybackx, 2017-02-13:
This one is bigger than the one from @uzi18. 
```
acer: {17} ./megaflash testesp RFLink.cpp.hex
AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Success. 170450 bytes at 115200 baud in 30.4s, 5591B/s 48% efficient
Success, took 32 seconds
```
Browsing its website for a bit shows me 57600 baud remarks. After setting the esplink to that, I get a message : 
```
20;00;Nodo RadioFrequencyLink - RFLink Gateway V1.1 - R45;
```

dannybackx, 2017-02-13:
I've done much of this work with a separate test hardware. I was preparing a setup with an Arduino Mega, a Motor shield (cheap ripoff of the old Adafruit design) and a bunch of sensors, to automate my chicken house door.

When the setup became too complicated, I ordered a new esp and a new Mega. If pasting the picture works, they're on the left. The other stuff is meant to be the production environment, and I've not been able to get my code to work on it yet. Looking into it...

[image]

uzi18, 2017-02-13:
@dannybackx : No img.
Will try full test to night.

dannybackx, 2017-02-13:
Don't worry about the image. Just stuff on the table next to me. 
I just uploaded the binary that goes with the latest changes above.
http://danny.backx.info/danny/backup/esplink-testing-20170131203642.tar.gz
http://danny.backx.info/danny/backup/megaflash-20170213203015
http://danny.backx.info/danny/backup/IMG_20170213_174715696.jpg

The last is that picture that didn't get uploaded on this forum.

uzi18, 2017-02-13:
Nice work @dannybackx 
Must be sure - programming with esp-link work perfect, because my project is used by some people already.
I want to use esp-link as website with configuration etc.+serial bridge as ethernet shield is very sloooow on SPI.

Lincolnshirenick, 2017-02-15:
@uzi18 thanks for your help, I have interface on order for flashing my boot, according to the github optiboot thread the bootloader version shipped in the Arduino tools is the one to use for Mega2560. Will let everyone know the outcome.

[58 more replies]

closed 2017-06-18 by tve