jeelabs/esp-link #88: problem getting sync with http based flasher in v2.2b0

TLMNicolas, 2016-01-03:
As per issue #86 there appears to be a problem with the http flashing of AVRs (stk500) in then ew v2.2b0 firmware.

The arduino's I have and another user have donot want to grab sync.

TLMNicolas, 2016-01-03:
To recap, myself and minida28 were unable to get sync with v2.2b0.
There were some changes to the http flasher in 2.2b0.

From some basic tests on my arduino, it may be related to the way the sync is requested that changed (from 0x30 0x020 to 0x20 0x20) and the arduino's bootloader not accepting that change (it specifically wants 0x30 0x20).

TLMNicolas, 2016-01-04:
I took the plunge and setup a VM to build esp-link. You setup the build enironment so perfectly. Making changes and pushing them to the ESP is super simple. Great Job!

Im able to report the flashing problem is related to the updates to the sync changes done to cgioptiboot.c

I had to revert back to the  STK_GET_SYNC and also had to undo the stateSync2/3 code for it to work.

tve, 2016-01-06:
Sounds like I have work to do... Thanks for checking things out!

codewise-nicolas, 2016-04-20:
Hello tve. Im looking forward to upgrading to the newer 2.2 releases but wanted to know if you had left the AVR flashing routines alone or reverted to the SYNC+EOP ?
(Checked the optiboot.c and it looks like the EOP+EOP is still in place, which doesnt work with my arduino's bootloader)

tve, 2016-04-20:
Not sure what you're looking at, is https://github.com/jeelabs/esp-link/blob/master/esp-link/cgioptiboot.c#L513-L518 not what you need?

codewise-nicolas, 2016-04-20:
Oh I'm so sorry,  that is correct.

I wonder if I was looking at an old commit of that file.
Again my apologies for wasting your time.

I'll give the new version a try.

tve, 2016-06-19:
The programming code was revisited for 2.2.beta2, please reopen if you see further issues.

closed 2016-06-19 by tve