jeelabs/esp-link #89: memory issue running out

TLMNicolas, 2016-01-03:
This is to keep track of the problem with the memory issue after the esplink has veeb running for some time and doing REST+MQTT.
With v2.2b0 of the firmware.

Error happens when loading web ui:
`951154> E:M 1584 `

TLMNicolas, 2016-01-03:
The memory issue could be that Im re-setting up the REST every time i go and do a REST request.
I do this in case the esp resets, the arduino will re-setup the rest and then make the request.
I was doing this when I was having issues with 2.7 as a stop gap for whenever the esp would reset the arduino could still do its REST request.

As requested tve here is the log from a REST request:

```
659947> CMD_parse_packet: cmd=11(REST_SETUP) argc=3 cb=0x00000237 ret=1
659948> CMD_Exec: Dispatching cmd=REST_SETUP
659949> REST: setup #1 host=api.thingspeak.com port=80 security=0
9955> CMD_Exec: Response: 0xbeef0001, cmd: 11
                                
660038> CMD_parse_packet: cmd=12(REST_REQUEST) argc=3 cb=0x00000000 ret=0     
660038> CMD_Exec: Dispatching cmd=REST_REQUEST                                
660040> REST: request #1 method=GET path=/update?api_key=<XXXXX>&field1=4935&field2=84697229&field3=22.31&field4=21.31 bodyLen=0 hdrLen=239 totLen=239 data=0x3fffa410 hdrLen=226                    
660056> REST: pCon state=0                                                    
660059> REST: Connect to host api.thingspeak.com:80                           
660064> CMD_Exec: no response (0)                                             
660081> REST DNS: found ip 144.212.80.11                                      
660082> REST: connecting...                                                   
660223> REST #1: connected                                                    
660224> REST #1: sending 226                                                  
660534> REST: Sent                                                            
660535> REST: status=200, body=5                                         
```

I dont see any `h=` anywhere in that output.

TLMNicolas, 2016-01-03:
I enabled the MQTT status reporting setting and reset the hardware.
Im seeing slowly deminishing heap size. So it is what you suggested tve that its not freeing up memory somewhere.

From newest to oldest
```
2016-01-03 01:45:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-43, "heap_free":10936}
2016-01-03 01:44:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-30, "heap_free":11000}
2016-01-03 01:43:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-31, "heap_free":11080}
2016-01-03 01:42:14
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-29, "heap_free":11128}
2016-01-03 01:42:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-29, "heap_free":11128}
2016-01-03 01:41:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-29, "heap_free":11192}
2016-01-03 01:40:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-29, "heap_free":11272}
2016-01-03 01:39:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-31, "heap_free":11320}
2016-01-03 01:38:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-24, "heap_free":11384}
2016-01-03 01:37:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-37, "heap_free":11464}
2016-01-03 01:36:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-35, "heap_free":11528}
2016-01-03 01:35:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-38, "heap_free":11592}
2016-01-03 01:34:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-27, "heap_free":11656}
2016-01-03 01:33:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-29, "heap_free":11704}
2016-01-03 01:32:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-38, "heap_free":11768}
2016-01-03 01:31:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-27, "heap_free":12032}
2016-01-03 01:30:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-27, "heap_free":11728}
2016-01-03 01:29:10
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-34, "heap_free":11944}
2016-01-03 01:28:14
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-37, "heap_free":12512}
```

TLMNicolas, 2016-01-03:
Another status update on the heap size from the mqtt auto push:
```
2016-01-03 06:28:55
Topic: /topic/esplife/2234
Qos: 1
{"rssi":-31, "heap_free":8672}
```
Keeps going down slowly.

tve, 2016-01-03:
looks like a bug indeed!

TLMNicolas, 2016-01-04:
The memory leak is indeed in the re-call of the REST.begin()

If i only call `rest.begin()` once in `setup()` and never again, the heap is very stable in size.
However, if i `rest.begin()` every time I want to make a `rest.get()` call then the heap slowly decreases and never comes back up.

Simple fix: dont call rest.begin() every time. Right ?
Well there is a bug where if for some reason the ESP resets, and you try to do a `rest.get()` without first having setup rest, then the ESP crashes and reboots. As you can imagine a nasty reboot cycle ensues.

Possible solutions (short of fixing the memory leak):
- A means of checking if REST has been setup
- In the rest.setup() dont do whatever it does if its already been setup.So that we can call it over and over with no worries.
- A check in the rest.get() if rest has not been setup error out and dont crash.

Let me know what you think

TLMNicolas, 2016-01-06:
@tve if you make a design decision as to which of the possible options above you would prefer to follow I will try and implement a solution.

tve, 2016-01-06:
I have to admit I've been swamped at work and have not looked at the code myself. First of all, esp-link needs to be robust and not crash if the arduino is misbehaving.
I'm trying to remove as much of this init stuff as possible. There need to be a call from the arduino side to remove any existing callbacks, which needs to be called at the beginning of the arduino sketch. This prevents old registrations from being invoked. There also need to be calls to get the wifi status.
I've started with some of this in https://github.com/jeelabs/el-client but I have not been able to get and end-to-end test to work, so I haven't pushed my latest round of changes. I'm hoping to make some progress before the end of the week. Dunno how that fits your schedule.