jeelabs/esp-link #92: mbed-compatible board

andriyadi, 2016-01-05:
Just wondering, can this project be used to flash/update firmware mbed-compatible boards?
Please advice.
Thanks.

tve, 2016-01-06:
I believe "mbed compatible board" is too vague a description. esp-link is a serial-wifi bridge, so if you can program your ARM processor over its UART then you can most likely program it using esp-link. For example, I know it works with NXP's LPC series. Some STM32 processors can also be programmed using the UART, but I have not tried. Esp-link is not an ST-Link programmer (which uses JTAG or SWD).

andriyadi, 2016-01-12:
@tve apology for late response.
I'm using STM32 Nucleo F401RE and code for it using mbed, instead of its native framework. I'll try it at the soonest.
Thanks.

ShaneKirkbride, 2016-06-06:
@andriyadi do you have a link to how you are programming the STM32 (I have a Nucleo F446RE) I'm trying to do just this but I don't want to reinvent the wheel. :-)

tve, 2016-06-18:
Closing ticket: I don't have an answer and there's nothing actionable for me. Feel free to reopen.

closed 2016-06-18 by tve