jeelabs/esp-link #94: Boot error after flashing on linux

gozotto, 2016-01-06:
I try to flash my NodeMCU DevBoard with esp-link release under linux.
(fs, ff and fm adapted to NodeMCU flash-chip)

esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -ff 40m -fm dio 0x0 0x00000 boot_v1.4\(b1\).bin 0x010000 user1.bin 0x3FE000 blank.bin
No error from esptool.py.

After boot on 76800 baud I've got the boot error:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 1396, room 16
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

backup boot failed.

user code done

And nothing on 115200.
Gone back to windows and use the NodeMcu-Flash-Tool to flash user1.bin and all is working again.
What's going wrong?

tve, 2016-01-06:
Your command line is pretty messed-up with 'dio' and an extra '0x0'

gozotto, 2016-01-06:
Sorry.. c&p error 
The second 0x0 can be ignored.
Corrected line (produce the error):
esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -ff 40m -fm dio 0x010000 user1.bin
The dio is for the SPI Flash mode.
See also: https://github.com/themadinventor/esptool/issues/79
for esptool.py.

tve, 2016-01-06:
You have the wrong address for user1.bin, it has to be 0x1000

gozotto, 2016-01-07:
Oh man... it was too late... reading this line so many times... and nothing seen.
That's the error and it works with esptool.py under linux.
The error was sitting in front of the screen.

closed 2016-01-07 by gozotto