jeelabs/esp-link #98: Set Parity?

AlphaLima, 2016-01-21:
Is there a way to the the sirial protocol to "even" parity?
Thx
 Ulli

fl1pfl0p, 2016-02-02:
An option to change data bits, parity bit, stop bit would be very useful. I have at least 2 projects that have this requirement, both with different settings. Editing the source and compiling for each can be a pain.

leodesigner, 2016-05-19:
Also it's needed for STM32 mcu serial boot loader. (STM32 uses serial connection with Even parity).

tve, 2016-10-23:
parity is supported on master...

closed 2016-10-23 by tve

zalexua, 2023-06-06:
But ODD parity left unsupported ...

I see some implementation in C code:
```
serial/console.c 
    if (buff[1] == 'N') flashConfig.parity = NONE_BITS;
    if (buff[1] == 'E') flashConfig.parity = EVEN_BITS;
    if (buff[1] == 'O') flashConfig.parity = ODD_BITS;
    if (buff[2] == '1') flashConfig.stop_bits = ONE_STOP_BIT;
    if (buff[2] == '2') flashConfig.stop_bits = TWO_STOP_BIT;
```

but not in HTML:
```
html/console.html
        <select id="fmt-sel" class="pure-button" href="#">
          <option value="8N1">8N1</option>
          <option value="8E1">8E1</option>
          <option value="8N2">8N2</option>
          <option value="8E2">8E2</option>
          <option value="7N1">7N1</option>
          <option value="7E1">7E1</option>
          <option value="7N2">7N2</option>
          <option value="7E2">7E2</option>
```

Even if I send proper HTTP POST data by shell command:
`curl http://10.33.0.42/console/fmt?fmt=8O1 -X 'POST'`
it responds with post data `{"fmt": "8N1"}` and use NONE parity.

Looks like definition here is not complete:
```
include/uart_hw.h
typedef enum {
    NONE_BITS = 0,
    ODD_BITS   = 0,
    EVEN_BITS = BIT4
} UartParityMode;
```