jeelabs/esp-link #100: Page reload problems under Windows 8 (esp-link 2.2beta1)

mi-kal, 2016-02-10:
On Windows, the menu and cards load much slower than under Linux Mint (e.g. 5 seconds) and, under special circumstances, never load: When pages such as "WiFi AP" are forcibly re-loaded with the reload button of Firefox 44.0.1 (+ shift to bypass the cache), the first card "AP state" loads after a few seconds, but the other spinning circle ("AP Settings" card) never goes aways and the associated card never loads. However, the spinning circle in the tab of Firefox goes away, so the browser seems to think it has loaded the page - or it timed out. If I then click on "WiFi AP" in the menu to reload the same page, it loads immediately - so the content apparently loaded and was cached, it just didn't display (?). The "Services" page is worst: it re-loads the menu but none of the cards. The home page doesn't have this issue. If the user starts there (with an empty browser cache) and goes to the other tabs, all work. Nothing like this happens under Linux with Firefox, so I guess it's not a Javascript issue. Could this be solved with Javascript: detect when the browser thinks that the page loaded (or gave up) and reload it, now with the browser cache filled? I tried limiting the connections via Firefox's about:config (network.http.max-connections, network.websocket.max-connections) to 4 or 8 but that had no effect.

Problem 2: As soon as I connect to the ESP8266 AP, my normal web browsing under Windows gets slow because there are now long timeouts before web pages load. This problem disappears if I disable automatic retrieval of the DNS server in the WiFi connection's TCP/IPv4 settings. Could this be solved in the esp-link firmware (let computers know there is no DNS server?)

Problem 3 (minor): Couldnt get the host name to work in any situation, only the IP works. Not in AP mode, not in STA mode and not with browsers on Windows 8, Linux Mint and a chomebook. Tried with and withuth mDNS, whatever that is. However, the host name shows up in my router's menu next to the MAC.

mi-kal, 2016-02-10:
The reload problem completely disappeard as soon as I switched from STA+AP to AP.

mi-kal, 2016-02-11:
When this happens - card(s) not showing up after reload - the debug log shows

 41858> E:M 1584
 41858> E:M 1584
 41858> E:M 1584
 41859> E:M 1584
 41859> E:M 1584
 41859> E:M 1584
 41859> E:M 1584

I googled that to mean out of memory. Doesn't happen wandering from page to page, just when reloading a page. Only under Windows. Only in STA+AP mode. As far as I can tell so far, always 7 messages of this type (regardless of page). All fancy features (SLIP, MQTT, mDNS, SNTP) off.

minida28, 2016-02-11:
I think this is the same with #89 ?

openSoftMan, 2016-02-13:
I was experiencing the same issues as you describe i.e. pages stopped loading, "E:M 1584" errors, and the telnet port would stop.  The connection led would show I was still connected and the router wifi status page confirmed that.  My issues were OS/browser though.

I wasn't using any other features either; I use just the bridge (STA only).  Just a quick question, when you get these errors and without restarting the ESP, can you go into your router and restart wifi?  ...do the pages back?

I was using a WRT-350n with the classic 08-12-10-r14929 dd-wrt firmware.  I have since switched out my router to an E3000, running the latest K3.x build and my problems went away.

mi-kal, 2016-02-13:
> when you get these errors and without restarting the ESP, can you go into your router and restart wifi? ...do the pages back?

Not sure what you mean with restart wifi in the router. I'm in STA+AP mode but I'm not connected to a router - which is off. It's just slow and has reload problems in AP+STA mode but I don't loose the connection between PC and ESP. If the page is partly blank after a non-cached reload, I click on the URL and press return and the page shows up fine.

I've in the meantime changed the esp link source code to disconnect after 15s to stop the endless reconnect attempts of the ESP SDK firmware. I'm on the Debug Log page, I reboot the ESP and after ca. 15s, the Wifi status is "idle" and the mode is still 3 (STA+AP):

 11424> no ALAN-ABCDEF found, reconnect after 1s
 11425> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)
 11525> reconnect
 11525> f -180, scandone
 13651> no ALAN-ABCDEF found, reconnect after 1s
 13652> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)
 13752> reconnect
 13652> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)
 13752> reconnect
 13752> f r-180, Wifi check: mode=AP+STA status=3
 15344> Wifi disconnected, state 3

The debug log shows little further output after that, only:

 15344> scandone
 15345> Enabling/continuing uart log
 57742> chg_A3:0
 88642> chg_A3:-180
104092> chg_A3:0
119542> chg_A3:-180
150442> chg_A3:0
165893> chg_A3:-180
181342> chg_A3:0

Until I shift-reload the debug log page. Then it throws E:M 1584 errors.

296165> E:M 1584
296165> E:M 1584
296165> E:M 1584
296165> E:M 1584
296165> E:M 1584
296165> E:M 1584
296166> E:M 1584

mi-kal, 2016-02-13:
Log with heap (printed every second), Windows 8:

 25357> Heap: 10424
 26357> Heap: 10424
 27357> Heap: 10424
 28357> Heap: 10424
 29357> Heap: 10424
 30357> Heap: 10424
 31358> Heap: 10424
 32357> Heap: 10424
 33357> Heap: 10136
--> shift-reloading "Debug Log" page under Windows now
 33539> E:M 1584
 33539> E:M 1584
 33540> E:M 1584
 33540> E:M 1584
 33540> E:M 1584
 33540> E:M 1584
 33540> E:M 1584
 34357> Heap: 9512
 35357> Heap: 9512
 36357> Heap: 9512
 37357> Heap: 9512
 38357> Heap: 10016

This goes on for a while but nothing happems until, several seconds later, the menu shows up. The other dynamic content does not appear (no log entries in log form). 

Booting into Linux and repeating the shift-reloading test:

370357> Heap: 10424
371357> Heap: 10424
372357> Heap: 10424
373357> Heap: 10424
374357> Heap: 10424
375357> Heap: 10424
376357> Heap: 10424
377357> Heap: 10424
--> shift-reloading "Debug Log" page under Linux now
378357> Heap: 9344
379357> Heap: 10008
380357> Heap: 10008
381357> Heap: 10008
382357> Heap: 10008
383357> Heap: 10008
384357> Heap: 10632

tve, 2016-06-18:
Looks like in Windows 8 the browser you're using is opening 7 or 8 connections simultaneously and the esp-8266 runs out of memory trying to serve them at the same time. I need to look into what it takes to limit the number of concurrent connections...

minida28, 2016-06-19:
I remember I was experiencing similar issues (slow loading of esp web menu, sometimes it won't reload at all). I thought it was caused by bad esp unit or poor wifi connections or low serial buffer settings. Tried everything I knew but never really cure the problem. Unfortunately I never really look in detail what was going on in the debug log console.

But when I was playing with UART debug log settings in the esp-link GUI, an then setting it to OFF, I would say I never experience those issues anymore... (or at least very rare to occur??).

So maybe it's worth a try.. try settings the debug log to completely OFF..?

I also wonder, can the issue above be related to this (from El-Client mqtt example file):
```C++
// Initialize a connection to esp-link using the normal hardware serial port both for
// SLIP and for debug messages.

ELClient esp(&Serial, &Serial);
```
Can using one port for both  SLIP and debug messages cause this; causing simultaneous connection within the browser?

tve, 2016-06-20:
Using one port normally works OK, but you have to be careful with the sequencing, i.e., if you output debug info while constructing a packet it's going to garble stuff.

The issue of running out of memory with 7 concurrent connections may be fixed with ef6a611. I'm going to close this ticket for now, feel free to reopen if it recurs.

closed 2016-06-20 by tve