jeelabs/esp-link #102: h/w CTS flow control (input GPIO13)

jeco1964, 2016-02-11:
In many cases it is necessary to slowdown the dataflow from esp module to an attached slow serial device (specially when it is in serialbridge mode). The ESP hardware is supposed to support this issue ( CTS input -->GPIO13) if it is enabled and configured (as active High or Low). Can u pls integrate this feature in this excellent  project, if it is possible ???
It will be very useful, using the serialbridge mode, to make older serial devices (printers, plotters etc) wifi aware!!!

FrankBoesing, 2016-10-14:
Is RTS/CTS FlowControl implemented now ?

tve, 2016-10-14:
this ticket wouldn't be open if it had ;-)

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve

linker3000, 2018-08-30:
Came here to ask about this - I'd like to use an ESP8266/esp-link with a Z80 board that requires hardware handshake. Any chance this could be implemented - at least an input to the ESP8266 so that the host device can control inbound data flow?

Thanks

wolfgangr, 2019-02-04:
I quoted this problem of buffer overrun
https://github.com/jeelabs/esp-link/issues/430#issuecomment-460062159

to Mecrisp-Stellaris-Wizard Matthias Koch and got the answer:

> hardware handshaking with RTS/CTS is a very nice solution, too. 
> Terry did that for STM32F051.

so I'd add my 3 pence on the nice-to-have pile.
But I hope that interrupt-buffered RX will do the job as well.