jeelabs/esp-link #104: Flood of reconnects makes fall-back to AP unusable

mi-kal, 2016-02-13:
2. When set up in STA mode to connect to a router and router is not accessible, esp-link will fall back to AP+STA after 15s (as intended) but the ESP firmware will make endless reconnect attempts, every 1 seconds, that make the AP essentially unusable (not as intended). Same issue descibed here: http://bbs.espressif.com/viewtopic.php?t=671. Log:

211593> scandone
211593> no ALAN-ABCDEF found, reconnect after 1s
211594> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)
211694> reconnect
211694> f -180, scandone
213820> no ALAN-ABCDEF found, reconnect after 1s
213821> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)
213921> reconnect
213921> f r-1, scandone
216047> no ALAN-ABCDEF found, reconnect after 1s
216048> Wifi disconnected from ssid ALAN-ABCDEF, reason no_ap_found (201)

It's difficult or impossible to connect to the AP if the module is in that state. wiflash attempts (if one mistakenly assumes that the module is corrupted) fail mid-point. What about stopping these reconnect attempts afer 15s in resetTimerCb()?

static void ICACHE_FLASH_ATTR resetTimerCb(void *arg)
{
    int x = wifi_station_get_connect_status();
    int m = wifi_get_opmode() & 0x3;

    DBG("Wifi check: mode=%s status=%d\n", wifiMode[m], x);

    // not in AP mode?

    if (m != 2)
    {
        if (x == STATION_GOT_IP)
        {
            // not in STA mode?

            if (m != 1)
            {
                #ifdef CHANGE_TO_STA
                    // we're happily connected, go to STA mode
                    DBG("Wifi got IP. Going into STA mode..\n");
                    wifi_set_opmode(1);
                    os_timer_arm(&resetTimer, RESET_TIMEOUT, 0); // check one more time after switching to STA-only
                #endif
            }

            // might have to turn UART logging off now (connected)

            log_uart(false);

            // no more resetTimer after this point, gotta use physical reset to recover if in trouble
        }
        else
        {
            // not STA+AP?

            if (m != 3)
            {
                DBG("Wifi connect failed. Going into STA+AP mode..\n");

                wifi_set_opmode(3);
                wifi_softap_set_config(&apconf);
            }

            if ((x == STATION_WRONG_PASSWORD) || (x == STATION_NO_AP_FOUND))
            {
                // disconnect to stop firmware from endless reconnect attempts

                DBG("Wifi disconnected, state %d\n", x);

                wifi_station_disconnect();
            }
            else
            {
                // continue waiting for connect, to go to STA at some point

                os_timer_arm(&resetTimer, RESET_TIMEOUT, 0);
            }

            // might have to turn UART logging on now (disconnected)

            log_uart(true);

            DBG("Enabling/continuing uart log\n");
        }
    }
}

2. After a mode change, the firmware sometimes displays "Error changing mode" even though there was no error, just loss of connection. What about making that warning sound less like an error?" "Error changing mode or connection lost (IP may have changed)"?

tve, 2016-02-16:
Thanks for flagging this. I was hoping not to add a reconnect attempt timer with back-off myself, but it sounds like I have to...

tve, 2016-06-19:
I cannot reproduce the problem using the latest version (master beyond 2.2.b2). Perhaps there is an improvement in the new SDK... What I did: set to STA+AP mode, change STA assoc to SSID "foobar", see reconnect attempts every second (not great..), connect via AP, web pages work fine, rebooted using debug log page button, came up again with wrong STA, could reconnect fine via AP, finally fixed incorrect STA setting. I also switch to STA-only mode, changed the SSID to foobar, then restarted using power-cycle, came up in STA+AP and stayed there until I connected to the AP and fixed the STA setting.
Please reopen if you see this issue again.

closed 2016-06-19 by tve