jeelabs/esp-link #105: Problem while Compiling @ Win

stockklauser, 2016-02-13:
Hello, 

i have enabled all settings from the Windows.md file. But everytime i get the following error 

`C:\Espressif\esp-link>make all
VERSION: esp-link master - 2016-02-13 23:38:43 - development
make[1]: Entering directory `/c/Espressif/esp-link/espfs/mkespfsimage'
make -C mman-win32
make[2]: Entering directory `/c/Espressif/esp-link/espfs/mkespfsimage/mman-win32'
gcc -o mman.o -c mman.c -Wall -O3 -fomit-frame-pointer
ar cru libmman.a mman.o
ranlib libmman.a
make[2]: Leaving directory `/c/Espressif/esp-link/espfs/mkespfsimage/mman-win32'
gcc -c -I.. -Imman-win32 -std=gnu99 -DESPFS_GZIP -o main.o main.c
gcc -o mkespfsimage.exe main.o -Lmman-win32 -lmman  -lz
make[1]: Leaving directory `/c/Espressif/esp-link/espfs/mkespfsimage'
Compression assets with htmlcompressor. This may take a while...
Compression assets with yui-compressor. This may take a while...
Zugriff verweigert - HTML_COMPRESSED
Datei -TYPE nicht gefunden
Datei F nicht gefunden
Datei -NAME nicht gefunden
Zugriff verweigert - HTML_COMPRESSED
Datei -TYPE nicht gefunden
Datei F nicht gefunden
Datei -NAME nicht gefunden
Zugriff verweigert - HTML_COMPRESSED
Datei -TYPE nicht gefunden
Datei F nicht gefunden
Datei -NAME nicht gefunden
Zugriff verweigert - .
Datei ! nicht gefunden
Datei -NAME nicht gefunden
1 -rw-r--r-- 1 thoma Administratoren 16 Feb 13 23:38 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
Dump : /c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -x build/httpd.user1.out
Disass: /c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump -d -l -x build/httpd.user1.out
FW firmware
ls -ls eagle*bin
  3 -rw-r--r-- 1 thoma Administratoren   3013 Feb 13 23:39 eagle.app.v6.data.bin
226 -rw-r--r-- 1 thoma Administratoren 230820 Feb 13 23:39 eagle.app.v6.irom0text.bin
 13 -rw-r--r-- 1 thoma Administratoren  13248 Feb 13 23:39 eagle.app.v6.rodata.bin
 31 -rw-r--r-- 1 thoma Administratoren  31232 Feb 13 23:38 eagle.app.v6.text.bin
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `COMPILE=gcc PATH=/c/Espressif/xtensa-lx106-elf/bin/:/c/ProgramData/Oracle/Java/javapath:/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/c/Program Files (x86)/Microsoft SQL Server/110/Tools/Binn/:/c/Program Files/Microsoft SQL Server/110/Tools/Binn/:/c/Program Files/Microsoft SQL Server/110/DTS/Binn/:/c/Users/thoma/.dnx/bin:/c/Program Files/Microsoft DNX/Dnvm/:/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit/:/c/Program Files/IDM Computer Solutions/UltraEdit:/c/Program Files/Git/cmd:/c/Program Files (x86)/Atmel/sam-ba_2.15/drv/:/c/Program Files (x86)/Atmel/sam-ba_2.15:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/MinGW/bin:/usr/bin:/c/espressif/git-bin:/c/espressif/java-bin:/c/Python27 python /c/Espressif/ESP8266_SDK/tools/gen_appbin.py build/httpd.user1.out 2 0        15       4        0'
make: *** [firmware/user1.bin] Error 2`

stockklauser, 2016-02-18:
Problem is solved! Thank you! 

It was required to edit the Makefile and some other Files

closed 2016-02-18 by stockklauser

dmitrijkulinich, 2017-01-06:
Hi! Can you tell me please what exactly files must be edited and what to edit?
I have the same problem:
`/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `COMPILE=gcc PATH=/c/Espressif/esp-open-sdk/xtensa-lx106-el
f/bin/:/c/Python27:/c/Espressif/xtensa-lx106-elf/bin:/c/Espressif/utils/ESP8266:
/mingw/bin:/usr/bin:/c/ProgramData/Oracle/Java/javapath:/c/Program Files (x86)/N
VIDIA Corporation/PhysX/Common:/c/Program Files/Broadcom/Broadcom 802.11 Network
 Adapter/Driver:/c/Program Files (x86)/Intel/iCLS Client/:/c/Program Files/Intel
/iCLS Client/:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows
/System32/WindowsPowerShell/v1.0/:/c/Program Files/Intel/Intel(R) Management Eng
ine Components/DAL:/c/Program Files/Intel/Intel(R) Management Engine Components/
IPT:/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/c/Pr
ogram Files (x86)/Intel/Intel(R) Management Engine Components/IPT:/c/Program Fil
es (x86)/Intel/OpenCL SDK/2.0/bin/x86:/c/Program Files (x86)/Intel/OpenCL SDK/2.
0/bin/x64:/c/Program Files (x86)/Skype/Phone/:/c/Program Files/Git/cmd:/c/Progra
m Files/TortoiseGit/bin:/c/Program Files/TortoiseSVN/bin:/d/install/apache-maven
-3.3.9/bin:/c/Program Files (x86)/MySQL/MySQL Fabric 1.5  python /c/Espressif/es
p_iot_sdk_v2.0.0.p1/tools/gen_appbin.py build/httpd.user1.out 2 0        15
  4        0 >/dev/null'
make: *** [firmware/user1.bin] Error 2`

Thank you!