jeelabs/esp-link #106: REST HTTP request to CouchDB problems

ksuie, 2016-02-14:
When I use esp-link(v2.1.7 and v2.2.beta1) REST request to CouchDB(port 5984)
`rest.begin("192.168.2.25", 5984, false);
rest.get("/");`
Debug Log:
REST: status=200, body=0
Can't get any response body.

But REST request to Lighttpd(port 80) is ok.
REST: status=200, body=15
Any suggestions?

tve, 2016-02-16:
What do you expect to receive from couchdb in the response? Could it be that the response it too big to fit?

ksuie, 2016-02-17:
`$ curl http://192.168.2.25:5984
{"couchdb":"Welcome","uuid":"824b5da6197f7d370930e7a7923abec4","version":"1.6.1","vendor":{"version":"1.6.1","name":"The Apache Software Foundation"}}
`
Response size: 150 bytes.
Add receive buffer to 512 bytes
`uint16_t code = rest.waitResponse(response, 512);`
still same problem, REST: status=200, body=0

I use lighttpd to measure response body size, less then 42 bytes is ok.
How can I get bigger response? thanks.

tve, 2016-02-20:
Moved to el-client repo: https://github.com/jeelabs/el-client/issues/4

closed 2016-02-20 by tve