jeelabs/esp-link #107: A way to have a long-lasting request

dbetz, 2016-02-15:
First, thanks for your work on esp-link! It looks like it's going to be a great basis for the project I'm working on.

I have a question though. I have some requests that may take a while to complete. I'd like to have the initial call to my CGI function return HTTPD_CGI_MORE but I'm not really expecting more POST data. I just need time to process the request. Then, later when the request has been completed, I'd like to send the response to the request. It looks like I can call httpdSend to send a response but there does not seem to be any way to flush the buffer since xmitSendBuff is a static function in httpd.c. Could this be made a global function so it can be called outside of httpdProcessRequest? I'd be happy to make the change if you think this is the right approach. If not, could you please suggest how I would do this?

Thanks!
David

tve, 2016-02-16:
Looks to me like it should work!

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve